esp8266/Arduino #6198: ESP shuts down when instantiating std::regex object

Gei0r, 2019-06-12:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-01 (D1 mini)
- Core Version:			2.5.2
- Development Env:		Arduino IDE + Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			WeMos D1 R1
- Flash Size: 		4MB (no SPIFFS)
- CPU Frequency:		80Mhz

### Problem Description

When instantiating a ``std::regex`` object, the D1 mini shuts down.
It only works for *very* simple regexes, such as /a/, but not for (a little) more complicated ones.

I also tried enabling C++ exceptions and catching something, but nothing is caught.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <regex>

void setup() {
  Serial.begin(115200);
  Serial.println("Hi there!"); delay(20);
  static const std::regex holdRegex("hold ([\\d,=]+) (\\d+)");
  Serial.println("After instantiation"); delay(20);
}

void loop() {
}
```

### Debug Messages

```
Hi there!

Exception (3):
epc1=0x4020b4d8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4025a301 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff610 end: 3fffffc0 offset: 01a0
3ffff7b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff7c0:  4025a2f8 00000000 4025a314 402085fa  
3ffff7d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff7e0:  00000000 feefeffe feefeffe feefeffe  
3ffff7f0:  feefeffe feefeffe feefeffe 3ffeed68  
3ffff800:  40207ef0 00000000 3ffeed68 4020897c  
3ffff810:  feefeffe feefeffe feefeffe feefeffe  
3ffff820:  feefeffe feefeffe feefeffe feefeffe  
3ffff830:  feefeffe feefeffe feefeffe feefeffe  
3ffff840:  feefeffe feefeffe feefeffe 3ffeed68  
3ffff850:  40207ef0 3ffffb34 3ffeed68 40208e90  
[... 171 lines cut ...]
0x40205480: std::_Function_handler    > >::_M_invoke(std::_Any_data const&, std::__detail::_PatternCursor const&, std::__detail::_Results&) at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_nfa.h line 97
0x402027f3: std::__detail::_Compiler   >::_M_quantifier() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 866
0x40202c66: std::__detail::_Compiler   >::_M_term() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 795
0x40202908: std::__detail::_Compiler   >::_M_alternative() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 773
0x40202c5d: std::__detail::_Compiler   >::_M_term() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 798
0x40202933: std::__detail::_Compiler   >::_M_alternative() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 777
0x40202933: std::__detail::_Compiler   >::_M_alternative() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 777
0x40202933: std::__detail::_Compiler   >::_M_alternative() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 777
0x40202933: std::__detail::_Compiler   >::_M_alternative() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 777
0x40202933: std::__detail::_Compiler   >::_M_alternative() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 777
0x4020297c: std::__detail::_Compiler   >::_M_disjunction() at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 759
0x402142fe: operator delete(void*) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/del_op.cc line 48
0x40201146: std::_Function_base::_Base_manager    > >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 1954
0x40202da4: std::__detail::_Compiler   >::_Compiler(char const* const&, char const* const&, std::regex_traits &, unsigned int) at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 730
0x40201110: std::_Function_base::_Base_manager    > >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40205480: std::_Function_handler    > >::_M_invoke(std::_Any_data const&, std::__detail::_PatternCursor const&, std::__detail::_Results&) at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_nfa.h line 97
0x4020d803: wctob at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/wctob.c line 22
0x40204f74: malloc(size_t) at C:\Users\bw1ebed0\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1677
0x40202ef0: std::__detail::__compile   >(char const* const&, char const* const&, std::regex_traits &, unsigned int) at c:\users\bw1ebed0\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/regex_compiler.h line 1105
0x4021d080: std::locale::_S_initialize_once() at ../../../../../../dl/gcc-xtensa/libstdc++-v3/src/c++98/locale_init.cc line 267
0x4021d0a6: std::locale::_S_initialize() at ../../../../../../dl/gcc-xtensa/libstdc++-v3/src/c++98/locale_init.cc line 280
0x40203dc6: delay(unsigned long) at C:\Users\bw1ebed0\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 57
0x402030b9: setup() at D:\UserData\bw1ebed0\Documents\Arduino\Regex/Regex.ino line 9
0x402038f4: loop_wrapper() at C:\Users\bw1ebed0\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 122
```

devyte, 2019-06-12:
The entire std C++ isn't supported/tested. In this case I'd say the issue is likely either stack size or free heap, but it could be some dependency as well, such as RTTI which we don't have.

Gei0r, 2019-06-14:
Well ok I rewrote the code to not use regexes, but it's not ideal. Is there a way to get closer the the particular problem in this case?

devyte, 2019-06-14:
I'm for getting as close to standard as possible, so if you could pursue this it would be great.
As I mentioned above, there are two avenues to investigate that come to mind: heap/stack usage, and whether regex somehow requires RTTI.
Maybe @earlephilhower has some ideas how to pursue this?

earlephilhower, 2019-06-26:
It was done before I came onboard, but IIRC the RTTI is disabled in  the toolchain because it inflates the code size and adds bytes to every C++ object.

Theoretically, it's just a switch that needs to be enabled on the g++ toolchain (not just the IDE platform.txt, but the libstdc++ compilation in the esp-quick-toolchain).

Gei0r, 2019-06-27:
I haven't looked into it further, but I don't think that RTTI is the problem. Can't imagine why compiling a regex object would have anything to do with RTTI.

earlephilhower, 2019-07-09:
Yes, it doesn't look like a RTTI issue on second glance a little later.  What it does look like is a memory error generating the regex parse tree.  Your attached crash dump is with exceptions enabled.  Without exceptions enabled, I get abort() called from `esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/regex_compiler.h` line 974:

````
  template<typename _InIter, typename _TraitsT>
    bool
    _Compiler<_InIter, _TraitsT>::
    _M_bracket_expression()
    {
      if (_M_match_token(_ScannerT::_S_token_bracket_begin))
        {
          _RMatcherT __matcher(_M_match_token(_ScannerT::_S_token_line_begin),
                               _M_traits);
          if (!_M_bracket_list(__matcher)
              || !_M_match_token(_ScannerT::_S_token_bracket_end))
>>>>            __throw_regex_error(regex_constants::error_brack);
          _M_stack.push(_StateSeq(_M_state_store,
                                  _M_state_store._M_insert_matcher(__matcher)));
          return true;
        }
      return false;
    }
````

And on further investigation, GCC 4.8 __does not support regex properly__ : https://stackoverflow.com/questions/12530406/is-gcc-4-8-or-earlier-buggy-about-regular-expressions

If you're adventurous, check out my GCC 7.2 PR #6267 .  It might get your regex running.

Closing as this is a known issue in gcc 4.8...

earlephilhower, 2019-07-14:
An update in GCC 9.1.  It runs, but there is not enough RAM to actually compile the `std::regex` and also does seem to need RTTI (I believe dynamic_cast is requires it).  So `std::regex` is not going to be your friend on the 8266.  There's glob-style regex support in the current head, though, if that would be useful.

````
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x4023311b: std::type_info::operator==(std::type_info const&) const at /home/earle/src/test/esp-quick-toolchain/repo/gcc-gnu/libstdc++-v3/libsupc++/typeinfo line 122
EXCVADDR: 0x40265508
Memory allocation of 163852 bytes failed at 0x402078ed: malloc(size_t) at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.cpp line 95

Decoding stack results
0x40233230: __cxxabiv1::__vmi_class_type_info::__do_dyncast(int, __cxxabiv1::__class_type_info::__sub_kind, __cxxabiv1::__class_type_info const*, void const*, __cxxabiv1::__class_type_info const*, void const*, __cxxabiv1::__class_type_info::__dyncast_result&) const at /home/earle/src/test/esp-quick-toolchain/repo/gcc-gnu/libstdc++-v3/libsupc++/vmi_class_type_info.cc line 91
0x402149d0: __ssputs_r at /home/earle/src/test/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 183
0x40214ef5: _svfprintf_r at /home/earle/src/test/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 660
0x40237414: __cxxabiv1::__dynamic_cast(void const*, __cxxabiv1::__class_type_info const*, __cxxabiv1::__class_type_info const*, ptrdiff_t) at /home/earle/src/test/esp-quick-toolchain/repo/gcc-gnu/libstdc++-v3/libsupc++/dyncast.cc line 73
0x402078ed: malloc(size_t) at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.cpp line 95
0x40100a3e: umm_malloc(size_t) at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 1684
0x40225a87: std::use_facet   >(std::locale const&) at /home/earle/src/test/esp-quick-toolchain/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_classes.tcc line 132
0x402061d4: std::__detail::_Compiler   >::_Compiler(char const*, char const*, std::locale const&, std::regex_constants::syntax_option_type) at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/9.1.0/bits/shared_ptr_base.h line 1020
0x40207a8c: realloc_loc(void*, size_t, char const*, int) at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.cpp line 142
0x402062c1: std::__detail::__compile_nfa  , char const*>(char const*, char const*, std::__cxx11::regex_traits ::locale_type const&, std::regex_constants::syntax_option_type) at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/9.1.0/bits/shared_ptr_base.h line 1184
0x4020116c: std::__detail::_Scanner ::_M_eat_escape_ecma() at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/9.1.0/bits/regex_scanner.tcc line 310
0x40207502: delay(unsigned long) at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.cpp line 55
0x40219414: std::locale::locale() at /home/earle/src/test/esp-quick-toolchain/repo/gcc-gnu/libstdc++-v3/src/c++98/locale_init.cc line 269
0x40206399: setup() at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/9.1.0/bits/move.h line 99
0x40207040: loop_wrapper() at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 129
````

earlephilhower, 2019-07-15:
After much hacking, `std::regex` can compile at runtime with the latest GCC 9.1 pre-release, and I can probably backport the changes fo the existing 4.8 GCC.

You can give your app a try with PR #6294 if you want to do further experiments.

Gei0r, 2019-07-16:
Wow, amazing. I'll give it a try on the weekend.

commit ac25f327 (2019-07-21): Upgrade to 2.5.0-4 toolchain w/improved pgm_read_x, C++ exceptions (#6273)

* Upgrade to 2.5.0-4 toolchain w/improved pgm_read_x

Rebuild the entire toolchain (including standard libraries) with the
latest pgm_read_xxx headers included (which fix unaligned dword reads
from progmem and run faster/smaller, and a pgm_read_byte change which
removes an instruction on each read saving flash).

Pull in latest bearssl while we're at it, too, which speeds up EC
handshakes and reduced ROM usage, too.

* Fix C++ exceptions

Exception code now only does 32b aligned reads from progmem to access
the eh_tabl [... cut]
closed 2019-07-21 by earlephilhower