esp8266/Arduino #6200: ESP8266 crashes when trying to concat strings

Najsr, 2019-06-14:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12E
- Core Version:			2.5.2
- Development Env:		Arduino IDE / Visual Studio
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU
- Flash Size: 		4MB
- lwip Variant:   v2 Higher Bandwidth
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

When I try to concat 2 strings, ESP crashes

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

static const char CON[] PROGMEM = "<!DOCTYPE html><html><head><title>MY PAGE INDEX</title><style type='text/css'>* {-webkit-box-sizing: border-box;-moz-box-sizing: border-box;box-sizing: border-box;}.signpost {text-decoration: none;display: block;width:250px;height:50px;background-color: #663399;margin:25px auto;position: relative;color: white;text-align: center;vertical-align: middle;line-height: 50px;font-weight: bold;}.signpost:after {content:'';position: absolute;top:0;height:0; width:0; border-width: 25px;border-style:solid;border-color:#663399;border-top-color:transparent;border-bottom-color:transparent;}.signpost-right:after {border-right-color:transparent;left:100%;}.signpost-left:after {border-left-color:transparent;right: 100%;}a:link { text-decoration: none; color: inherit;}.restart {width: 50%;margin: auto;padding: 5px;text-align: center;border-radius: 20px;background-color: #6495ED;}</style></head><body><a href='ap'><div class='signpost signpost-right'>Access Point Setup</div></a><a href='wifi'><div class='signpost signpost-left'>Wi-Fi networks configuration</div></a><div class='restart'>Reason of restart: <span id='reason'>";
static const char CAT[] PROGMEM = "Web Server pin is set to LOW";

void setup() {
  Serial.begin(115200);
  String data = String(CON);
  data += CAT;
  Serial.println(data);
}

void loop() {
}
```

### Crash event
```
Exception (3):
epc1=0x4020580c epc2=0x00000000 epc3=0x00000000 excvaddr=0x402383c9 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 01a0
3fffff70:  402383c9 0000001c 00000000 feefeffe  
3fffff80:  feefeffe 00000000 3ffee234 40201046  
3fffff90:  00000000 00000000 ffefeffe feefeffe  
3fffffa0:  feefeffe feefeffe 3ffee25c 40201814  
3fffffb0:  feefeffe feefeffe 3ffe84f4 40100461  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

Decoded:
```

Decoding stack results
0x40201046: setup() at C:\Users\David\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/WString.h line 121
0x40201d60: loop_wrapper() at C:\Users\David\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 122
```

fsommer1968, 2019-06-14:
Hi Najsr, check the F-Macro....., maybe is´s caused by unaligned access?

Najsr, 2019-06-14:
> Hi Najsr, check the F-Macro....., maybe is´s caused by unaligned access?

Well I do not understand how to do that. Could you briefly introduce me?
Btw this started happening with version 2.5.1 and now 2.5.2

Najsr, 2019-06-14:
```cpp
static const char FF[] PROGMEM = "12";
static const char FT[] PROGMEM = "56";

void setup() {
  Serial.begin(115200);
  String data = String(FF);
  data += "34";
  Serial.println(data);
  data += FT;
  Serial.println(data);
}

void loop() {
}
```
Even code simple as this causes crash...

Pablo2048, 2019-06-14:
Please read this https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html . And then just use 
```
String data = String(FPSTR(FF));
```

fsommer1968, 2019-06-17:
Hi Najsr,

wouldn´t be a good idea to close this PR now? I suppose that the link provided solved your problem.

Najsr, 2019-06-17:
Yeah my bad, forgot about it. Thanks for the help btw :)

closed 2019-06-17 by Najsr