esp8266/Arduino #6204: Nodemcu suddenly doesn't work anymore after uploading IDE sketch

TechMaxed, 2019-06-15:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			esp12
- Core Version:			v0.9
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

After reinstalling Windows 10 and my Arduino IDE, my sketches don't work in nodemcu anymore. I was able to add the url on the preferences and I set the board to both Nodemcu v0.9 and v1 but still it doesn't work. Here is the console log: (Note: I also tried uploading blink and it wouldn't run)
`esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: dc:4f:22:61:2f:d2
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 320240 bytes to 227296...

Writing at 0x00000000... (7 %)
Writing at 0x00004000... (14 %)
Writing at 0x00008000... (21 %)
Writing at 0x0000c000... (28 %)
Writing at 0x00010000... (35 %)
Writing at 0x00014000... (42 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (57 %)
Writing at 0x00020000... (64 %)
Writing at 0x00024000... (71 %)
Writing at 0x00028000... (78 %)
Writing at 0x0002c000... (85 %)
Writing at 0x00030000... (92 %)
Writing at 0x00034000... (100 %)
Wrote 320240 bytes (227296 compressed) at 0x00000000 in 238.9 seconds (effective 10.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

earlephilhower, 2019-06-26:
Please use the forum https://www.esp8266.com/ for questions like this as it's not an Arduino core issue.  Check cables, connections, power supply, etc.

closed 2019-06-26 by earlephilhower

TechMaxed, 2019-08-28:
@earlephilhower Can you delete this issue because I posted on the wrong forum. Thanks