esp8266/Arduino #6205: MDNS.begin() and ArduinoOTA.begin() work once and then after reset crashes.

cyrusbuilt, 2019-06-16:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[8b899c1] - v2.5.2
- Development Env:		[Platformio]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

There appears to be some sort of issue in the flash using any library that has a setHostname() method (I'm merely guessing at this point).  Both MDNS.begin() and ArduinoOTA.begin() work the first time the sketch boots, then upon reset, I get a wdt reset when calling begin().  It also works exactly once, and then never again.  Erasing the flash with esptool fixes this temporarily.  In other words, I can clear the flash, re-upload the sketch, reset, and then the first boot works flawlessly.  Subsequent reboots (or power on/off) results in a wdt reset when calling MDNS.begin() or ArduinoOTA.begin().  At first I thought the issue was just with the MDNS library since I had seen similar problems, so I removed all the MDNS code only to see the same issue happen when calling ArduinoOTA.begin().

I resolved some of the other issues I've had by adding the following lines with a delay prior to calling WiFi.begin():

WiFi.persistent(false);
WiFi.disconnect(true); 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#ifndef ESP8266
    #error This firmware is only compatible with ESP8266 controllers.
#endif
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>
#include "LED.h"
#include "Relay.h"
#include "TaskScheduler.h"
#include "ResetManager.h"
#include "ESPCrashMonitor-master/ESPCrashMonitor.h"
extern "C" {
  #include "user_interface.h"
}

#define FIRMWARE_VERSION "1.0"

// Configuration
#define ENABLE_OTA                              // Comment this line to disable OTA updates.
#define DEFAULT_SSID "cyrusbuilt"           // Put the SSID of your WiFi here.
#define DEFAULT_PASSWORD "*******"   // Put your WiFi password here.
#define WEBSERVER_PORT 80                       // The built-in webserver port.
#define SERIAL_BAUD 115200                      // The BAUD rate (speed) of the serial port (console).
#define CHECK_WIFI_INTERVAL 30000               // How often to check WiFi status (milliseconds).
#define CHECK_SENSORS_INTERVAL 3000             // How often to check sensors (milliseconds).
#define ACTIVATION_DURATION 500                 // How long the activation relay should be on.
#define DEVICE_NAME "CyGarage"                  // The device name.
#ifdef ENABLE_OTA
    #define OTA_HOST_PORT 8266                     // The OTA updater port.
[... 668 lines cut ...]
~ld

INIT: CyGarage v1.0 booting ...

INIT: Initializing crash monitor... DONE
=========================================

Reset reason: 1, Hardware Watchdog
Exception cause: 4
Exception vector address: 0
EPC1: 1073745768
EPC2: 0
EPC3: 0
DEPC: 0

Reset cause: 4
Boot device: 3
Boot mode:   6
Normal boot: 0
Reset pin:   0
Watchdog:    1
Boot device RAM:   0
Boot device flash: 1
Reset reason code: 1
```

d-a-v, 2019-06-17:
Your MCVE is not "M"inimal.
ArduinoOTA and mDNS behave well in STA and non persistent mode, even with a hostname (pingable).

cyrusbuilt, 2019-06-19:
Fair enough.  I made it code-complete in case I was doing anything else stupid that caused the issue. So I erased the flash again, then stripped the code down to the following:

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>

#define FIRMWARE_VERSION "1.0"

// Configuration
#define SERIAL_BAUD 115200
#define WEBSERVER_PORT 80
#define DEFAULT_SSID "cyrusbuilt"
#define DEFAULT_PASSWORD "*******"
#define DEVICE_NAME "CYGARAGE"
#define OTA_HOST_PORT 8266
#define OTA_PASSWORD "your_ota_password_here"

MDNSResponder mdns;
ESP8266WebServer server(WEBSERVER_PORT);

void initSerial() {
    Serial.begin(SERIAL_BAUD);
    Serial.println();
    Serial.println();
    Serial.print(F("[NET][INIT] CyGarage v"));
    Serial.print(FIRMWARE_VERSION);
    Serial.println(F(" booting ..."));
    Serial.println();
}

void initMDNS() {
    Serial.print(F("INIT: Starting MDNS responder... "));
    if (WiFi.status() == WL_CONNECTED) {
        delay(500);
        if (!mdns.begin(DEVICE_NAME)) {
            Serial.println(F(" FAILED"));
        }

        Serial.println(F(" DONE"));
    }
    else {
        Serial.println(F(" FAILED"));
    }
}

void getDeviceStatus() {
    server.send(200, "text/plain", "OK");
}

void getFirmwareVersion() {
    server.send(200, "text/plain", FIRMWARE_VERSION);
}

void serverOk() {
    server.send(200, "text/plain", "Server Ok...");
}

void activateDoor() {
    Serial.print(F("INFO: Activating door @ "));
    Serial.println(millis());
    server.send(200, "text/plain", "OK");
}

void initWebServer() {
    Serial.print(F("INIT: Initializing web server... "));
    if (WiFi.status() == WL_CONNECTED) {
        server.on("/", serverOk);
        server.on("/device/status", HTTP_GET, getDeviceStatus);
        server.on("/device/activate", HTTP_POST, activateDoor);
        server.on("/version", HTTP_GET, getFirmwareVersion);
        server.begin();
        Serial.println(F("DONE"));
    }
    else {
        Serial.println(F("FAIL"));
    }
}

void connectWifi() {
    WiFi.persistent(false);
    WiFi.disconnect(true);
    delay(1000);
    WiFi.mode(WIFI_STA);
    Serial.println(F("DEBUG: Beginning connection..."));
    WiFi.begin(DEFAULT_SSID, DEFAULT_PASSWORD);
    
    Serial.println(F("DEBUG: Waiting for connection..."));
    int currentTry = 0;
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(F("Try "));
        Serial.println(currentTry);
        currentTry++;
        delay(500);
    }

    if (WiFi.status() != WL_CONNECTED) {
        Serial.println(F("ERROR: Failed to connect to WiFi!"));
    }
    else {
        Serial.println(F("Connected!!"));
    }
}

void initOTA() {
    Serial.print(F("INIT: Initializing OTA updater... "));
    if (WiFi.status() == WL_CONNECTED) {
        ArduinoOTA.setPort(OTA_HOST_PORT);
        ArduinoOTA.setHostname(DEVICE_NAME);
        ArduinoOTA.setPassword(OTA_PASSWORD);
        ArduinoOTA.onStart([]() {
            String type;
            if (ArduinoOTA.getCommand() == U_FLASH) {
                type = "sketch";
            }
            else {
                type = "filesystem";
            }
            Serial.println("INFO: Starting OTA update (type: " + type + ") ...");
        });
        ArduinoOTA.onEnd([]() {
            Serial.println(F("INFO: OTA updater stopped."));
        });
        ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
            Serial.printf("INFO: OTA Update Progress: %u%%\r", (progress / (total / 100)));
        });
        ArduinoOTA.onError([](ota_error_t error) {
            Serial.printf("ERROR: OTA update error [%u]: ", error);
            switch(error) {
                case OTA_AUTH_ERROR:
                    Serial.println(F("Auth failed."));
                    break;
                case OTA_BEGIN_ERROR:
                    Serial.println(F("Begin failed."));
                    break;
                case OTA_CONNECT_ERROR:
                    Serial.println(F("Connect failed."));
                    break;
                case OTA_RECEIVE_ERROR:
                    Serial.println(F("Receive failed."));
                    break;
                case OTA_END_ERROR:
                    Serial.println(F("End failed."));
                    break;
            }
        });
        ArduinoOTA.begin();
        Serial.println(F("DONE"));
    }
    else {
        Serial.println(F("FAIL"));
    }
}

void setup() {
    initSerial();
    connectWifi();
    initMDNS();
    initWebServer();
    initOTA();
}

void loop() {
    delay(100);
}
```

But I'm still getting the following:

```
[NET][INIT] CyGarage v1.0 booting ...

DEBUG: Beginning connection...
DEBUG: Waiting for connection...
Try 0
Try 1
Try 2
Try 3
Try 4
Try 5
Connected!!
INIT: Starting MDNS responder... 
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

hreintke, 2019-06-20:
@cyrusbuilt 
Running your code in my environment gives : 

```
SDK:2.2.1(cfd48f3)/Core:unspecified=0/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:6b9587f

[NET][INIT] CyGarage v1.0 booting ...

scandone
del if0
usl
mode : null
mode : sta(5c:cf:7f:24:2f:6e)
add if0
DEBUG: Beginning connection...
DEBUG: Waiting for connection...
Try 0
Try 1
Try 2
Try 3
Try 4
Try 5
scandone
state: 0 -> 2 (b0)
Try 6
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt

connected with FRITZ!Box Fon WLAN 7360, channel 6
dhcp client start...
Try 7
Try 8
ip:10.0.0.212,mask:255.255.255.0,gw:10.0.0.138
Connected!!
INIT: Starting MDNS responder...  DONE
INIT: Initializing web server... DONE
INIT: Initializing OTA updater... DONE
pm open,type:2 0
```

cyrusbuilt, 2019-06-20:
Well.... That figures.  Maybe I have a defective chip.  I'm using an Adafruit Feather Huzzah ESP8266 Breakout. I bought 2 of them, so I'll try the same code on the other one and see if I experience the same issue.  I used to be able to get it to work at least once with my original code as long as I cleared the flash first.  But after one reset, it would fail indefinitely on either MDNS.begin() or ArduinoOTA.begin().  I thought it was some combination or sequence of calls that was causing it, but I suppose it may just be an underlying hardware issue.  I'll try with the other Huzzah I have and see if I experience the same problem with that one.  If so, maybe I'll have to consider a redesign using a different hardware platform.

cyrusbuilt, 2019-06-22:
Apparently its just a defective chip.  I flashed both the above code and the original code on a new one and both work fine. Closing issue.

closed 2019-06-22 by cyrusbuilt