esp8266/Arduino #6206: Hellomesh example gets stuck, then causes WDT

RudyFiero, 2019-06-17:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [*] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [*] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [*] I have searched the issue tracker for a similar issue.
- [*] If there is a stack dump, I have decoded it.
- [*] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[none]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [256000|other] (serial upload only)

### Problem Description

I load the Hellomesh.INO example. Do a manual reset. It gets through setup. Gets into loop. It enters the scan routine.  It never makes it to     timeOfLastScan = millis();  It never completes this and never gets to complete loop().

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMesh.h>
#include <TypeConversionFunctions.h>
#include <assert.h>

/**
   NOTE: Although we could define the strings below as normal String variables,
   here we are using PROGMEM combined with the FPSTR() macro (and also just the F() macro further down in the file).
[... 318 lines cut ...]
:ref 1

```
This is where it waits for a few seconds before it continues and then continues.

```

### Debug Messages

 ****************************** STACK DUMP ******************************
Decoding 11 results
0x402044bc: TransmissionResult::TransmissionResult(NetworkInfo const&, transmission_status_t) at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFiMesh\src/TransmissionResult.cpp line 42
0x40203f6c: ESP8266WiFiMesh::attemptTransmission(String const&, bool, bool, bool, bool) at c:\users\rudy\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/stl_vector.h line 922
:  (inlined by) ESP8266WiFiMesh::attemptTransmission(String const&, bool, bool, bool, bool) at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFiMesh\src/ESP8266WiFiMesh.cpp line 529
0x402068e6: uart_write at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/uart.cpp line 498
0x402049ac: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x401001c0: millis at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/core_esp8266_wiring.cpp line 186
0x40204cc5: Print::write(char const*) at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Print.h line 60
0x40204d50: Print::println() at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Print.cpp line 178
0x402015cd: loop at C:\Users\Rudy\AppData\Local\Temp\arduino_modified_sketch_801160/HelloMesh.ino line 163
0x402084a0: WiFiServer::write(unsigned char const*, unsigned int) at ?? line ?
0x40205b2c: loop_wrapper() at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/core_esp8266_main.cpp line 125
0x401005b9: cont_wrapper at C:\Users\Rudy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/cont.S line 81

```

RudyFiero, 2019-06-17:
I have tried all IwIP v2 variants and they all crash.  Only V1.4 works.

devyte, 2019-06-17:
@aerlon @d-a-v this probably needs you

aerlon, 2019-06-18:
@RudyFiero 
I found a problem with lwIP2 that got fixed recently and committed to the master branch yesterday (#6194). With that fix applied I'm unable to reproduce your issue, instead the result below is produced. Could you try the most recent master branch and see if that fixes your issue?

```
Note that this library can use static IP:s for the nodes to speed up connection times.
Use the setStaticIP method as shown in this example to enable this.
Ensure that nodes connecting to the same AP have distinct static IP:s.
Also, remember to change the default mesh network password!

Setting up mesh node...
lwIP version is at least 2.0.3. Static ip optimizations enabled.

++++ end of setup 5976
++++ led 15 5977
++++ in scan 5977
++++ in scan after String request 5977
Scanning... AP acquired: MeshNode_456, Ch:1 (-42dBm) ... Connecting... 964
Transmitting
Request sent: Hello world request #0 from MeshNode_123.
Response received: Hello world response #42 from MeshNode_456.
++++ in scan defore timeOfLastScan= 7043
++++ in scan after timeOfLastScan= 7043
Transmission successful.
++++ in latestTransmissionOutcomes check 7050
end of scan 
7051
++++ end of loop 7062
++++ led 15 7062
++++ acceptRequest() 7062
++++ end of loop 7072
++++ acceptRequest() 7072
++++ end of loop 7082
++++ acceptRequest() 7083
++++ end of loop 7093
++++ acceptRequest() 7093
++++ end of loop 7103
++++ acceptRequest() 7103
++++ end of loop 7113
++++ acceptRequest() 7113
++++ end of loop 7123
++++ acceptRequest() 7123
++++ end of loop 7133
++++ acceptRequest() 7133
++++ end of loop 7143
++++ acceptRequest() 7143
++++ end of loop 7153
++++ acceptRequest() 7153
++++ end of loop 7164
++++ acceptRequest() 7164
++++ end of loop 7174
++++ acceptRequest() 7174
++++ end of loop 7184
++++ acceptRequest() 7184
++++ end of loop 7194
++++ acceptRequest() 7194
++++ end of loop 7204
++++ acceptRequest() 7204
++++ end of loop 7214
++++ acceptRequest() 7214
++++ end of loop 7224
++++ acceptRequest() 7224
++++ end of loop 7234
++++ acceptRequest() 7234
++++ end of loop 7244
++++ acceptRequest() 7245
++++ end of loop 7255
++++ acceptRequest() 7255
++++ end of loop 7265
++++ acceptRequest() 7265
++++ end of loop 7275
[... 566 lines cut ...]
Transmission successful.
++++ in latestTransmissionOutcomes check 10086
end of scan 
10088
++++ end of loop 10099
++++ acceptRequest() 10099
++++ end of loop 10109
++++ led 15 10109
++++ acceptRequest() 10109
++++ end of loop 10119
++++ acceptRequest() 10119
++++ end of loop 10129
++++ acceptRequest() 10129
++++ end of loop 10139
++++ acceptRequest() 10140
++++ end of loop 10150
++++ acceptRequest() 10150
++++ end of loop 10160
++++ acceptRequest() 10160
++++ end of loop 10170
++++ acceptRequest() 10170
++++ end of loop 10180
++++ acceptRequest() 10180
++++ end of loop 10190
```

RudyFiero, 2019-06-18:
@aerlon  I installed the current git version. I compiled with V2 Lower Memory and it is working as expected. No more resetting. Joins with a board with 1.4 and with V2.

I'm glad that it had been fixed. I tried a couple of weeks ago (with 2.5.2) and it failed then. I didn't have enough time to go through the it properly and then raise an issue. I'm on vacation now so I decided to take another crack at it.

closed 2019-06-18 by RudyFiero

aerlon, 2019-06-18:
Great!

RudyFiero, 2019-06-18:
Thank you guys for all the work you do.