esp8266/Arduino #6208: ESP 12E After running some time, Randomly device is restart

UdayPert, 2019-06-18:
I am building code in esp12E with NONOS SDK 3.0.0 firmware version 2.5.0

Arduino IDE = 1.8.9
Board: Generic ESP8266 Module
CPU Frequency: 80MHz
Crystal Frequency: 26MHz
Flash Size: 4m(1m SPIFFS)
Flash Mode: DOUT(comaptible)
Flash Frequency: 40Mhz

     We are developing a home automation project, for that we are using ESP-12E. We build UDP/TCP protocols, SPI File systems, Some other functions for my application. Now the problem is I build and flash the code into ESP8266, after running sometimes it's restarting with some exceptions. I am not understanding why the exceptions are coming. Because the same code is running in TI CC3200 Controller also, It's running perfectly.
 
Here I am attaching my Exception logs for different codes, Can anyone help me, please.

powerlog.log](https://github.com/esp8266/Arduino/files/3299980/powerlog.log)
[withoutpower.log](https://github.com/esp8266/Arduino/files/3299982/withoutpower.log)

d-a-v, 2019-06-18:
Please try with core release 2.5.2 using nonos-sdk 2.2.1.
There are too many issues with the sdk used in core release 2.5.0 (which is not the real v3.0 by the way).

UdayPert, 2019-06-18:
I am using Arduino 1.8.9.
  when I go to board manager for selecting SDK, There if I select ESP8266 it's showing all versions, there I am selecting 2.5.2 version after installation it's directly showing 3.0 SDK how can I select, Code version 2.5.2 and SDK 2.2.1 in Arduino IDE

d-a-v, 2019-06-18:
On core-2.5.1 and 2.5.2, if you are using the "generic esp8266" board,
sdk selection appears in Tools menu.
With any other board, sdk is 2.2.1.
When booting in debug mode, sdk version is shown up on serial console.

UdayPert, 2019-06-18:
You are saying about Espressif FW: nonos-SDK 2.2.1(legacy)

If this is right means I will test and update you

d-a-v, 2019-06-18:
> You are saying about Espressif FW: nonos-SDK 2.2.1(legacy)

Yes this is the default one for all boards in core-2.5.1 and core-2.5.2 (and also core-2.4.2).

UdayPert, 2019-06-19:
Hi, I did What you said yesterday, But I am getting same restarts.  I am attaching files of configuration and restart log image
[image]
[image]

devyte, 2019-06-20:
Issue template instructions were not followed: field info wasn't filled in, no MCVE, not even the stack dumps were decoded. 
I don't see a core issue here, nor a way to investigate a core issue.
I suggest reading the issue template instructions carefully, reducing your code to a minimal sketch that shows the issue, reading the docs l, especially the section on crashes. Once done, consider again whether your issue is in the core and open a new ticket. 
BTW, the ESP has quirks, which means that just because your code works on another platform doesn't mean it will work here without modifications. Some quirks that come to mind are: 
1. ISRs must have the icache ram attr specified, or you'll get random crashes 
2. all execution paths in your setup/loop code must adhere to timing restrictions or you'll get random WDTs 
3. The ESP's stack is tiny at 4KB, so don't use big local vars or you'll get random crashes. 
4. Beware use of async libs such as the async udp/tcp, async webserver, or even our own Ticker. There are additional restrictions that apply to code in the callbacks passed to them, and if those restrictions aren't met, you'll get random crashes. 
There are others. 
Finally, if you're using 3rd party libs, they should explicitly state support for the ESP. If the lib author doesn't state such support, then there is no guarantee the lib will work correctly on the ESP and you could get random crashes due to the lib code not handling the quirks. 
Closing.

closed 2019-06-20 by devyte