esp8266/Arduino #6221: ESP-01 SPIFFS write fail on flash chip PUYA P25Q80H

christian-0s, 2019-06-22:
### Basic Infos

- [ :heavy_check_mark: ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ :heavy_check_mark: ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ :heavy_check_mark: ] I have tested that the issue is present in current master branch (aka latest git).
- [ :heavy_check_mark: ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ :heavy_check_mark: ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-01]
- Core Version:			[653f58e2]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Write on SPIFFS fails, with latest github core and also with 2.0.0 patched with [Patch file](https://github.com/letscontrolit/ESPEasy/blob/4744d638e4084ca4051df56ffc82f3f8c245e517/patches/puya_core2_4_x.patch)
If a SPIFFS image is loaded via "Sketch data upload" can be read without problems.
I tried with different SDK versions (legacy, testing and pre-3) but behavior is the same.

Photo of the flash chip:
[image]
 
__________________________

Firmware: 
 Chip Id: 009F41C3
 Core version: 653f58e2
 SDK version: 2.2.2-dev(c0eb301)
 Boot version: 31
 Boot mode: 1
__________________________

Flash chip information: 
 Flash chip Id: 00146085 (for example: Id=001640E0 Manuf=E0, Device=4016 (swap bytes))
 Sketch thinks Flash RAM is size: 1.00 MB
 Actual size based on chip Id: 1.00 MB ... given by (2^( "Device" - 1) / 8 / 1024
 Flash frequency: 40.00 MHz
 Flash write mode: QIO
__________________________

File system (SPIFFS): 
 Total KB: 228.20 KB
 Used KB: 0.00 KB
 Block size: 4096
 Page size: 256
 Maximum open files: 5
 Maximum path length: 32

SPIFFS directory {/} :
__________________________

CPU frequency: 80 MHz

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

[... 384 lines cut ...]
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
SPIFFS_read rc=-10012
read from file: 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
done

```

TD-er, 2019-06-22:
For core 2.5.0 and newer you should set a define and then do a complete rebuild. (the define must be set when building the core libraries)
`PUYA_SUPPORT=1`

Look for "puya" in my platformio.ini file: https://github.com/letscontrolit/ESPEasy/blob/mega/platformio.ini

christian-0s, 2019-06-22:
I included ```#define PUYA_SUPPORT 1``` at the beginning of the sketch before building, is this not sufficient?

d-a-v, 2019-06-22:
> is this not sufficient? 

No

With arduino IDE you can create `platform.local.txt` right next to `platform.txt`, with
```
compiler.c.extra_flags=-DPUYA_SUPPORT=1
compiler.cpp.extra_flags={compiler.c.extra_flags}
```

I wonder @earlephilhower @devyte if we should enable this by default
(and use `-DPUYA_SUPPORT=0` if one explicitely does not want it)

devyte, 2019-06-22:
@d-a-v the puya patch affects only those boards with a puya chip. Enabling it by default would impact everyone else, including real non-clone boards. 
Instead, I suggest documenting the whole thing, including the above instructions, if it isn't already done, and directing anyone with puya problems to readthedocs.

d-a-v, 2019-06-22:
That's true. The overhead is 
* code size in flash for `spi_flash_write_puya()`
* execution time of this code everytime `flash_write()` is called:
```
#if PUYA_SUPPORT
    if (getFlashChipVendorId() == SPI_FLASH_VENDOR_PUYA) {
        rc = spi_flash_write_puya(offset, data, size);
    }
    else
#endif
```

`getFlashChipVendorId()` calls fw's `spi_flash_get_id()`.
It we could add a cache variable for `getFlashChipVendorId() == SPI_FLASH_VENDOR_PUYA` then the overhead would be minimal (if current code has a noticeable overhead).

christian-0s, 2019-06-23:
> > is this not sufficient?
> 
> No (I mean Yes) (don't you not understand me negatively ?)
> 
> With arduino IDE you can create `platform.local.txt` right next to `platform.txt`, with
> 
> ```
> compiler.c.extra_flags=-DPUYA_SUPPORT=1
> compiler.cpp.extra_flags={compiler.c.extra_flags}
> ```

Many thanks, now it works.

closed 2019-06-23 by christian-0s

SergiuToporjinschi, 2019-08-21:
offf that's crazy 😒 I lost 1 day searching for this...