esp8266/Arduino #6223: Conflicting interruptlock between core and dht library

supersjimmie, 2019-06-24:
I installed a fresh Arduino IDE, and added the latest esp8266 core library from git.
Then I added the DHT library, which seems to have a conflict with the core?

```
In file included from C:\Users\me\Documents\Arduino\IoT\IoT.ino:182:0:

C:\Users\me\Documents\arduino-1.8.9\libraries\DHT-sensor-library/DHT.h:67:7: error: redefinition of 'class InterruptLock'

 class InterruptLock {

       ^

In file included from C:\Users\me\Documents\arduino-1.8.9\hardware\esp8266com\esp8266\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:28:0,

                 from C:\Users\me\Documents\arduino-1.8.9\hardware\esp8266com\esp8266\libraries\SoftwareSerial\src/SoftwareSerial.h:27,

                 from C:\Users\me\Documents\Arduino\IoT\IoT.ino:157:

C:\Users\me\Documents\arduino-1.8.9\hardware\esp8266com\esp8266\cores\esp8266/interrupts.h:19:7: error: previous definition of 'class InterruptLock'

 class InterruptLock {

       ^

```

d-a-v, 2019-06-24:
Our class name exists for long time but is only recently used
@dok-net (SoftwareSerial) will comment

I think we can rename this generic name in our core
depending on what other maintainers think of it.

In the meantime, you can manually/locally rename InterruptLock in DHT.h.
If it's an esp8266 specific library, you can also tell the author to check if he could use the core version of this class.

dok-net, 2019-06-24:
@d-a-v @supersjimmie There's a PR in Adafruits repo now: https://github.com/adafruit/DHT-sensor-library/pull/135

dok-net, 2019-06-26:
@supersjimmie Are you getting a working combination by default download methods already now? The closing integration of public release 5.2.2 via ESP8266 Arduino submodule libraries/SoftwareSerial is still awaiting, and without trying it all by myself, I expect the name conflict to continue until then?

devyte, 2020-02-02:
Closing because #6225 is merged.

closed 2020-02-02 by devyte