esp8266/Arduino #6232: MQTT Connection with BearSSL and TLS 1.2 results in exception 28

vishalkothari, 2019-06-26:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			        ESP-12
- Core Version:			2.5.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  		Nodemcu 1.0
- Flash Size: 		4MB
- lwip Variant:          v2 Lower Memory
- CPU Frequency:	80Mhz
- Upload Speed:      115200

### Problem Description
I am trying to connect to MQTT  over TLS 1.2 using BearSSL. I am getting exception 28 when trying to connect. 
I tried using web sockets API instead MQTT and it seems to work after some retries with exception 28.

I looked at #5347, and my rootCert seems to be ok as Websocket connection works.
I also looked at #4134, but latest version of umm_malloc doesnt seem to be compatible with ESP core 2.5.2

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <time.h>
#include <sys/time.h>                   // struct timeval
#include <coredecls.h>
#include <PubSubClient.h> //from https://github.com/Imroy/pubsubclient

#define TZ              5.5       // (utc+) TZ in hours
#define DST_MN          0      // use 60mn for summer time in some countries
#define TZ_MN           ((TZ)*60)
#define TZ_SEC          ((TZ)*3600)
#define DST_SEC         ((DST_MN)*60)

const char *ssid = "<ssid>";
const char *pass = "<password>";

const char *   host = "<broker>.amazonaws.com";
const uint16_t  port = 8883;

BearSSL::WiFiClientSecure wifiClient;
PubSubClient pubsubclient(wifiClient, host, port); 

// Set time via HTTP GET service, as required for x.509 validation
time_t setClockHttp() {
  HTTPClient http;
  time_t now;
  //.....
  //Get epoch from http service
  //....

  timeval tv = { now, 0 };
  timezone tz = { TZ_MN + DST_MN, 0 };

  settimeofday(&tv, &tz);
  gettimeofday(&cbtime, NULL);
  now = cbtime.tv_sec;
  return now;
[... 213 lines cut ...]
0x4024328b: ppTxPkt at ?? line ?
0x40236643: ieee80211_output_pbuf at ?? line ?
```
OR
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 19 results
0x4020ab40: BearSSL::PrivateKey::isRSA() const at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/BearSSLHelpers.cpp line 728 (discriminator 1)
0x40208b6c: esp_yield at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/core_esp8266_main.cpp line 91
0x4020914b: delay at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/core_esp8266_wiring.cpp line 54
0x402033a1: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/ClientContext.h line 136
:  (inlined by) WiFiClient::connect(IPAddress, unsigned short) at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 170
0x4020485d: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.cpp line 231
0x4020b580: std::function ::swap(std::function &) at ?? line ?
0x4020b580: std::function ::swap(std::function &) at ?? line ?
0x40206ed4: PubSubClient::connect(MQTT::Connect&) at C:\vishal data\arduino-1.8.3\libraries\pubsubclient-master\src/PubSubClient.cpp line 185
0x40207cbc: String::String(char const*) at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/WString.cpp line 36
0x40207e70: String::String(String const&) at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/WString.cpp line 41
0x4020710d: PubSubClient::connect(String) at C:\vishal data\arduino-1.8.3\libraries\pubsubclient-master\src/PubSubClient.cpp line 168
0x4020b458: std::function ::swap(std::function &) at ?? line ?
0x40203e00: BearSSL::WiFiClientSecure::_installClientX509Validator() at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.cpp line 927
0x402010bc: loop at C:\vishal data\iot\esp code\esp-awsiot-bearssl/esp-awsiot-bearssl.ino line 235
```

Any inputs on this will be very helpful. Thanks.

earlephilhower, 2019-06-26:
The problem looks to be here:
````
**  BearSSL::X509List cert(ca_cert);
  wifiClient.setTrustAnchors(&cert);  
**  BearSSL::X509List client_crt(client_cert);
**  BearSSL::PrivateKey key(client_key);
  wifiClient.setClientRSACert(&client_crt, &key);
````

You're passing in stack local variables to the persistent connection.  Those variables (cert, client_crt, key) cease to be valid the moment you exit the function, and they're needed later on as part of handshaking so that causes very bad things to happen.

Move them to global space and all should be well.

vishalkothari, 2019-06-27:
Thanks for your feedback @earlephilhower. 
I made the changes you mentioned and modified sketch looks like below

```
#include <ESP8266HTTPClient.h>
#include <time.h>
#include <sys/time.h>                   // struct timeval
#include <coredecls.h>
#include <PubSubClient.h> //from https://github.com/Imroy/pubsubclient

#define TZ              5.5       // (utc+) TZ in hours
#define DST_MN          0      // use 60mn for summer time in some countries
#define TZ_MN           ((TZ)*60)
#define TZ_SEC          ((TZ)*3600)
#define DST_SEC         ((DST_MN)*60)

const char *ssid = "<ssid>";
const char *pass = "<password>";

const char *   host = "<broker>.amazonaws.com";
const uint16_t  port = 8883;

static const char ca_cert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIE0zCCA7ugAwIBAgIQGNrRniZ96LtKIVjNzGs7SjANBgkqhkiG9w0BAQUFADCB
yjELMAkGA1UEBhMCVVMxFzAVBgNVBAoTDlZlcmlTaWduLCBJbmMuMR8wHQYDVQQL
ExZWZXJpU2lnbiBUcnVzdCBOZXR3b3JrMTowOAYDVQQLEzEoYykgMjAwNiBWZXJp
....
4fQRbxC1lfznQgUy286dUV4otp6F01vvpX1FQHKOtw5rDgb7MzVIcbidJ4vEZV8N
hnacRHr2lVz2XTIIM6RUthg/aFzyQkqFOFSDX9HoLPKsEdao7WNq
-----END CERTIFICATE-----
)EOF";

static const char client_cert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIDWjCCAkKgAwIBAgIVAOFlUZIDMrUht5zbLrjKog1uWrUKMA0GCSqGSIb3DQEB
CwUAME0xSzBJBgNVBAsMQkFtYXpvbiBXZWIgU2VydmljZXMgTz1BbWF6b24uY29t
...
mJKLnqz8EkxQB6qd2/7XimHrmYoo/DI1KjHKfxEFxUnwkOp1wC6sh78bPXordDTL
gBWeKhwdZHZ3d6TdkY+tpmbSE13+n1+4kUit901F4NIDecdMlIN3zJqQwTPsyw==
-----END CERTIFICATE-----
)EOF";

static const char client_key[] PROGMEM = R"KEY(
-----BEGIN RSA PRIVATE KEY-----
MIIEowIBAAKCAQEAvdXSakGZNJoGlnY7B/Q7FZNRToZzl5QUAYaOvxZRawnKWfd0
b70e7rSSphbf+GNtvKM7S2iXtMH9qz3Vmv7qVM0CxfwzFi//z3m3ncmrqOSkLSuT
....
ut8HxDMV+AmxMDPTnC+5dEhduIw8r1mMBiYr7TXcu5vnDOgzlCq+S6RyWuOVcF2m
YJ8BWgR5mp4KsuPj/eczZLnKgtDzVLoGjg5VE4dPYKypBYHzeeqX
-----END RSA PRIVATE KEY-----
)KEY";

BearSSL::WiFiClientSecure wifiClient;
PubSubClient pubsubclient(wifiClient, host, port); 
BearSSL::X509List cert(ca_cert);
BearSSL::X509List client_crt(client_cert);
BearSSL::PrivateKey key(client_key);

// Set time via HTTP GET service, as required for x.509 validation
time_t setClockHttp() {
  HTTPClient http;
  time_t now;
  //.....
  //Get epoch from http service
  //....

  timeval tv = { now, 0 };
  timezone tz = { TZ_MN + DST_MN, 0 };

[... 107 lines cut ...]
0x40104278: call_user_start_local at ?? line ?
0x4010427e: call_user_start_local at ?? line ?
0x4010000d: call_user_start at ?? line ?
0x40100530: cont_ret at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/cont.S line 142
0x401004dd: cont_continue at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/cont.S line 51
0x4025b720: node_remove_from_list at ?? line ?
0x401020a2: wDev_ProcessFiq at ?? line ?
0x4021f9c8: sha2big_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2big.c line 165
:  (inlined by) br_sha384_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2big.c line 207
0x40220233: sha2small_out at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2small.c line 249
0x4025b464: node_remove_from_list at ?? line ?
0x4022cfe1: br_multihash_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/multihash.c line 141
0x4024412b: pp_attach at ?? line ?
0x4024417a: pp_attach at ?? line ?
0x40244286: pp_attach at ?? line ?
0x40243223: ppTxPkt at ?? line ?
0x402365db: ieee80211_output_pbuf at ?? line ?
0x401047f7: wdt_feed at ?? line ?
0x40100d72: pp_post at ?? line ?
0x40213999: glue2esp_linkoutput at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 299
0x40100d72: pp_post at ?? line ?
....
```

Please let me know your inputs on this.

vishalkothari, 2019-06-29:
@earlephilhower, I also tested this on core version 2.5.0 and I get below error 

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 92 results
0x40231f4e: ieee80211_crypto_decap at ?? line ?
0x4010471f: chm_get_current_channel at ?? line ?
0x40238ccb: sta_input at ?? line ?
0x40242757: pp_tx_idle_timeout at ?? line ?
0x40242017: ppPeocessRxPktHdr at ?? line ?
0x402098ff: loop_task(ETSEventTag*) at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/core_esp8266_main.cpp line 133
0x40104814: call_user_start_local at ?? line ?
0x4010481a: call_user_start_local at ?? line ?
0x4010000d: call_user_start at ?? line ?
0x40100a5c: cont_ret at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/cont.S line 142
0x40100a09: cont_continue at C:\Users\vikothar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/cont.S line 51
0x4025aa10: node_remove_from_list at ?? line ?
0x40104656: lmacTxFrame at ?? line ?
0x4010425b: lmacMSDUAged at ?? line ?
0x40103bce: lmacRecycleMPDU at ?? line ?
0x40103659: lmacProcessTXStartData at ?? line ?
0x40103656: lmacProcessTXStartData at ?? line ?
0x40102576: wDev_ProcessFiq at ?? line ?
0x40104420: lmacProcessAckTimeout at ?? line ?
0x401023cc: wDev_ProcessFiq at ?? line ?
0x40104656: lmacTxFrame at ?? line ?
0x4010425b: lmacMSDUAged at ?? line ?
0x40103bce: lmacRecycleMPDU at ?? line ?
0x40104165: lmacMSDUAged at ?? line ?
0x40104420: lmacProcessAckTimeout at ?? line ?
0x4010254b: wDev_ProcessFiq at ?? line ?
0x401023cc: wDev_ProcessFiq at ?
```

earlephilhower, 2019-07-06:
@vishalkothari, None of the traces you attached seem to have crashes in BearSSL.  It looks like memory corruption somewhere, as you see lots of ROM code in the backtraces.

I can't use the MCVE as it's obviously missing a valid keypair (you don't want to publish your AWS keys online!).  Are there any public MQTT services you could connect to and report the error with (and share the pub/priv keys and cert)?

vishalkothari, 2019-07-12:
@earlephilhower, Thanks. I tried to run the code against broker test.mosquitto.org with self signed certificates. Surprisingly, it worked well. I switched AWS rootCA from Verisign certificate which is marked as Legacy to new AWS Root CA 1 and regenerated Client cert and keys. This resulted in successful connection. Please note that verisign root CA still works ok with other MQTT clients like MQTT.fx but not from ESP.

I am still getting disconnected from AWS every 20 seconds or so. But, since the connection is working fine on test.mosquitto.org, I will close this issue.

closed 2019-07-12 by vishalkothari