esp8266/Arduino #6237: attachinterrupt(); Causes ESP to crash

ubuntusuperuser, 2019-06-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Wemos D1 mini v2.3.0
- Core Version:			2.5.1
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Size: 		      4MB
- lwip Variant:                v2 Lower Memory
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:             115200

### Problem Description

Detailed problem description goes here.

when "attachInterrupt();" is called the esp crashes immediately
this happens with V2.5.2 and V2.5.1 but not with V2.5.0 and lower (as downloaded from the Arduino board manager)

I Don't believe its a hardware problem because tested this on multiple boards with a multiple good power supply's 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#define interruptPin 2

void setup() {
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), ISR, RISING);
}

void loop() {
}

void ISR() {
}

```

### Debug Messages

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vac02aff5
~ld

```

Pablo2048, 2019-06-28:
Your ISR has missing ICACHE_RAM_ATTR decorator...

earlephilhower, 2019-06-28:
@Pablo2048 is correct, thanks.  Closing.

closed 2019-06-28 by earlephilhower