esp8266/Arduino #6238: ESP 12E  restarts with an exception 9 error....

123tron, 2019-06-28:
hi to all ...the following code is a code for an application which gets its input from an mobile app... whenever the code receives data from the app ... esp restarts with the exception 9 error...kindly suggest a solution... i have attached the stack and error statement at the end

and also the decoded exception....

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <EEPROM.h>
#define MAX_SRV_CLIENTS 2
WiFiClient serverClients[MAX_SRV_CLIENTS];
unsigned int q1,q11,RX_D,flag_D,RX_ip,flag_ip,q12,old_flag=0,V1,v11,flag_rep,C1,c11;
char RX[50],D_Type[50],ip_Type[50],D_rep[101];
unsigned int ip1,ip2,ip3,ip4,pn1,start=0,start1=0,start2=0,start3=0,f1; 
WiFiServer server1(80);
WiFiServer server2(1800);
unsigned int o,o1,Radj,Gadj,Badj,Length_D,ipvalue=0,flag,recieve_data,recieve_data1,recieve_data2,i,i1,j,restartflag,y,flagN=1,flagY=1,flagB=1,flag1,y2,y3,i2,i3,i4,flagP=0,z=0,pn=0,flag2,flagP1=0,i5,i6;
char d,h[50],h1[101],h2[50],e,N[50],P[50],N1[50],P1[30],old[30];
//HTTPClient http; 
String colour,ddd,www,pwd,host,nm,data,data1,PN,hostnm,port,ssid,password,DN,ipout,Newip,Newpn,Nm1,Nm2,D_rep1,D_type1,hst4;
char old_ssid[]="MOWITRAA12345678";
char old_dn[]="MOWITRA";
char old_port[]="80";
//char a[] = "http://app.nanonss.com/v.asmx/o?b=";
//char b[] = "&c=EMP00011011123&d=2&e=3";
//char c[120];
char hst1[]="@",hst2[15],hst3[]="!",hstname[100],hst5[100],V_data[100];

char buff[]="\0";
unsigned int addr=0,x=0,count=0;
extern "C" {
#include "user_interface.h"
}
void setup() 
{
  EEPROM.begin(512); 
  Serial.begin(9600);
  delay(10);
  
  Length_D=int(EEPROM.read(0));
  delay(10);
  int Length_P=int(EEPROM.read(1));
  delay(10);
  int Length_IP=int(EEPROM.read(0x22));
  delay(10);
  int Length_PO=int(EEPROM.read(0x23));
  delay(10);
  
  for(int l=0;l<Length_IP;l++) //ip address given 
  {
   if((EEPROM.read(0x24+l))=='.') {ipvalue++;l++;}
   if(ipvalue==0)
   ip1=ip1*10+int(EEPROM.read(0x24+l))-0x30;                                 //Read one by one with starting address of 0x24+l  
   if(ipvalue==1)
   ip2=ip2*10+int(EEPROM.read(0x24+l))-0x30;                                 //Read one by one with starting address of 0x24+l  
   if(ipvalue==2)
   ip3=ip3*10+int(EEPROM.read(0x24+l))-0x30;                                 //Read one by one with starting address of 0x24+l 
   if(ipvalue==3)
   ip4=ip4*10+int(EEPROM.read(0x24+l))-0x30;                                 //Read one by one with starting address of 0x24+l 
  } 
  
  for(int l=0;l<Length_PO;l++)                                              //port adress of output ip
  {
  pn1=pn1*10+int(EEPROM.read(0x31+l))-0x30; //Read one by one with starting address of 0x31+l   
  }  
  
  for(int l=0;l<Length_D;l++)                                             //Device name
  {
   nm = nm + char(EEPROM.read(0x02+l)); //Read one by one with starting address of 0x02    
  }  
  
[... 490 lines cut ...]
3fffff40:  40205f78 00000000 00001388 40204a92  
3fffff50:  3ffef0c8 00000000 40206020 0102a8c0  
3fffff60:  00000000 00000000 00000000 00000000  
3fffff70:  40206020 0102a8c0 00000000 feefeffe  
3fffff80:  00000007 00000007 3fffff40 3fffff30  
3fffff90:  00000021 3ffef0c8 feefeffe feefeffe  
3fffffa0:  00000000 00000000 00000001 3ffef26c  
3fffffb0:  3fffdad0 00000000 3ffef265 402059fc  
3fffffc0:  feefeffe feefeffe 3ffe8534 40100a49  
<<<stack<<<

///////////////////////

decoded exception:::::::::::::::::::::::::

0x40205c94: Stream::readBytes(unsigned char*, unsigned int) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Stream.h line 92
0x40203c40: ESP8266WiFiSTAClass::status() at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 498
0x402032dc: loop at E:\5.UMAMAHESWARAN\Extra studied\chip_name\chip_name_last20062019/chip_name_last20062019.ino line 377
0x40202c09: loop at E:\5.UMAMAHESWARAN\Extra studied\chip_name\chip_name_last20062019/chip_name_last20062019.ino line 158
0x40205f78: pgm_read_byte_inlined at pgmspace.cpp line ?
0x40204a92: WiFiServer::begin(unsigned short) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiServer.cpp line 96
0x40206020: pgm_read_byte_inlined at pgmspace.cpp line ?
0x40206020: pgm_read_byte_inlined at pgmspace.cpp line ?
0x402059fc: loop_wrapper at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 125
0x40100a49: cont_wrapper at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/cont.S line 81

d-a-v, 2019-06-28:
We are not supposed to debug your app. 
However this is not good:
```
size_t len1 = serverClients[C1].available();
uint8_t sbuf1[len1];
```
You must limit `len1` within a fair size like 128 or allocate in heap and test the result.
```
size_t len1 = std::min(128, serverClients[C1].available());
uint8_t sbuf1[len1];
```
Then enable debug options (tools menu), and also add some `print(Esp.getFreeHeap())`.

123tron, 2019-07-01:
sorry for the late reply.... i have done a blunder mistake in array ....now the system works fine...thanks for ur help .....

closed 2019-07-01 by 123tron