esp8266/Arduino #6240: Zip Download fails

Buzzardbait, 2019-06-28:
I see this issue was encountered several years ago and now closed but I can't download the ESP8266 Core zip file and install it. It downloads as Arduino-Master.zip but IDE 1.8.9 says it's not a valid zip file. I'm on Windows 8.1 and I have successfully downloaded Github zips before. Must be something really stupid but why such a fundamental problem? Certifications have something to do with it? What to do about that?

earlephilhower, 2019-06-28:
The IDE cannot use source ZIPs downloaded from github.  Please see the README if you want to use the IDE (there is a specific URL to add to boards manager), or how to make clone using GIT/python (also in the readme.md).

closed 2019-06-28 by earlephilhower

Buzzardbait, 2019-06-28:
Thanks. I'll check that out.