esp8266/Arduino #6245: Esp8266 CH340G ESP_12E Errors

notSoTechnical, 2019-07-01:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

notSoTechnical, 2019-07-01:
Sketch uses 240025 bytes (22%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32960 bytes (40%) of dynamic memory, leaving 48960 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

TD-er, 2019-07-01:
Maybe you could add some more info on what actually the problem is.
I only see an error regarding upload of the sketch, but nothing about the used flash tool (hardware/software).
Make sure your ESP unit is connected through a good USB cable and maybe you can also try using an USB3 port (is capable of handling more current)
Also you may want to look at the available COM ports in the device manager (assuming you're using Windows, it is not mentioned in the opening post) and also if it remains present when trying to flash the unit.
Is there anything connected to the board? For example a lot of Wemos D1 mini boards I recently purchased have all had voltage regulators which are not capable of handling > 150 mA.
This is not enough and you may need to add some capacitors over the GND and 3v3 line of the board.

notSoTechnical, 2019-07-01:
Thanks for replying.  I followed all the steps from this website: https://www.how2electronics.com/home-automation-with-nodemcu-alexa/#Source_CodeProgram

I tried:
1. Replaced my micro USB cable;
2. Changed USB port;
3. Pressed the "flash" before and while plugging the USB cable; 
4. Decreased the upload speed;
5. Rewiring the board.  Supplied 5V to the 3V pin of Node MCU;
-----Still not working!  
6. Installed a driver (CP210x USB to UART Bridge VCP Drivers) from https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers
It works!  The COM Port showed a new port (I assumed that was the Node MCU).  

**However, Alexa couldn't find the device after I uploaded the code.**  

**Serial monitor:** 
Sketch uses 240025 bytes (22%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32960 bytes (40%) of dynamic memory, leaving 48960 bytes for local variables. Maximum is 81920 bytes.
Uploading 244176 bytes from /var/folders/gb/17sgptk51yqdb44sb6f5d8t00000gn/T/arduino_build_115413/nodemcu_home_automation.ino.bin to flash at 0x00000000
................................................................................ [ 33% ]
................................................................................ [ 66% ]
...............................................................................  [ 100% ]

**Output:** 
Sketch uses 240025 bytes (22%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32960 bytes (40%) of dynamic memory, leaving 48960 bytes for local variables. Maximum is 81920 bytes.
Uploading 244176 bytes from /var/folders/gb/17sgptk51yqdb44sb6f5d8t00000gn/T/arduino_build_115413/nodemcu_home_automation.ino.bin to flash at 0x00000000
................................................................................ [ 33% ]
................................................................................ [ 66% ]
...............................................................................  [ 100% ]

I don't know what to do now... I'm very new to this.

TD-er, 2019-07-01:
If installing a CP210x driver does help, then at least the topic title is wrong and you don't have a CH340 USB to serial converter.  (CP210x is a small square chip, CH340 is a rectangular DIL chip)

Also wiring 5V to the ESP chip is not a good idea. That's almost a guarantee for damaging the chip.

Apparently the hardware issues are over or else you cannot upload data to the device.

I don't know anything about the Alexa sketch, so I guess you could ask the author of that sketch.
This repository is about the core libraries (esp8266/Arduino) and not about interfacing to Alexa.

You should at least make sure your device is connected to the network, so somewhere you should have to enter your WiFi credentials.
Either in the code you copied from that howto page, or the ESP node will start an access point to which you have to connect using your phone (or something with WiFi) and enter the credentials on a web page served by the ESP node.  (very likely the IP-address you  then have to visit is 192.168.4.1)
So you have to check:

- Did you set the WiFi credentials somewhere before compiling and uploading to the node?
- Do you need to connect to a (new) accesspoint served by the ESP node when it boots? (and after connecting, set the WiFi credentials)

5chufti, 2019-07-03:
looking at the above mentioned website I see that you are not the only one having problem with "device discovery". As this seems to be a problem with the used sketch/library and NOT of the esp arduino core, please close this issue (as to "free the dev's desk"). 
You may be better off contacting the author of the sketch/library or looking for help on boards like https://www.esp8266.com

closed 2019-07-03 by notSoTechnical