esp8266/Arduino #6249: ESP V2.5.2 AP WiFi is not seen on just one mobile (V2.5.0 is OK)

caipifrosch, 2019-07-03:
#### Platform

- Hardware:					[ESP-12]
- Core Version:			[V2.5.2 Release]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		       [4MB]
- lwip Variant:             [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	       [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:           [other] (serial upload only)

### Problem Description

WiFi Setup of XIAOMI Mi A2 (Android one) does not find the ESP AP with Versions 2.5.2 and 2.5.1, while other mobile, PC or iPad devices will show ESP SSID. Same applies to latest Git (03.07.2019).

Reverting back to V2.5.0 always fixes this issue. 

Tested back and forth multiple times: 
ESP is not seen by Mi A2 mobile on any Version above V2.5.0.

Thanks a lot for your help/advice in advance.

### Sketch / AP Setup

void setupAP(void) {

  Serial.print("WLAN: Starting Access Point... SSID: ");
  Serial.println(network.apSSID);

  WiFi.persistent(false);

  WiFi.disconnect();
  WiFi.softAPdisconnect(true);
  WiFi.hostname(network.apSSID);

  wifi_set_opmode(STATIONAP_MODE);

  WiFi.softAPConfig((IPAddress)network.apIP, (IPAddress)network.apIP, (IPAddress)network.wlanNM);

  // start AP
  Serial.print("WLAN: Starting SoftAP on Channel ");
  Serial.print(String(network.apCH,DEC));

  if (!WiFi.softAP(network.apSSID, network.apPass, network.apCH, 0, MAX_AP_CONS))
  {

    // AP setup failed
    Serial.println("... failed!");

    ESP.restart();

  }
  else {

    // AP setup and running...
    Serial.println(" ... started.");

    launchWeb(1);

  }

}

TD-er, 2019-07-04:
Is it unable to spot the AP's SSID of the node, or is it not capable of making (or maintaining) a connection?

caipifrosch, 2019-07-04:
This specific mobile is unable to spot the AP SSID. Other devices will see AP SSID.

TD-er, 2019-07-04:
Have you also tried using some network analyzer app?
Not that such an app will help you connect, but it may be some indication of the layer where the problems arise.

caipifrosch, 2019-07-04:
SDK is pre3 in V2.5.0 and SDK V2.2.1 in core V2.5.2

My best guesss is, that this may be related to:

#5784 - SDK issues (SDK reverted from pre3 to 2.2.1) 

What kind of tool do you suggest?

TD-er, 2019-07-04:
I have several on my phone.
The one I use the most for WiFi issues is [WiFiAnalyzer (open-source)](https://play.google.com/store/apps/details?id=com.vrem.wifianalyzer)
But there are several, also with the same name ;)

devyte, 2019-07-05:
Given that this is specific to that phone, I'd say it's a compatibility issue in Espressif's SDK, especially since the ssid visibility is due to the beacon broadcast. 
I suppose you could try to tweak some parameters, like the beacon time etc, and see if that helps you. See the SoftAP source and Espressif doc 2c api reference for details on which AP parameters can be given to the sdk.
I'm closing this, because I don't think it's a core issue, but feel free to discuss further. If you do find a core issue, @ me or just open a new issue and fill in the required info.

closed 2019-07-05 by devyte