esp8266/Arduino #6253: LeaMDNS lambda capture initializer C++ standard warnings

earlephilhower, 2019-07-04:
We build the core with `--std=c++11`, but the following warning about using something only available under C++14 or later comes up with `-wextra`.

@LaborEtArs, can you please take a quick look at these and see if we're good or if the compiler is silently ignoring your initializer?

````
C++ ../../libraries/ESP8266WiFi/src/ESP8266WiFiSTA-WPS.cpp.32.o
../../libraries/ESP8266mDNS/src/LEAmDNS.cpp: In member function ‘bool esp8266::MDNSImplementation::MDNSResponder::setHostProbeResultCallback(esp8266::MDNSImplementation::MDNSResponder::MDNSHostProbeFn1)’:
../../libraries/ESP8266mDNS/src/LEAmDNS.cpp:1054:40: warning: lambda capture initializers only available with -std=c++14 or -std=gnu++14
     return setHostProbeResultCallback([resp=std::ref(*this), pfn](const char* p_pcDomainName, bool p_bProbeResult) { pfn(resp, p_pcDomainName, p_bProbeResult); });
                                        ^~~~
../../libraries/ESP8266mDNS/src/LEAmDNS.cpp: In member function ‘bool esp8266::MDNSImplementation::MDNSResponder::setServiceProbeResultCallback(esp8266::MDNSImplementation::MDNSResponder::hMDNSService, esp8266::MDNSImplementation::MDNSResponder::MDNSServiceProbeFn1)’:
../../libraries/ESP8266mDNS/src/LEAmDNS.cpp:1083:55: warning: lambda capture initializers only available with -std=c++14 or -std=gnu++14
     return setServiceProbeResultCallback(p_hService, [resp=std::ref(*this), p_fnCallback](const char* p_pcServiceName, const hMDNSService p_hMDNSService, bool p_bProbeResult) {
                                                       ^~~~

````

dok-net, 2019-07-05:
I'm the one that patched these lines, going for the minimum diff in the first round 2d9253e46c , starting off from:

`return setHostProbeResultCallback(std::bind(pfn, std::ref(*this), _1, _2));
`
and

`return setServiceProbeResultCallback(p_hService, std::bind(p_fnCallback, std::ref(*this), _1, _2, _3));`

I would like to think that this is equivalent:

`return setHostProbeResultCallback([this, pfn](const char* p_pcDomainName, bool p_bProbeResult) { pfn(*this, p_pcDomainName, p_bProbeResult); });
`
and

```
return setServiceProbeResultCallback(p_hService, [this, p_fnCallback](const char* p_pcServiceName, const hMDNSService p_hMDNSService, bool p_bProbeResult) {
  p_fnCallback(*this, p_pcServiceName, p_hMDNSService, p_bProbeResult);
});

```
@hreintke , can you comment on why the former `std::ref(*this)` form was chosen over binding `this `as copy dereferencing it in the actual parameter list?

hreintke, 2019-07-06:
@dok-net @d-a-v Because of my knowledge.
It was : `return setServiceProbeResultCallback(p_hService, std::bind(p_fnCallback, *this, _1, _2, _3));`
But the bug in there was that the *this was copied as all std::bind parameters are by value.
Looking for solutions I found std::ref and used it. Did not consider other options as this worked OK.

PR #6278 (2019-07-10): Fix compiler warning about lambda capture initializers
Fixes #6253
closed 2019-07-10 by earlephilhower