esp8266/Arduino #6255: I2C

TuxRobotics, 2019-07-04:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			ESP-12e
- Core Version:		6c5269a74b19c7a0c5889f0f48a31e3cf24bfeda
- Development Env:	Arduino IDE 1.8.5
- Operating System:	Linux 4.15.0-45-generic

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			?
- Flash Size: 		4MB, 2MB SPIFFS
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	?
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

The Wire library always sets the first bit of the first data byte after an address.  The example sketch below is copied directly out of the examples, with the only modification being the addresses and data sent.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Wire.h>
#include <PolledTimeout.h>

#define SDA_PIN 4
#define SCL_PIN 5
const int16_t I2C_MASTER = 0x42;
const int16_t I2C_SLAVE = 0x56 >>1;

void setup() {
  Wire.begin(0x03);
  Wire.setClock(400000);  // use 400 kHz I2C
}

byte x = 0;

void loop() {
  using periodic = esp8266::polledTimeout::periodic;
  static periodic nextPing(1000);

  if (nextPing) {
    Wire.beginTransmission(I2C_SLAVE); // transmit to device #8
    Wire.write((uint8_t)0);        // sends one byte
    Wire.write((uint8_t)0);        // sends one byte
    Wire.endTransmission();    // stop transmitting

    x++;
  }
}

```

### Debug Messages

Runs normally, first bit of the data byte is always set:

```
Debug messages go here
```

TuxRobotics, 2019-07-04:
Avoided the issue by using brzo i2c library instead.

closed 2019-07-04 by TuxRobotics

devyte, 2019-07-05:
@TuxRobotics beware brzo, it is GPL, unlike our core. My understanding is that it's ok to use with our core only for personal use, and can't be distributed, because we rely on closed source prebuilt libs that aren't ours, and are therefore incompatible with any GPL.

About the issue, I'm rather surprised that nobody has seen it before. Is there any chance you could investigate our lib and try to find a fix?