esp8266/Arduino #6256: HTTP_UPDATE_FAILD Error (-5): HTTP error: connection lost

akhtarusafzai, 2019-07-04:
Hello All, 
I am trying to update my firmware for NodeMCU through OTA. I am using the basic HTTP_UPDATE example code given in arduino. The Bin File is placed on Github and trying to download that and upload it to the board.

#### Platform

- Hardware:					[NodeMCU Amica]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT Compatible]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

/**
   httpUpdate.ino

    Created on: 27.11.2015

*/

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

#ifndef APSSID
#define APSSID "Akhtar"
#define APPSK  "akhtar1234"
#endif

ESP8266WiFiMulti WiFiMulti;

void setup() {

  USE_SERIAL.begin(115200);
  // USE_SERIAL.setDebugOutput(true);

  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  for (uint8_t t = 4; t > 0; t--) {
    USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
    USE_SERIAL.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(APSSID, APPSK);

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    WiFiClient client;

    // The line below is optional. It can be used to blink the LED on the board during flashing
    // The LED will be on during download of one buffer of data from the network. The LED will
    // be off during writing that buffer to flash
    // On a good connection the LED should flash regularly. On a bad connection the LED will be
    // on much longer than it will be off. Other pins than LED_BUILTIN may be used. The second
    // value is used to put the LED on. If the LED is on with HIGH, that value should be passed
    ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);

    t_httpUpdate_return ret = ESPhttpUpdate.update(client, "https://github.com/akhtarusafzai/office/blob/master/Blink.bin");
    // Or:
    //t_httpUpdate_return ret = ESPhttpUpdate.update(client, "server", 80, "file.bin");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        USE_SERIAL.println("HTTP_UPDATE_OK");
        break;
    }
  }
}
### Debug Messages

```
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
HTTP_UPDATE_FAILD Error (-5): HTTP error: connection lost
HTTP_UPDATE_FAILD Error (-5): HTTP error: connection lost
HTTP_UPDATE_FAILD Error (-5): HTTP error: connection lost
HTTP_UPDATE_FAILD Error (-5): HTTP error: connection lost
```

devyte, 2019-07-05:
The basic example clearly shows an http url, while github requires https, i. e. secure. Therefore that example won't work in your case. You have to use a WiFiClientSecure client, and set it up correctly before passing it as arg to the updater. 
Closing due to user error.

closed 2019-07-05 by devyte

Jhony9625, 2019-07-15:
Hi, I'm using the same example of httpUpdate and I get wrong http code the server link is this: http://qualitysystems.com.co/Blink.bin

I searched the internet and found this 2016 post https://github.com/esp8266/Arduino/issues/2022 I tried what it says but it did not work for me. I do not know what else to do to solve this problem.

JiriBilek, 2019-07-15:
@Jhony9625: I assume the error is reported in file `ESP8266httpUpdate.cpp`, line 419 or nearby. 
If so, edit the file, add debugging printouts to find out what's going on. You can also turn on debugging messages in Arduino IDE (debug level HTTP_UPDATE), maybe it helps. Otherwise you have to understand the httpUpdate code and find out the problem yourself.

akhtarusafzai, 2019-07-17:
@Jhony9625 I have resolved the issue. I was passing a https address to http_update method. it won't work without ssl certificates fingerprint. Try to give it local server address running on your pc. like http://192.168.0.12/Blink.bin

Jhony9625, 2019-07-17:
I was able to solve it, I leave the link where they gave me the solution. Thank you all