esp8266/Arduino #6259: _yield panics adafruit gfx

tablatronix, 2019-07-05:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				NODEMCU v2
- Core Version:			staging,stable,2.0.4
- Development Env:		platformio
- Operating System:		osx

### Settings in IDE

- Module:  			nodemcuv2, standard build
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I am getting _yield panics, and cannot for the life of me figure out why

I do not have an MCVE yet, still trying to make one smaller...

[image]

https://github.com/adafruit/Adafruit-GFX-Library/issues/236

d-a-v, 2019-07-05:
@tablatronix Are you calling `Adafruit_SPITFT::writeColor()` from an ISR or a callback ?

tablatronix, 2019-07-05:
Nope, afaik there is no isr or callbacks in this code at all

```
Exception Cause: Not found

0x40206d19: yield at ??:?
0x40208d00: Adafruit_SPITFT::writeColor(unsigned short, unsigned int) at ??:?
0x40209bcb: SPIClass::beginTransaction(SPISettings) at ??:?
0x40208e09: Adafruit_SPITFT::writeFillRectPreclipped(short, short, short, short, unsigned short) at ??:?
0x402090bd: Adafruit_SPITFT::fillRect(short, short, short, short, unsigned short) at ??:?
0x402060c5: Print::write(char const*) at ??:?
0x40209f11: Adafruit_GFX::fillScreen(unsigned short) at ??:?
0x4020326f: ShowMenu() at ??:?
0x40104ee8: ets_timer_arm_new at ??:?
0x402074bc: delay at ??:?
0x402074d2: delay at ??:?
0x40203773: setup at ??:?
0x40206cd0: loop_wrapper() at core_esp8266_main.cpp:?
0x401007d5: cont_wrapper at ??:?
0x401000c0: app_entry at ??:?
0x4010f000: ?? ??:0
0x4010f15c: ?? ??:0
0x4000044c: ?? ??:0
0x40000000: ?? ??:0
0x40000010: ?? ??:0
0x40000020: ?? ??:0
0x40000030: ?? ??:0
0x40000040: ?? ??:0
```

tablatronix, 2019-07-05:
It does act strange though, as if there is, sometimes delays() do not work, and loop stops after one loop and I get hardware wdt, I am at a complete loss atm

d-a-v, 2019-07-05:
`ets_timer_arm_new()` is used by `delay()`

Any chance to see the code or having more details on when the gfx function is called ?

tablatronix, 2019-07-05:
trying to make a reproducible sample

last thing is a display.println("something");
going to condense to static code and see what happens

is there a way to test ETS_INTR_WITHINISR in user code?

or what cont->pc_ret and pc_yield are ?

d-a-v, 2019-07-05:
`cont_can_yield()` will tell

devyte, 2019-10-31:
@tablatronix at least one stability issue that could be relevant has been fixed in the meantime. Is this issue still reproducible?

tablatronix, 2019-10-31:
I can test again and close

devyte, 2019-11-09:
ping

tablatronix, 2019-11-09:
Will give it a spin soon

devyte, 2019-11-20:
reping

tablatronix, 2019-11-20:
Heh I have been trying to find the sketch I was using lol.

devyte, 2020-02-02:
Ok, closing due to lack of feedback in several months.

closed 2020-02-02 by devyte