esp8266/Arduino #6265: HTTPClient with https : LoadStoreAlignmentCause

Stunkymonkey, 2019-07-06:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12E
- Core Version:			2.5.5 via Arduino IDE
- Development Env:		Arduino IDE
- Operating System:		Arch Linux

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			? do not know what this is
- Flash Size: 	                4MB (1MB SPIFF)
- lwip Variant:                   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	        ? unable to find it
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

The function below is getting executed every 15Minutes. When the `setup()` starts it gets executed the first time. But everytime it crashes.
If i uncomment this function eveything works fine.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

String message = "{\"temp\":\"" + String(temperature) + "\", \"humidity\":\"" + String(humidity) + "\"}";
std::unique_ptr<BearSSL::WiFiClientSecure> client(new BearSSL::WiFiClientSecure);
// fingerprint verification is skipped
client->setInsecure();
http.begin(*client, "https://stunkym.uber.space/esp8266/sensor.php");
delay(500);
http.addHeader("Content-Type", "application/json");
http.addHeader("token", "secret");
print("got here");
int httpCode = http.POST(message);
print("finally works");
print(httpCode);
if (httpCode > 0) {
  if (DEBUG && httpCode == HTTP_CODE_OK) {
    http.writeToStream(&Serial);
    print("");
  }
} else {
  print("unknown http error" + String(http.errorToString(httpCode)));
}
http.end();

```

### Debug Messages

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40223328: HTTPClient::connected() at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 460
EXCVADDR: 0x076007d3

Decoding stack results
0x4020b989: HTTPClient::connect() at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1143
0x402199d0: String::changeBuffer(unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.cpp line 179
0x4020c171: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 655
0x40218468: HardwareSerial::write(unsigned char const*, unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 158
0x40219a52: String::reserve(unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.cpp line 148
0x4020c378: HTTPClient::POST(unsigned char*, unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 586
0x4020c3a4: HTTPClient::POST(String) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 591
0x4020324d: sendSensorData() at /home/felix/code/arduino/esp8266-smartclock/ESP_clock/Sensor.ino line 74
0x40201a28: getTemperature() at /home/felix/code/arduino/esp8266-smartclock/ESP_clock/Sensor.ino line 23
0x40203e59: setup() at /home/felix/code/arduino/esp8266-smartclock/ESP_clock/ESP_clock.ino line 188
0x4021af00: loop_wrapper() at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 122

```

sometimes this appears:
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40223328: HTTPClient::connected() at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 460
EXCVADDR: 0x00000034

Decoding stack results
0x4020b989: HTTPClient::connect() at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1143
0x402199d0: String::changeBuffer(unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.cpp line 179
0x4020c171: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 655
0x40218468: HardwareSerial::write(unsigned char const*, unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 158
0x40219a52: String::reserve(unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.cpp line 148
0x4020c378: HTTPClient::POST(unsigned char*, unsigned int) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 586
0x4020c3a4: HTTPClient::POST(String) at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 591
0x4020324d: sendSensorData() at /home/felix/code/arduino/esp8266-smartclock/ESP_clock/Sensor.ino line 74
0x40201a28: getTemperature() at /home/felix/code/arduino/esp8266-smartclock/ESP_clock/Sensor.ino line 23
0x40203e59: setup() at /home/felix/code/arduino/esp8266-smartclock/ESP_clock/ESP_clock.ino line 188
0x4021af00: loop_wrapper() at /home/felix/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 122
```

earlephilhower, 2019-07-06:
We need a full, compilable, MCVE sketch.  Your snippet isn't enough to build anything.  Please winnow it down to a setup() with all relevant variable definitions, etc. that reproduced the failure.

Given the failure lines, I think you've got some memory corruption in your code or hitting OOM as the line in question is simply dereferncing the passed-in client (i.e. the client variable itself is NULL or contains garbage).

Stunkymonkey, 2019-07-07:
this is the MCVE
```
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <ESP8266HTTPClient.h>

HTTPClient http;

void setup() {
  Serial.begin(115200);
  String message = "{\"temp\":\"23.45\", \"humidity\":\"44.56\"}";
  Serial.println(message);
  std::unique_ptr<BearSSL::WiFiClientSecure> client(new BearSSL::WiFiClientSecure);
  // fingerprint verification is skipped
  client->setInsecure();
  http.begin(*client, "https://stunkym.uber.space/esp8266/sensor.php");
  delay(500);
  http.addHeader("Content-Type", "application/json");
  http.addHeader("token", "pzEMiC5cPArgLaqg7Suf");
  Serial.println("got here");
  int httpCode = http.POST(message);
  Serial.println("posted");
  Serial.println(httpCode);
  if (httpCode > 0) {
    if (httpCode == HTTP_CODE_OK) {
      http.writeToStream(&Serial);
      Serial.println("");
    }
  } else {
    Serial.println("unknown http error" + String(http.errorToString(httpCode)));
  }
  http.end();
}

void loop() {
}
```
and when flashing it onto the esp it works fine. so yes oom could be the problem.
but how to fix it?

earlephilhower, 2019-07-07:
Hard to say anything other than "use less memory."  I would preallocate a single WiFiClientSecure globally to ensure that the large memory chunk it needs gets taken early on, because even with lots of free RAM you can end up with fragmentation where there is no single spot big enough for the needed buffers.  Good luck.

closed 2019-07-07 by earlephilhower

Abedi98, 2020-11-25:
> this is the MCVE
>
> ```
> #include <WiFiClient.h>
> #include <WiFiClientSecure.h>
> #include <ESP8266HTTPClient.h>
>
> HTTPClient http;
>
> void setup() {
>   Serial.begin(115200);
>   String message = [...]

I have exactly the same problem. How did you solve this problem?

Stunkymonkey, 2020-11-25:
RAM is the problem. So the the soultions are:
- use another board
- use HTTP and maybe bridge it
- reduce functionality to have more RAM available

duncansu, 2022-04-03:
hey guys,do you solve your problem?And howto do it?thank you