esp8266/Arduino #6271: getHeapFragmentation() returns 98 when normally returns 1

adrionics, 2019-07-08:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				WROOM-02
- Core Version:			2.5.2
- Development Env:		Arduino
- Operating System:		MacOS

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			OTA
- Flash Size: 		2MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		nodemcu
- Flash Frequency:	80Mhz
- CPU Frequency:		160MHz
- Upload Using:		OTA
- Upload Speed:   

### Problem Description

This is a weird one so I do not have an MCVE.

What happens is this: The device works for quite some time, then one day stopped and on examination, was crashing during the setup() routine.

From very early in the startup sequence, I notice the getHeapFragmentation() function returns 98 where usually it returns 1.

In order to do some kind of troubleshooting, I got the customer to read the flash content and I used that .bin file to flash a device of mine. The problem was not there. Flashing multiple devices and multiple times does not get the error to show itself.

The numbers in bold are the result of getHeapFragmentation(), to the left of that is getFreeHeap() and to the right getMaxFreeBlockSize():

This is an example of a working device. 

```
FlashChipRealSize: 2097152
 0h00:02.706 41584/**1**/41448/3136/3/0/0 1970/01/01T08:00:02z: BME280 (0x76)
 0h00:03.730 41544/**1**/41448/3056/3/0/0 1970/01/01T08:00:03z: Set hostname... 
 0h00:03.732 41536/**1**/41448/3056/3/0/0 1970/01/01T08:00:03z: HostName-e8db
 0h00:03.735 40936/**1**/40904/3056/3/0/0 1970/01/01T08:00:03z: 84:f3:eb:98:e8:db
 0h00:03.741 40416/**1**/40384/3056/3/0/0 1970/01/01T08:00:03z: SSID: THE_SSID
.
.
.

```
And the non working one:

```
FlashChipRealSize: 2097152
 0h00:02.556 44112/**98**/42872/3136/3/0/0 1970/01/01T08:00:02z: BME280 (0x76)
 0h00:03.578 44072/**98**/42872/3056/3/0/0 1970/01/01T08:00:03z: Set hostname... 
 0h00:03.579 44064/**98**/42872/3056/3/0/0 1970/01/01T08:00:03z: HostName-9234
 0h00:03.582 43464/**99**/42872/3056/3/0/0 1970/01/01T08:00:03z: 84:f3:eb:6e:92:34
Abort called

Stack trace outputted here
.
.
.

```

I am not yet sure if just flashing the .bin we read off the faulty device will fix it, we will try that when the customer gets a chance, but I suspect it will need to be wiped first.

I was wondering if anyone has any idea why this may be happening. I have never seen such a high fragmentation number in the startup of this code. Also, the getMaxFreeBlockSize shows no sign of a problem, neither does getFreeHeap. If it was flash corruption, it would be copied to the test device? 

Any insights would be appreciated.

devyte, 2019-07-08:
No idea. If the device was working, then stopped working for no apparent reason, and the exact same flash dump flashed on a different device works, my suspicion would be some kind of hw failure.
Other than that, some kind of mem corruption during startup could be happening due to some condition that was never hit before. Or it could be some mem alloc/free loop, e. g. construction of globals, that causes severe fragmentation. 
Without a MCVE that reproduces the issue there's nothing we can do. I'm closing for now, but if you come up with additional info that could open a path for investigation, @ me and I'll reopen, or just open a new issue and fill out the required info.

closed 2019-07-08 by devyte

adrionics, 2019-07-08:
Thanks, I thought I would just mention it in case someone comes across a similar type of behavior. It is quite hard to deal with something like this that only happens randomly and seldom. I think it happened before but I did not notice the fragmentation number (it may have been present) before I erased and flashed that one.

adrionics, 2019-07-08:
An update. Using esptool.py to download the flash and then upload it again to the same device fixed the problem. How could that be? Is there some other memory that is cleared on a flash upload? Perhaps the RTC area?