esp8266/Arduino #6275: Linker error `.text1' will not fit in region `iram1_0_seg'

Jason2866, 2019-07-09:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8266]
- Core Version:			[latest git hash]
- Development Env:		[VSC with Platformio|]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[|Wemos D1 mini]
- Flash Mode:			[DOUT]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [LWIP2_HIGHER_BANDWIDTH_LOW_FLASH]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Linker error with version master when compiling https://github.com/arendst/Sonoff-Tasmota

```
Indexing .pioenvs/sonoff/libFrameworkArduino.a
Linking .pioenvs/sonoff/firmware.elf
/home/gitpod/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/sonoff/firmware.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sonoff/firmware.elf] Error 1
============================================= [ERROR] Took 124.28 seconds =============================================
```

TO REPRODUCE
Steps to reproduce the behavior:

Compiling with options enabled in platformio.ini (file from project Tasmota)

```
env_default = sonoff

platform                  = ${core_stage.platform}
build_flags               = ${core_stage.build_flags}

```
EXPECTED BEHAVIOUR
Successfull linking

earlephilhower, 2019-07-09:
You're out of IRAM space.  Please post/examine the generated .MAP file and see what's filling up the very limited  space.

Jason2866, 2019-07-09:
Was is the reason for decreased iram space? There is more iram space in core 2.5.2.

**** Core 2.5.2

```
Linking .pioenvs\sonoff\firmware.elf
Building .pioenvs\sonoff\firmware.bin
Retrieving maximum program size .pioenvs\sonoff\firmware.elf
Checking size .pioenvs\sonoff\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  55.0% (used 45040 bytes from 81920 bytes)
PROGRAM: [=====     ]  48.9% (used 500348 bytes from 1023984 bytes)
Creating BIN file ".pioenvs\sonoff\firmware.bin" using ".pioenvs\sonoff\firmware.elf"

readelf -l firmware.elf:

Elf file type is EXEC (Executable file)
Entry point 0x4010065c
There are 5 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x0000d4 0x00000000 0x00000000 0x00000 0x00000     0x1000
  LOAD           0x001000 0x3ffe8000 0x3ffe8000 0x01424 0x01424 RW  0x1000
  LOAD           0x002428 0x3ffe9428 0x3ffe9428 0x00000 0x09bd8 RW  0x1000
  LOAD           0x003000 0x40100000 0x40100000 0x07838 **0x07838** R E 0x1000
  LOAD           0x00b010 0x40201010 0x40201010 0x7162c 0x7162c R E 0x1000

 Section to Segment mapping:
  Segment Sections...
   00
   01     .data .noinit .rodata
   02     .bss
   03     .text .text1
   04     .irom0.text
```

**** Core Stage (20190709)
```
Linking .pioenvs\sonoff\firmware.elf
Building .pioenvs\sonoff\firmware.bin
Retrieving maximum program size .pioenvs\sonoff\firmware.elf
Checking size .pioenvs\sonoff\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  55.1% (used 45172 bytes from 81920 bytes)
PROGRAM: [=====     ]  48.5% (used 496768 bytes from 1023984 bytes)
Creating BIN file ".pioenvs\sonoff\firmware.bin" using ".pioenvs\sonoff\firmware.elf"

readelf -l firmware.elf:

Elf file type is EXEC (Executable file)
Entry point 0x401000b8
There are 5 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x0000d4 0x00000000 0x00000000 0x00000 0x00000     0x1000
  LOAD           0x001000 0x3ffe8000 0x3ffe8000 0x01420 0x01420 RW  0x1000
  LOAD           0x002420 0x3ffe9420 0x3ffe9420 0x00000 0x09c60 RW  0x1000
  LOAD           0x003000 0x40100000 0x40100000 0x07c70 **0x07c70** R E 0x1000
  LOAD           0x00b010 0x40201010 0x40201010 0x703fc 0x703fc R E 0x1000

 Section to Segment mapping:
  Segment Sections...
   00
   01     .data .noinit .rodata
   02     .bss
   03     .text .text1
   04     .irom0.text
```
From the above info section 03 contains the iram usage. The default linker allows for max 0x08000 bytes iram. As can be seen Stage uses a lot more iram (0x7c70) than 2.5.2 (0x7838).

earlephilhower, 2019-07-09:
You need to look at the actual MAP file to get sizes and see what's taking up space, the ELF isn't going to be generated when the IRAM section overflows but the MAP is always generated even when there's an overflow .  I know the TwoWire stuff seems to be quite large, but I don't use it so can't comment.

Jason2866, 2019-07-09:
Thanks for looking into. We knew what functions iram space needs in Tasmota.
The question is, is there a chance to get back more iram space?
Or will this reduced iram space be standard for upcoming core 2.6.0?

earlephilhower, 2019-07-09:
If you can get a list of things that have grown (say, from the MAP files of projects built under older revs and git head) that you're using, we can look at it...or you can, that's the magic of open source :)

Right now, unfortunately, there's no info in this bug that we can use.

Jason2866, 2019-07-09:
@earlephilhower  I have uploaded (in my post before). Is anything else needed?
THX!

earlephilhower, 2019-07-09:
Actually, since it looks like both are compiling just fine now, can you attach the ELF?  Since the ELF is being generated, it's 100x easier to just dump the symbols from IRAM and to reverse-work the .map file.  

But, that said, I think this is due to a bugfix in head vs. 2.5.2.  2.5.2 ended up with the memory manager stored in *flash* not iram.  Git head moved it back where it belongs, which is where it needs to be, and that's another 0x400 bytes if my eyeballing is correct.

And that matches your difference, so I think we're good here.  It's working as in tended...

````
 *fill*         0x0000000040101963        0x1
 .text.umm_disconnect_from_free_list
                0x0000000040101964       0x62 .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
 *fill*         0x00000000401019c6        0x2
 .text.umm_assimilate_up
                0x00000000401019c8       0x82 .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
                                         0x8a (size before relaxing)
 *fill*         0x0000000040101a4a        0x2
 .text.umm_assimilate_down
                0x0000000040101a4c       0x55 .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
                                         0x59 (size before relaxing)
 *fill*         0x0000000040101aa1        0x3
 .text._umm_free
                0x0000000040101aa4       0xcb .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
                                         0xe7 (size before relaxing)
 *fill*         0x0000000040101b6f        0x1
 .text.umm_make_new_block$constprop$1
                0x0000000040101b70       0x5a .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
                                         0x5e (size before relaxing)
 *fill*         0x0000000040101bca        0x2
 .text._umm_malloc
                0x0000000040101bcc      0x143 .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
                                        0x163 (size before relaxing)
 *fill*         0x0000000040101d0f        0x1
 .text.malloc   0x0000000040101d10       0x31 .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
                                         0x39 (size before relaxing)
                0x0000000040101d18                malloc
 *fill*         0x0000000040101d41        0x3
 .text.calloc   0x0000000040101d44       0x47 .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
                                         0x53 (size before relaxing)
                0x0000000040101d48                calloc
 *fill*         0x0000000040101d8b        0x1
 .text.realloc  0x0000000040101d8c      0x187 .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
                                        0x1db (size before relaxing)
                0x0000000040101d90                realloc
 *fill*         0x0000000040101f13        0x1
 .text.free     0x0000000040101f14        0xf .pioenvs\sonoff\libFrameworkArduino.a(umm_malloc.cpp.o)
                                         0x17 (size before relaxing)
                0x0000000040101f14                free
 *fill*         0x0000000040101f23        0x1
````

Jason2866, 2019-07-10:
LD.exe fails and is deleting *.elf file. How can i tell Platformio no to do so?

earlephilhower, 2019-07-10:
That's OK, but the MAP you uploaded showed plenty of space:
````
  0x0000000040107d9c                _etext = .
````
So IRAM had 0x7d9c (of 0x7fff) space used, which is legal.  

https://github.com/esp8266/Arduino/blob/1b3ac4f5e916108bf231e4e0082dafb888816572/tools/sdk/ld/eagle.flash.1m.ld#L13

So, an ELF should have been generated.

In any case, the difference is explained.  A bug in 2.5.2 had malloc in flash, not IRAM, so it was potentially unsafe to call from IRQs (and, in fact, it was always in iram until the 2.5.2 release when the change from umm.c to umm.cpp caused it not to go there).

Tasmota is stuffing a ton of stuff in IRAM.  It would be worth the time for the project to see exactly what's really needed, because the memory manager is not optional. :)

Jason2866, 2019-07-10:
Stripped 2 features in Tasmota, does compile successfull again.
More iram space would be good ;-)
Thx for your help!

closed 2019-07-10 by Jason2866

Brackbouille, 2019-10-12:
Hello 
I tried to compile Tasmota 6.6.0 with VSC, atom  and i have this issue, it mean that that come from platformio 4. Could you help me to solve that problem, i can't find a solution

Regards

Brackbouille, 2019-10-12:
Ok found myself change platform and build from 2.3.0 to 2.4.2 in platformio.ini