esp8266/Arduino #6276: sdfat calls missing

gemu2015, 2019-07-09:
Basic Infos

  This issue complies with the issue POLICY doc.
  I have read the documentation at readthedocs and the issue is not addressed there.
  I have tested that the issue is present in current master branch (aka latest git).
  I have searched the issue tracker for a similar issue.
  If there is a stack dump, I have decoded it.
  I have filled out all fields below.

Platform
Hardware:	[ESP8266]
Core Version:	[latest git hash]
Development Env:	[VSC with Platformio|]
Operating System:	[OSX]

Settings in IDE
Module: [|Wemos D1 mini]
Flash Mode:	[DOUT]
Flash Size: [4MB/1MB]
lwip Variant: [LWIP2_HIGHER_BANDWIDTH_LOW_FLASH]
Reset Method:	[ck]
Flash Frequency:	[40Mhz]
CPU Frequency:	[80Mhz]
Upload Using:	[SERIAL]
Upload Speed: [115200]

Problem Description

after 2.50 the following calls are missing

the calls FAT_DATE and FAT_TIME are missing
also the callback SdFile::dateTimeCallback(dateTime);
is no longer supported thus the file creation time can not be set.

the defines USE_STANDARD_SPI_LIBRARY and USE_LONG_FILE_NAMES
are hardcoded now.
should be ifdefd to be overwritten by compiler without changing sources

especially USE_STANDARD_SPI_LIBRARY is important to use several spi devices on the bus.

earlephilhower, 2019-07-09:
The defines will be easy to bracket.  I didn't think anyone wanted 8.3 in 2019, even my 2560 based 3d printer handles LFNs, but I suppose if you're working with other, older HW it might be needed.

The time stuff needs some thought as to how to fit it into the current architecture.  Might just punt and add a call to get the underlying (new) SdFat object for a file, which would give you access to the dateTimeCB.

earlephilhower, 2019-07-09:
Just did some checking, @gemu2015.  The two macros you mention are set by the main library author, not myself, and look to have been there for quite some time.  I imagine that's for a good reason, as he's very active on the library.

I will add options to set them from the compiler (change your platform.locals.txt) but can't guarantee they'll do anything useful or even work.

gemu2015, 2019-07-10:
ok its meant for an extension to tasmota (theo arends)
Theo requires all prs to run with 2.3, 2.42 and 2.52
if developed the extension (scripter) when 2.52 wasn't available
now i get problems to stay compatible.
especially  USE_STANDARD_SPI_LIBRARY=0 may be a problem because i use several spi devices together (eg. lcd displays with sd cards on board)
so a custom spi would break the lcd interface.
the file creation date and time are also nice to have (callback)
the long filenames would not be a problem although i like to use as little ram as possible
(on tasmota ram is very limited)

earlephilhower, 2019-07-10:
Wow, 2.3 is over 3 years old.  Good luck to you, and to anyone else trying to keep it up and running!

earlephilhower, 2019-07-10:
Also, FWIW, the free heap of a blank WiFi-enabled sketch in 2.5.2 is ~54KB now, so RAM is less of a challenge.  I think it was around

earlephilhower, 2019-07-20:
Give #6315 a try, @gemu2015 .  It's got timestamps for all filesystems now and brackets around the sdfat config options.

gemu2015, 2019-07-23:
@earlephilhower 

thank you very much!
 i am currently short on time but will test in the near future

earlephilhower, 2019-11-18:
Closing as SD.h and LittleFS now have timestamp support.

closed 2019-11-18 by earlephilhower