esp8266/Arduino #6277: Random "SOFT WDT reset" crashes

arushgupta2007, 2019-07-09:
#### Platform

- Hardware:					Nodemcu Amica ESP8266
- Core Version:			2.5.0
- Development Env:		Arduino IDE
- Operating System:		Windows

#### Settings in IDE

- Module:  			Nodemcu 1.0 (ESP-12E Module)
- Flash Size: 		4MB (no SPIFFS )
- lwip Variant:   v2 Lower Memory
- CPU Frequency:		80Mhz
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I am using my Nodemcu to control the status of my bulb. For which I am using 2 triacs, ACS712 Current sensor and zero cross detection circuit. I am also using Firebase as my cloud service, so that I can see and change the status of my bulb using my phone. 
[link to firebase library I am using](https://github.com/mobizt/Firebase-ESP8266/)

I am experiencing random crashes after some time. It reads Soft WDT Reset 

### Sketch

```cpp

[code]
#include <ACS712.h>
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

#define FIREBASE_HOST "..."
#define FIREBASE_AUTH "..."
#define WIFI_SSID "..."
#define WIFI_PASSWORD "..."

ACS712 sensor(ACS712_05B, A0);
int delayto = 2000;
bool has_changed = false;
String path = "";
unsigned long previousMillis = 0;
const long interval = 1000;
int till = 0;
//int debug = 0;

FirebaseData firebaseData;
FirebaseData firebaseDatadiffrent_obj;
DynamicJsonBuffer json_buffer;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);

  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  pinMode(D2, INPUT);
  pinMode(D7, OUTPUT);
  pinMode(D8, OUTPUT);

[... 166 lines cut ...]
0x402091f5: FirebaseESP8266::sendFirebaseRequest(FirebaseData&, char const*, unsigned char, unsigned char, char const*, char const*, unsigned int) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 3545
0x4020b1d6: FirebaseESP8266::firebaseConnect(FirebaseData&, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 1359
0x4022900b: basic_read at ssl/tls1.c line 1374
0x401001dd: __digitalWrite at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring_digital.c line 82
0x40207382: zcd() at C:\Users\...\Documents\Arduino\esp_home_control/esp_home_control.ino line 143
0x40100349: interrupt_handler at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring_digital.c line 166
0x40100338: interrupt_handler at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring_digital.c line 162
0x4020d7c4: ESP8266WiFiSTAClass::status() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 563
0x4020b878: FirebaseESP8266::firebaseConnect(FirebaseData&, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 1718
0x4020ce18: FirebaseESP8266::sendRequest(FirebaseData&, unsigned char, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 1828
0x4020d02a: FirebaseESP8266::buildRequest(FirebaseData&, unsigned char, unsigned char, std::string const&, char const*, bool, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 199
0x402118e1: Print::write(char const*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/Print.h line 60
0x4021cef0: operator new(unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
0x4020d1ab: FirebaseESP8266::setBool(FirebaseData&, std::string const&, bool, bool, std::string const&, std::string const&) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 728
0x4021d7f2: std::basic_string  , std::allocator  >::basic_string(char const*, std::allocator  const&) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 218
0x4010014c: millis at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring.c line 183
0x4020d21b: FirebaseESP8266::setBool(FirebaseData&, String const&, bool) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266.cpp line 688
0x40211eb7: String::reserve(unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 146
0x40211f00: String::copy(char const*, unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 182
0x40211fec: String::operator=(String const&) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 220
0x40207795: loop() at C:\Users\...\Documents\Arduino\esp_home_control/esp_home_control.ino line 85
0x40212400: loop_wrapper() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 125

```
Can anyone please help me out?

devyte, 2019-07-09:
A whole lot of issues have been fixed since 2.5.0. I suggest trying 2.5.2 and/or latest git.
I don't see an issue at this time, so I'm closing. If you add info that points to a core issue, @ me and I'll reopen, or just open a new issue and fill in the required info.

closed 2019-07-09 by devyte

arushgupta2007, 2019-07-10:
@devyte  I am still having the problem using 2.5.2. 

###Stack Dump
```
0x4020e713: loop_task(ETSEvent*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 133
0x4010020c: delayMicroseconds(unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 208
0x40100234: __digitalWrite(uint8_t, uint8_t) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 85
0x401000de: zcd() at C:\Users\...\Documents\Arduino\esp_home_control/esp_home_control.ino line 144
0x401003b1: interrupt_handler(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 169
0x401003a0: interrupt_handler(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 165
0x40100234: __digitalWrite(uint8_t, uint8_t) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 85
0x401002e0: interrupt_handler(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 133
0x40231fcf: SHA1ProcessMessageBlock at crypto/sha1.c line 169
0x402320b5: SHA1_Update at crypto/sha1.c line 81
0x402300e2: bi_subtract at crypto/bigint.c line 324
0x40210b2e: _umm_free(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1350
0x4022fd2a: bi_free at crypto/bigint.c line 219
0x4022fd2a: bi_free at crypto/bigint.c line 219
0x401003a3: interrupt_handler(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 165
0x401003a0: interrupt_handler(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 165
0x401002e0: interrupt_handler(void*) at C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 133
0x40232a35: SHA256_Process at crypto/sha256.c line 97
0x40234cc0: SHA256_Update at crypto/sha256.c line 231
0x401003b1: interrupt_handler(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 169
0x401002e0: interrupt_handler(void*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 133
0x4022f32e: AES_cbc_encrypt at crypto/aes.c line 370
0x4022f29e: AES_cbc_encrypt at crypto/aes.c line 283
0x4020e758: esp_yield() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 91
0x4020ece3: delay(unsigned long) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 54
0x4020812d: ClientContext::_write_from_source(DataSource*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/ClientContext.h line 467
0x40208930: axTLS::__ax_port_write(int, uint8_t*, size_t) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/ClientContext.h line 361
0x4022bae0: send_packet at ssl/tls1.c line 1133
0x402088c8: axTLS::__ax_port_read(int, uint8_t*, size_t) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClientSecureAxTLS.cpp line 472
0x4022c45b: basic_read at ssl/tls1.c line 1374
0x4022bb6c: ssl_write at ssl/tls1.c line 355
0x40208f43: axTLS::WiFiClientSecure::write(unsigned char const*, unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/SSLContext.h line 382
0x4022c930: ssl_read at ssl/tls1.c line 314
0x4020da69: Print::write(char const*) at C:\Users\....\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Print.h line 60
0x40208c90: axTLS::WiFiClientSecure::available() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/SSLContext.h line 273
0x4020dac4: Print::print(char const*) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\Print.cpp line 122
0x40207396: FirebaseHTTPClient::http_sendHeader(char const*) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266HTTPClient.cpp line 58
0x402073c6: FirebaseHTTPClient::http_connect() at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266HTTPClient.cpp line 76
0x40207421: FirebaseHTTPClient::http_sendRequest(char const*, char const*) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266HTTPClient.cpp line 65
0x40218b49: __itoa at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/itoa.c line 59
0x4020354d: FirebaseESP8266::sendFirebaseRequest(FirebaseData&, char const*, unsigned char, unsigned char, char const*, char const*, unsigned int) at C:\Users\...\Documents\Arduino\libraries\Firebase-ESP8266-master\src\FirebaseESP8266 [... cut ...]
0x4020e1c0: String::copy(char const*, unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 211
0x4020e35c: String::operator=(String const&) at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 259
0x40201429: loop() at C:\Users\...\Documents\Arduino\esp_home_control/esp_home_control.ino line 85
0x4020e808: loop_wrapper() at C:\Users\...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125

```

###Code
Same as above

devyte, 2019-07-10:
Several comments from that last dump:
- I get the impression you're facing oom. Check free heap and max free block size at different places in your code
- We can't debug firebase. If the crash is in there, and I think it is, you'll have to discuss with the lib author. Try reducing your code to not use firebase and see if the issue is still there. 
- the dump shows use of axtls. That was deprecated some time ago, then the default was changed to bssl. In the (near) future it will be dropped altogether, so firebase should update to ise bssl sooner rather than later.
BTW, axtls doesn't behave well when faced with oom. 
- why do you have a global dynamic json buffer? I haven't been keeping up with that lib's most recent development, but at least thw last version I used did not allow reuse of a buffer, because mem would not be freed until destruction. I seem to remember that a reset would be added, if that happened you need to use it. I mention this, because you may have a mem leak there. 
- Try with 160MHz cpu. Use of ssl is cpu intensive.

mobizt, 2019-07-10:
Nested interrupt causes heap corruption by stack growth due to the poster uses delay in his ISR.

devyte, 2019-07-10:
Ah jeez I completely missed that, thanks @mobizt.
@arushgupta2007 you can't delay() in an ISR.

arushgupta2007, 2019-07-12:
🎉 It worked!! 🎉

arushgupta2007, 2019-07-12:
Thanks @mobizt  and @devyte  for your time and support!

doublejosh, 2020-02-24:
Apparently also... `delayMicroseconds()` *does _not_* satisfy the watchdog, but `delay()` does!