esp8266/Arduino #6281: Compiler warning for cast in Ticker code

Niek, 2019-07-10:
Sample code:
```c++
#include <Ticker.h>

void testFn(unsigned char i) {}

void setup() {
  Ticker t;
  unsigned char i = 1;
  t.once_ms(100, testFn, i);
}
```
I get this warning when compiling the latest git version:
```bash
libraries/Ticker/Ticker.h: In instantiation of 'void Ticker::once_ms(uint32_t, void (*)(TArg), TArg) [with TArg = unsigned char; uint32_t = unsigned int]':

libraries/Ticker/Ticker.h:110:94: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   _attach_ms(milliseconds, false, reinterpret_cast<callback_with_arg_t>(callback), (void*)arg);
```
It originates from this cast: https://github.com/esp8266/Arduino/blob/48ace77b4857073ee65dfec9ed3ddc9fd5e8d099/libraries/Ticker/Ticker.h#L110
It seems that it's caused by the ``(void*)`` cast in the last parameter.

earlephilhower, 2019-07-10:
You have a problem in your code.

The function signature for a callback is `void (cb)(void *data)`, not `void (cb)(unsigned char)`.  Change the callback to the proper signature and then, inside the function, convert from `void*` to the `unsigned char` with something like below.
````
void testFn(void *cbData) {
    unsigned char i = static_cast<unsigned char>(cbData);
   ....
}

Niek, 2019-07-10:
Are you saying the callback function can't take other types as argument, as explained in the sample? https://github.com/esp8266/Arduino/blob/master/libraries/Ticker/examples/TickerParameter/TickerParameter.ino
Or is it the unsigned part that makes it problematic?
By the way the code works fine in core 2.5.2.

earlephilhower, 2019-07-10:
Actually, I'm mistaken and you're right.  The template is supposed to handle that from how I read it:
````
        template<typename TArg>
        void once_ms(uint32_t milliseconds, void (*callback)(TArg), TArg arg)
````

The (void*) in the template body looks to be incorrect and should be able to do promotion to a pointer silently, w/o error.  We need a fix and a test case that hits this.

Niek, 2019-07-11:
Thanks! I can confirm it's fixed in https://github.com/esp8266/Arduino/commit/8c3760161447c345b054b8af386b412f7da0268d - closing this issue

closed 2019-07-11 by Niek