esp8266/Arduino #6283: panic() called from ISR causes Hardware WDT

mhightower83, 2019-07-11:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in the current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					**ESP-12**
- Core Version:			[5a47cab77d396afeb842c9161327f9f4dca7c833]
- Development Env:		**Arduino IDE**
- Operating System:		**Ubuntu**

### Settings in IDE

- Module:  			**Adafruit HUZZAH**
- Flash Mode:			**qio**
- Flash Size: 		**4MB**
- lwip Variant:   **v2 Lower Memory**
- Reset Method:		**nodemcu**
- Flash Frequency:	**40Mhz**
- CPU Frequency:		**80Mhz**
- Upload Using:		**SERIAL**
- Upload Speed:   **115200** (serial upload only)

### Problem Description

I have found, that there are some actions that an ISR can take, that can unexpectedly result in a "Hardware WDT". Calls made from an ISR to panic(), abort(), assert(), or the __unhandled_exception() will result in: no message, no stack trace, just: 
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
```

Obviously, it would be best to avoid these in our ISRs. However, it might not be obvious we are doing these calls from an ISR. Take the case of running a build with just "Debug: Serial" selected in the IDE.
This will enable the `UMM_POISON`, build flag. If an ISR steps on the poison, then later does a `free()`, a message is printed (with non-IRAM based `printf()` function) then an attempt is made within `umm_malloc.cpp` to call `panic()`. When I tested this the umm poison message printed; however, the `panic()` details never appeared. Alter a long period, the "Hardware WDT" event appeared.

What I think I am seeing is, calls from an ISR that reach a panic(), abort(), assert(), or the __unhandled_exception() will eventually arrive at  `raise_exception()` in module `core_postmortem.cpp`. Then onto the inline assembly for "syscall". After a long period of silence, I see the "Hardware WDT" event. My test method at that time was less than ideal because I was using a printf function that was not in IRAM. However, it did print all the way up to the "syscall", before the WDT appeared and nothing at __wrap_system_restart_local(). 

It is interesting to find that there are causes for the "Hardware WDT", that are not caused by being stuck in a loop. 

In Arduino Core 2.5.0 and up, a failed "new" operator without `(std::nothrow)` in an ISR will call `abort()` resulting in WDT. I am concerned that there might be a lot more cases out there. 

This one is unconfirmed however looks like a good one to verify:

https://github.com/esp8266/Arduino/blob/7c6701512f44d27b998bd59119631f645dc37dba/cores/esp8266/core_esp8266_main.cpp#L104-L115

https://github.com/esp8266/Arduino/blob/7c6701512f44d27b998bd59119631f645dc37dba/cores/esp8266/cont_util.cpp#L65-L68

It looks like a call from an ISR, that reaches yield(), would call panic() and the message would never be seen.

I could say more but I'll wait and see what you say, just in case I made a wrong turn in my assessment of this issue.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <interrupts.h>
#include <assert.h>

#define CC4US_DELAY(us)   ((us) * clockCyclesPerMicrosecond())

void ICACHE_RAM_ATTR timer0_handler(void) {
  timer0_detachInterrupt();
  // From an ISR context these calls will fail with a hardware WDT
  // No event related information or stack trace is printed.
  panic();
  // abort();
  // assert(1==0);
}

void setup(void) {
  Serial.begin(115200);
  delay(20);
  WiFi.mode(WIFI_OFF);
  Serial.printf("\n\nUp and running.\n");
  {
    AutoInterruptLock(15);
    timer0_isr_init();
    timer0_attachInterrupt(&timer0_handler);
    timer0_write(ESP.getCycleCount() + CC4US_DELAY(150));
  }
}

void loop() {
}

```

### Debug Messages

```
Up and running.

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v5a47cab7
~ld
```

d-a-v, 2019-07-11:
After a quick look it looks like `asm("syscall")` line does nothing and WDT is triggered by the following one supposed to be not reached.

https://github.com/esp8266/Arduino/blob/8c3760161447c345b054b8af386b412f7da0268d/cores/esp8266/core_esp8266_postmortem.cpp#L224-L226

d-a-v, 2019-07-11:
Replacing  `__asm__ __volatile__ ("syscall");` by `*((int*)0)=0;` works better (an illegal address exception is displayed before the panic message though).

Never dug into this code before.
I am now trying to understand where the exception vector (called by `syscall`) is set up.

mhightower83, 2019-07-11:
FWIW: This is logic flow that I think I see  at `raise_exception()`:

For ps.intlevel = 0 in non ISR context:
   * syscall - has not visable effect - call and returns.
   * while(1) - sit and wait for soft wdt
   * hit soft wdt - system passes control onto `__wrap_system_restart_local()`
   * postmortem prints everyone is happy

For ps.intlevel != 0 and from a hardware ISR context:
   * syscall - has not visable effect - call and returns.
   * while(1) - sit and wait for soft wdt
   * hardware WDT stikes, `__wrap_system_restart_local()` is never called
   * postmortem never prints everyone is sad

Assumption: Between ps.intlevel !=0 and hardware interrupt priority soft WDT
cannot be serviced. Yea, I tried forcing ps.intlevel to 0 to see.

d-a-v, 2019-07-12:
@mhightower83 #6172 is a first fix for this issue.
At least it solves displayed message issues on user exceptions from ISR.
It works on the important hardware exceptions cases with gdb (cont/user or sys contexts)
There may be room for improvement, at least on gdb side.

mhightower83, 2019-07-12:
@d-a-v Thanks! I think this is going to help out a lot of people.
I think this also applies to the non-ISR case, where a lock (exp. `rsil(15)`) is active and a user exception happens.

PR #6288 (2019-07-15): fix raise_exception()
This is a proposal 

~It seems this has no *visible* effect:~
```
__asm__ __volatile__ ("syscall"); // triggers gdb or does nothing
```

On `panic`/`abort`/`assert` called from user context, WDT happens due to the following `while(1);` and finally produces an exception with the right message (after the WDT reason and delay).
On ISR context, only the WDT reason is visible.

This commit permits to avoid WDT and show the right message from both user or ISR context.

Before #4482, `*((int*)0)=0` was used to trigger an hardware exception (similar to the WDT exception we have n [... cut]
closed 2019-07-15 by earlephilhower