esp8266/Arduino #6286: RequestHandlersImpl.h:67:28: error: expected ')' before '&' token       StaticRequestHandler(FS& fs, const char* path, const char* uri, const char* cache_header)

bkrajendra, 2019-07-11:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			2.5.0-3-20ed2b9
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I need SPIFFS and sdFat tu be running in same code.
My code was running fine until last time I updated to latest master.

So i tried compiling basing ESP8266Advanced Webserver example code, which runs find when compiled.
But when i put 
```
#define FS_NO_GLOBALS
```
 it throws error during compiling:
```
C:\Users\Rajendra\Documents\Arduino\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h:67:28: error: expected ')' before '&' token

     StaticRequestHandler(FS& fs, const char* path, const char* uri, const char* cache_header)

                            ^

AdvancedWebServer:152:1: error: expected '}' at end of input

 }

 ^

AdvancedWebServer:152:1: error: expected unqualified-id at end of input

AdvancedWebServer:152:1: error: expected '}' at end of input
```

earlephilhower, 2019-07-14:
Updated title to make it more searchable.

This is basically the expected behavior.

FS_NO_GLOBALS means that the "using fs:xxxx" lines are not included in the FS.h header, and so much of the core just can't work (especially, say, the static request handler which expects a file...).  You'd need to write your own FS class, implementing the existing FS interface, and prefix any includes  with "using myfsclass:FS", etc.

I think the main reason for this was that in prior versions the SD::File and the SPIFFS::File were incompatible and you needed to jump through some hoops to make them work in a single sketch.  They're now both fully compatible so that's no longer needed.

Did you maybe mean to do `NO_GLOBAL_INSTANCES` or `NO_GLOBAL_SPIFFS`?  Those are the ones which may save some heap, if you're low on memory.

bkrajendra, 2019-07-16:
@earlephilhower thanks for detailed explanation.
Yes im aware of the use of FS_NO_GLOBALS  and have made changes in my sketch accordingly. 
But this time its in library file RequestHandlersImpl.h and dont have idea where/how to change it.
Also this is specific to master branch I guess... its not there in stable channel.

I'll look into it more and get back here.

earlephilhower, 2019-07-16:
There is no need to define FS_NO_GLOBALS anymore if you're using SD and SPIFFS.  So you should really remove that flag from your code, there's no need for it.

Otherwise, if for some reason you really want to keep the define set, then before you include the webserver you need to tell it which FS to use with a "using XXX = FS;" line or it can't disambiguate.

bkrajendra, 2019-07-17:
ok. I just tried to compile my code again by removing FS_NO_GLOBALS and its compiling properly now.
 Im not understanding where i was wrong.

Initially code was having this issue so I added FS_NO_GLOBALS and it resolved compile error on my first PC.
When I moved code to another PC, I started getting this error in RequestHandlersImpl.h

Now when I again tried without FS_NO_GLOBAL onn this second PC everything is ok.
On first PC all works fine with FS_NO_GLOBAL.

I understand it might be version issue. As i mostly use master branch manual install on second PC and my first PC was using stable board manager installation method. 

But when i started getting this issue, I also switch my second PC to stable board manager installation method.

I need to look into this issue again.
By the way thanks for the help @earlephilhower !

earlephilhower, 2019-07-17:
Versions prior to 2.5.2 had two different, incompatible definitions for `class FS`.  One in SPIFFS, one in SD.  So you needed ugly workarounds like the NO_GLOBAL_FS and then needed to choose one or the other when building different objects.

The current version unifies them and replaces the old SD library completely.  So there is now only one `class FS` in the entire tree, and SPIFFS, LittleFS, and SD implement it so they're interchangeable.  `File` and `Dir` are also unified, so everything should "just work."

closed 2019-07-19 by bkrajendra