esp8266/Arduino #6289: EEPROM Time

mjoelmendes, 2019-07-11:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

-Hardware:	[ESP8266EX]
-Core Version:	[2.5.2]
-Development Env:	[Arduino IDE]
-Operating System:	[Windows]

### Settings in IDE

- Module: [Generic ESP8266 Module]
- Flash Mode:	[DOUT (COMPATIBLE)]
- Flash Size: [1MB]
- lwip Variant: [v2 Lower Memory]
- Reset Method:	[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:	[80Mhz]
- Upload Using:	[SERIAL]
- Upload Speed: [115200]
- Espressif FW: [nono-sdk 2.2.1]
- Problem Description

### Problem Description
Hello,
I'm writing to eeprom of esp so it takes me 3 + - seconds to commit is there no way to delay less?
I'm just writing a byte into an allocation whenever the state changes. I'm using this inside the httpclient lib when I get data

          Serial.println("1");
        EEPROM.write(ALOC_STATUS_OUTS + ch, 100); // Save output status to eeprom.
        EEPROM.commit();  
       Serial.println("2");
[image]

devyte, 2019-07-12:
The ESP doesn't have EEPROM. The EEPROM emulates the AVR's EEPROM api by using one flash sector for non volatile memory. Because it is one sector, it has to be written all at once, so to reduce rewriting for every byte, it holds a buffer in memory as a cache, and only writes back to the flash on commit.
The reason I explain the above isn't for the 3s time you're reporting, but rather to say: I hope you're not calling commit frequently, because if you are, you're burning out that one sector, and it will fail very soon.

Now, having said that, I suggest measuring the times around the write and commit calls with millis() instead of relying on the serial console.

mjoelmendes, 2019-07-12:
@devyte Thank you for the quick and enlightening response. I'll see you get the best solution for me

closed 2019-07-12 by mjoelmendes