esp8266/Arduino #6290: mDNS not work in AP mode

lags, 2019-07-11:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-8266MOD]
- Core Version:			[v2.5.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [256000] (serial upload only)

### Problem Description

When set in AP mode mDNS not work for respond with service associated.
Instead in STA mode work successfully with same mDNS configuration.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266mDNS.h>
#include <ESP8266WiFi.h>

IPAddress IP_AP(192, 168, 4, 1);
IPAddress NETMASK_AP(255, 255, 255, 0);

void setup() {
  Serial.begin(115200);
  WiFi.macAddress(macAddr);

  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(IP_AP, IP_AP, NETMASK_AP);
  WiFi.softAP("demoTest", "12345678");

  if (!MDNS.begin('demotest')) {
    Serial.print("$MDNS Error#\r\n");
  }
  MDNS.addService("http", "tcp", 80);
}

void loop() {
  MDNS.update();
}

```

earlephilhower, 2019-07-11:
Please try with the GIT head.  There has been discussion and changes to the networking which may get you going just fine.  If git head (see README.md for instructions on getting it) still fails, you can update this issue report.

lags, 2019-07-12:
I have follow [these lines](https://github.com/esp8266/Arduino#using-git-version-basic-instructions) for install git version but when restart arduino IDE, compile and flash, the result is same. Not work in AP mode. Maybe I must delete older folder "esp8266/2.5.2" before opening IDE?

earlephilhower, 2019-07-12:
If you installed the 2.5.2 release via boards manager, then delete it using the boards manager to ensure there aren't traces left.  Otherwise things can get "interesting" and you'll not be sure what's exactly being run...

lags, 2019-07-17:
I have delete installed version 2.5.2 from board manager and clean folder...
Follow the git guide but now if open arduino ide esp8266 not exsist in manage board.
The new path is: 
`C:\Users\nickname\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266com\esp8266`
It is correct?

lags, 2019-07-17:
Path problem solved editing with:
`C:\Users\nickname\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0`
but after compile and flash, the problem persist. mDNS not work in AP mode.

lags, 2019-07-17:
Solved...
Change Line
`MDNS.addService("http", "tcp", 80);`
to
`MDNS.addService(0, "http", "tcp", 80);`
and now mDNS work successfully with AP mode

closed 2019-07-17 by lags

Blend3rman, 2019-10-22:
I am struggling with this issue, and @lags 's suggestion does not work. In fact, upon doing

> Change Line
`MDNS.addService("http", "tcp", 80);`
to
`MDNS.addService(0, "http", "tcp", 80);`

, the mDNS service stops working in both AP and station modes. Keeping the original line makes mDNS work in station but not AP mode. Is there any fix for this issue @earlephilhower ?

Blend3rman, 2019-10-22:
Here is the logic flow of my application:

1. Disable persistent WiFi mode (`WiFI.persistent(false)`)
2. Set mode as `WIFI_AP_STA`
3. Set softAP config (`WiFi.softAPConfig()`)
4. Start softAP
5. Start MDNS (`MDNS.begin(hostString)`)
6. Add MDNS Service `("http", "tcp", 80)`

Is the order of configurations correct, or am I supposed to init the AP after the MDNS service?