esp8266/Arduino #6291: Add new WiFi.mode(WIFI_SHUTDOWN)

TD-er, 2019-07-12:
As discussed here:  https://github.com/esp8266/Arduino/issues/6172#issuecomment-510413645

It appears the WiFi modes require some steps to switch on and off which have to be taken in a very specific order.
It would be nice to have some generic commands/options to switch WiFi mode.
This would make it a lot easier to create some reliable "going to sleep" actions.

Things to consider:
- Controlling (and keeping state?) of DHCP server(s) (client for STA and server for AP)
- Keeping last known mode (AP/STA/AP+STA)
- Keeping last known connection parameters (BSSID/channel) to allow quick reconnect
- Calling some callback function for cleaning up existing connections.
- Flushing data in buffers.
- Considering some timeout to flush buffers
- Where to keep these stored settings (RTC memory?)

As a counter part operation, it would be nice to have the `WiFi.mode(WIFI_RESUME)` option.
This would allow to collect the stored values (or initiate some default steps when those are not available or corrupt) and resume WiFi operations.

Using this can make the ESP a lot more energy efficient since you only need to have WiFi connection when you need to send data.
Great for battery operated nodes or nodes where heat may quickly become an issue.

devyte, 2019-11-09:
@d-a-v the modes are already implemented. Is there anything pending, or can this be closed?

d-a-v, 2019-11-09:
closed by #6356

closed 2019-11-09 by d-a-v