esp8266/Arduino #6303: NodeMCU stops responding to MQTT messages after a while of inactivity

Debagnik, 2019-07-15:
Hi
I recently created a project for home automation.
I used Adafruit.io website for publishing messages in the MQTT server but one issue always come. The device works fine after resetting or switching on but stops responding after a while
Link to my project is here:https://github.com/Debagnik/arduino/blob/master/Google_Assistant_Home_Automation.ino

I am sorry for all the my noobiness
thanks in advance.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

TD-er, 2019-07-15:
You're only handling a MQTT reconnect in the loop() call, but you never check for a WiFi connection.
If the WiFi itself gets disconnected, you should try to reconnect (or set it to automatic reconnect?)
Also you may want to set the [persistence mode](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html) to disabled, or else you may wear out your flash in a few months since it will then update the WiFi settings on the flash every time it makes a connection.

Debagnik, 2019-07-15:
Thanks for your kind information. If would be very helpful if you could elaborate the solution. sorry for by noobiness

jandrassy, 2019-07-15:
> Also you may want to set the [persistence mode](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html) to disabled, or else you may wear out your flash in a few months since it will then update the WiFi settings on the flash every time it makes a connection.

only changed values are saved. see the SDK reference. OP connects with constatnts

TD-er, 2019-07-15:
> only changed values are saved. see the SDK reference. OP connects with constatnts

It does not save the BSSID or channel?
If it does save the BSSID, then the node may hop between AP and repeater for example and still wear out the flash.  Also change of channel may happen every now and then.

closed 2019-07-15 by devyte

jandrassy, 2019-07-15:
> > only changed values are saved. see the SDK reference. OP connects with constatnts
> 
> It does not save the BSSID or channel?
> If it does save the BSSID, then the node may hop between AP and repeater for example and still wear out the flash. Also change of channel may happen every now and then.

I believe that only the entered parameters are saved, because if I don't specify BSSID it should connect to any BSSID of specified SSID

TD-er, 2019-07-15:
@jandrassy I have not looked in the code of the library, but a reconnect like this can be done in less than 2.4 sec, which means it must know enough to skip the WiFi channel scan which takes about 2.4 sec (200 msec per channel, 12 channels)

Debagnik, 2019-07-15:
Well the code that i dumped into the MCU had a specific SSID and password. I didn't wanted to upload with the SSID so I removed that from the code before uploading.

Anyways I am extremely noob about this so the most of the technical terms above goes above my head. So I request the community to give the solution elaborately and stepwise

Thanks

TD-er, 2019-07-15:
The quickest (and dirty) solution is to call for a reboot as soon as you cannot make a MQTT reconnect.

`ESP.restart();`

jandrassy, 2019-09-03:
the wifi channel is stored temporary in RTC memory.
from https://www.espressif.com/en/support/explore/faq

> After ESP8266 connects to an AP, it will save the AP’s channel information in RTC memory.
> 
> When you reset ESP8266 by software or the ESP8266 wakes up from Deep-sleep mode, it will read the AP’s channel information from RTC memory, then try to connect the AP in the channel.
> But if the ESP8266 goes through a power cycle or is hardware reset, the RTC memory will be cleared. The ESP8266 will then scan through all the channels, which takes some time.
> 
>  Note that the auto connection function should be enabled for the ESP8266 to read the AP’s channel information, and to connect to the AP in that channel.
> 
> Only channel information is cached in RTC memory. The other information (such as SSID and Password) is saved in Flash, when you call the function wifi_station_set_config.

srd1994, 2020-05-04:
Hello There,

Hope you find this message and this reply helps you resolve your issue.

I've been recently working with Node MCU/ ESP 8266 module.
I'm using it to turn ON/ OFF Lights and Fans. Basically, a home automation project.

I FOUND THE SAME ISSUE OF NO RESPONSE FROM NODE MCU AFTER 10-15 MINS OF INACTIVITY.

This is quite irritating, as I always had to reset the module to make it work again.
**** BUT I HAVE A SOLUTION TO IT NOW!!!!!

Basically, Node MCU constantly reads the Subscription and acts accordingly to turn ON/OFF any devices.
*subscription is a pointer which reads the Subscription value being sent from Adafruit.

Adafruit_MQTT_Subscribe *subscription;
while ((subscription = mqtt.readSubscription(5000)))

If there is inactivity for a long period of time, there is a blank value sent and the connection between MQTT broker and NodeMCU is broken.
We must send a ping to let the broker know you're around!
So you need to add a simple line mqtt.ping(); just before closing your VOID LOOP bracket.
VOILA! This resolves the issue completely.
Now you are constantly pinging your broker and no connections breaks.

I've tried this in my project. I sent a command at night and tried to send another command in the morning, after 12 hours. AND IT WORKED!!!!!

Just add a ping in your VOID LOOP and get rid of this issue.
Good Luck!

More Info about Adafruit MQTT: https://learn.adafruit.com/mqtt-adafruit-io-and-you/intro-to-adafruit-mqtt