esp8266/Arduino #6304: HTTPUpdate problem

Jhony9625, 2019-07-15:
Hello, I am using the httpUpdate example and I get wrong http code the server link is this: http://qualitysystems.com.co/Blink.bin

I searched the internet and found this 2016 post https://github.com/esp8266/Arduino/issues/2022 I tried what it says but it has not worked and everything I find is very old. I do not know what else to do to solve this problem.

my code:
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

#ifndef APSSID
#define APSSID "HERMES"
#define APPSK  "55469347daz"
#endif

ESP8266WiFiMulti WiFiMulti;

void setup() {

  USE_SERIAL.begin(115200);
  // USE_SERIAL.setDebugOutput(true);

  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  for (uint8_t t = 4; t > 0; t--) {
    USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
    USE_SERIAL.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(APSSID, APPSK);

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    WiFiClient client;

    // The line below is optional. It can be used to blink the LED on the board during flashing
    // The LED will be on during download of one buffer of data from the network. The LED will
    // be off during writing that buffer to flash
    // On a good connection the LED should flash regularly. On a bad connection the LED will be
    // on much longer than it will be off. Other pins than LED_BUILTIN may be used. The second
    // value is used to put the LED on. If the LED is on with HIGH, that value should be passed
    ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);

    t_httpUpdate_return ret = ESPhttpUpdate.update(client, "http://qualitysystems.com.co/Blink.bin");
    // Or:
    //t_httpUpdate_return ret = ESPhttpUpdate.update(client, "server", 80, "file.bin");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        USE_SERIAL.println("HTTP_UPDATE_OK");
        break;
    }
  }
}

Jeroen88, 2019-07-16:
I tried your URL for redirects with [this](http://www.redirect-checker.org/index.php) tool. HTTPUpdate does not support redirects. I think if you try http://www.qtsolutions.com.co/Blink.bin instead it will work

liebman, 2019-07-16:
Redirects are optionally supported.  You need to call ```ESPhttpUpdate.followRedirects(true)``` to enable it.

Jhony9625, 2019-07-17:
@Jeroen88 Thank you very much for the help, it has worked perfectly, now I will investigate how to make it detect if there is a new update to upload the code automatically.

@liebman I will try to prove what you say, thank you

earlephilhower, 2019-07-26:
Looks like @Jeroen88  and @liebman pointed out the redirect issue, thanks.  Closing.

closed 2019-07-26 by earlephilhower

Kasperdelasopa, 2023-06-11:
Hello,

even if i enable the redirect i get the error:

`HTTP_UPDATE_FAILED Error (-5): HTTP error: connection lost`

I do the following:

```
WiFiClient otaclient;

ESPhttpUpdate.rebootOnUpdate(true);
ESPhttpUpdate.followRedirects(true);
t_httpUpdate_return ret = ESPhttpUpdate.update(otaclient, url);

switch(ret) {
	case HTTP_UPDATE_FAILED:
		Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
		break;

	case HTTP_UPDATE_NO_UPDATES:
		Serial.println("HTTP_UPDATE_NO_UPDATES");
		break;

	case HTTP_UPDATE_OK:
		Serial.println("HTTP_UPDATE_OK");
		break;
}
```

I try to download the binary from my GitHub account:

`https://github.com/Kasperdelasopa/AutoWater/raw/main/20230611_AutoWater.bin`

Can someone give me an hint what i can improve ?

Best regards 
Manuel