esp8266/Arduino #6308: WiFi.begin() always fails to connect

aryghm, 2019-07-17:
### Basic Infos

- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have only tested this issue in core version 2.5.0.
- [X ] I have searched the issue tracker for a similar issue.
- [X ] If there is a stack dump, I have decoded it.

#### Platform
 
- Hardware:			        ESP-12 LOLIN
- Core Version:			2.5.0
- Development Env:		other (vMicro for visual studio)
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			Unknown
- Flash Size: 		4MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   other (921600)

### Problem Description

I am trying to connect to wifi with my esp8266 to communicate to some application over tcp,
my problem is that I can't connect to a wifi router using WiFi.begin() in esp the problem is WiFi.begin() always fails to connect. I checked SSID and password they're both right I've enabled debug mode and I received some messages that I will share below.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Notice: the code below is not the exact same code, and I'm not going to copy all because it's about 3400 lines of code.
```cpp

#include <Arduino.h>

void setup() {
 //connect to your local wi-fi network
  WiFiConnect();

  //WiFi.begin(ssid, password);

  //Check WiFi connection
  if (IsWiFiConnected())
  {

	  //clear for new frame
	  display.clearDisplay();

	  display.setCursor(0, 0);
	  display.setTextSize(2);
	  display.print("Got IP: ");
	  display.println(WiFi.localIP());
	  display.display();
  }
  else
  {
	  //Set WiFi mode to access point
	  WiFi.mode(WIFI_AP);

	  display.clearDisplay();
	  display.setTextSize(1);
	  display.setCursor(0, 0);
	  display.println("Error On Connecting To WiFi, Running as access point.");
	  display.display();
	  delay(1000);
	  //Serial.println("Couldn't connect to wifi");
  }
}

void loop() {

}

void inline WiFiConnect()
{
	//clear for new frame
	display.clearDisplay();

	byte WifiatemptCount = 0;

	display.setCursor(0, 0);
	display.setTextSize(1);
	display.print("Conneting to WiFi");
	//WiFi.begin(ssid, password);
	WiFi.begin("Samaneh", "raymond9111");
	//check wi-fi is connected to wi-fi network
	//retry until maximum attemp count
	while ((WiFi.status() != WL_CONNECTED) && (WifiatemptCount <= 10)) {
		display.print("xxxxxyyyyyyy");
		delay(500);
		WiFi.begin("Samaneh", "raymond9111");
		display.print(".");
		display.display();
		WifiatemptCount++;
	}
	display.println();
	display.display();
}

```

### Debug Messages

```
mode : sta(b4:e6:2d:70:a0:82) + softAP(b6:e6:2d:70:a0:82)
add if0
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
del if0
mode : softAP(b6:e6:2d:70:a0:82)
```
any help appreciated.

TD-er, 2019-07-17:
You do call `WiFi.begin` again in the while loop.
In the while loop you should just wait for the wifi connection you initiated before the while loop.
Also you're only waiting for 5 seconds (10x 500 msec), which may be a bit short.

jandrassy, 2019-07-17:
every call to WiFi.begin starts connecting from the beginning, but WiFi can't be connected in 500 millis

earlephilhower, 2019-07-17:
Thanks, @jandrassy and @TD-er .

closed 2019-07-17 by earlephilhower

ThermalEnergy, 2020-08-30:
In order to connect to Wifi from ESP32 and other boards, the WiFi connection needs to be 2.4GHz, wireless mode must be legacy. It cannot be set to Auto or N only. Go to your router box web page, and change this setting under Advanced Settings > Wireless.

HisNerdship, 2021-02-15:
Thank you, ThermalEnergy, that had been holding me up for a couple of days!  I was trying to connect to the WiFi through MKR WiFi 1010 on a 5GHz connection.  Very annoyed with Arduino for not mentioning this in the Getting Started pages.  Well, maybe it is there, but it's very well hidden.
Muchas gracias, Señor!