esp8266/Arduino #6311: Add time and date to FS filesystems

earlephilhower, 2019-07-18:
#6276 brought it up, but it's a more general concern.  If we can solve the general problem then we don't need to do ugly hacks to pull out underlying, wrapped objects for users.

What is the proper way to enable timestamps through the existing FS interface.  Setting a file's create/access/write time (as appropriate and available) on creation, at other times, and getting those values from the FS?

SDFat has a global dateTimeCallback() which returns two `uint16_t`s in a very MSDOS-specific format, which is called by the underlying FS library when it needs to get a date/time:
https://github.com/greiman/SdFat/blob/master/examples/Timestamp/Timestamp.ino

The old Arduino SD.h fork also exposed these.

SPIFFS and LittleFS might be able to be extended to have date/time metadata, as well, so this should also be generally applicable.

@devyte, @me-no-dev , @d-a-v, any ideas, suggestions?

devyte, 2019-07-18:
There was a PR that enabled date/time for spiffs, but I think it was compile time. 
If memory serves, the problem was that the enabling makes spiffs not backwards compatible, so flashing a new sketch with that enabled would cause a failed mount and reformat.
I suggest enabling for 3.x.
No idea about LittleFS.

earlephilhower, 2019-07-18:
Yes, it was incompatible w/existing code I think.  My question/worry has to do with the API, how it's actually set/called.

SDFat has a callback you set per-file that gets called when it needs the time/date.  That might be the sanest to implement (and be no-op for spiffs/littlefs until they're tweaked) with a default CB applied that takes the current time from `time()` (assuming NTP has happened).

luc-github, 2019-07-18:
My PR for SPIFFS is planned for 3.0 due to backward compatibility, but littleFS is not yet in any release, so why to wait for 3.0 to implement date/time for littleFS which is may be in years ? 

I did not digged in latest LittleFS code yet to see the impact, but feature is definitly there : https://github.com/ARMmbed/littlefs/pull/31#issuecomment-448291274 

There is no backward compatibility for littlefs  issue yet, I think it would be better to add it now IMHO   
 
Also if implementation is done like in ESP32  (SD/SPIFFS/Fat), from user point of view the API has only one function : https://github.com/espressif/arduino-esp32/blob/master/libraries/FS/src/FSImpl.h#L40
`time_t getLastWrite()` which can simply send 0 if time is not yet implemented like in SPIFFS so no backward compatibility as it is new function.

earlephilhower, 2019-07-18:
I like the idea, @luc-github.  What about setting the time on a file on the ESP32, is it automatic (i.e. the FS just calls `time(NULL)`?)  I don't see any overrides in the API...

luc-github, 2019-07-18:
On ESP32 I did 2 PR :  one on IDF  https://github.com/espressif/esp-idf/pull/1128 to prepare the one for Arduino https://github.com/espressif/arduino-esp32/pull/738 because time is set in idf part

igrr, 2019-07-18:
For FATFS, file time is also set from `time(NULL)`: https://github.com/espressif/esp-idf/blob/d7e659df27cf00b5f831686bb94162a97bea62be/components/fatfs/diskio/diskio.c#L81

earlephilhower, 2020-05-26:
Closing as a won't-fix due to SPIFFS.  LittleFS and SDFS both have file-time support already.

closed 2020-05-26 by earlephilhower