esp8266/Arduino #6314: Hard resetting via RTS Pin on write

larrywal, 2019-07-18:
Hi folks - bought some new ESP8266 boards and downloaded NodeMCU drivers.  Every time I try to upload a sketch, I get the attached

I've tried with NodeMCU 0.9 (worked for me before with these boards), NodeMCU 1.0, as well as generic ESP8266.  Same issue every time.  Can somone help me debug?  Writing to Arduino Uno is working fine.

[image]

5chufti, 2019-07-18:
yes, where is the problem?
It writes your sketch, verifies ok and with reset via RTS starts it ... ???

earlephilhower, 2019-07-18:
Yes, everything looks just fine on the screenshot.  After uploading it *always* resets the chip to start the new program running.

closed 2019-07-18 by earlephilhower