esp8266/Arduino #6316: Update SDK

mbm60, 2019-07-19:
Hi
The ESP8266_NONOS_SDK [v3.01](https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v3.0.1) released with many bug fix.
Maybe we can update SDK to latest version now.

devyte, 2019-07-19:
Ref #5784.

Closing.

closed 2019-07-19 by devyte

cziter15, 2019-08-30:
There are a lot of iee80211 library, that may fix #2330 
It might be really worth to try update SDK. 

Are there a lot of hacks that don't work in 3.0.1?

d-a-v, 2019-08-30:
NONOSDK is EOL.
Espressif is putting their efforts into RTOS-SDK.
So the best next move is probably there.
I am personally waiting for this [issue](https://github.com/espressif/ESP8266_RTOS_SDK/issues/635) to be answered.
That will be a good reason to start switching (Imho).
#2330 is another good reason.

d-a-v, 2019-08-30:
About NONOS-SDK, espressif has made a 2.2.x branch in their repository.
Git version of this core is currently using the latest of this 2.2.x branch.

cziter15, 2019-08-30:
I see that 2.2.x branch has been updated to NONOSV22y directory.
When do you plan to release 2.6.0?

>
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

d-a-v, 2019-08-30:
Probably after we rename this "22y" menu option which was initially a test :)

cziter15, 2019-08-31:
It looks like there is significant improvement in connection stability with
22y aka +100 version. Did they added same fixes as in 3.0 to 22y?