esp8266/Arduino #6318: Wemos d1 mini not waking up

sntlyeet, 2019-07-19:
Im trying a simple deepsleep sketch to use it in one of my projects but when the deepsleep ends, the led flashes once and nothing happens. how can i fix this?

i also debuged it with Serial.setDebugOutput(true);

### Debug Messages
22:02:17.751 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
22:02:17.751 ->
22:02:17.751 -> load 0x4010f000, len 1384, room 16
22:02:17.751 -> tail 8
22:02:17.751 -> chksum 0x2d
22:02:17.751 -> csum 0x2d
22:02:17.785 -> v8b899c12
22:02:17.785 -> ~ld
22:02:51.502 ->
22:02:51.502 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
22:02:51.502 ->

my code:

#include <ESP.h>

void setup() {
  Serial.begin(2400);
  Serial.setDebugOutput(true);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print("Uaaah. I'm tired. Going back to bed for ");
    Serial.println(" minutes. Good night!");
    Serial.println("---");
    ESP.deepSleep(20 * 1000000);
    delay(100);
    Serial.println("dochnicht");
}

TD-er, 2019-07-19:
Just curious, why are you using such a slow baudrate for the serial port?
2400 bps is roughly about 240 characters per second.
Also the loop() function does not really run very long.
It is probably only awake for a few msec and then you don't have time to output anything to the serial port.

fsommer1968, 2019-07-23:
Set the baud rate to 74400 and test again and post the new logfile if still required.

suculent, 2019-07-25:
Or maybe user `Serial.flush();` after last print?

bjoernbusch, 2019-12-26:
I'm facing the same issue. I increased the baud rate to 74880, but the log doesn't show anything helpful. Any hints on this?

Tech-TX, 2019-12-29:
Put a 1 to 2 second delay before you hit the ESP.deepSleep() command if you're doing something like the original poster.  His code only runs for a few microseconds before it goes back into deep sleep, so the serial port never has a chance to get the first byte of his print statement sent out, and then the serial buffer is wiped when the chip resets and boots again after deep sleep.

bjoernbusch, 2019-12-29:
Thanks, @Tech-TX it was actually an issue with the resistor between D0 and RST. Somehow my 560 resistor was too much, I replaced it by a simple wire and now it works.

Tech-TX, 2019-12-29:
I was wrong about the delay(); I just pulled up my old test program and it works.  I'm using a D1 Mini.

```cpp
  // Deep sleep test for 5 seconds (5E6 microseconds) - the ESP8266 wakes up by itself if
  // GPIO16 (D0) is connected to the RESET pin or you press the RESET switch / ground RST.
  //
  // You'll need to ground GPIO0 (D3) and press the RESET switch to upload again after
  // this is running or the serial won't connect 'cos it's almost always ASLEEP.
  
void setup() {
  Serial.begin(74880);
//  while (!Serial) { } // Wait for serial to initialize (wasn't needed)
  Serial.println("I'm awake, but I'm going into deep sleep for 5 seconds");
}

void loop() {
  // your code goes here
    ESP.deepSleep(5E6); // good night!
    Serial.println("What... I'm not asleep?!?");  // it will never get here
}
```

Here's the output on the serial monitor:

> load 0x4010f000, len 3456, room 16 
> tail 0
> chksum 0x84
> csum 0x84
> v3197d2ac
> ~ld
> I'm awake, but I'm going into deep sleep for 5 seconds
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 3456, room 16 
> tail 0
> chksum 0x84
> csum 0x84
> v3197d2ac
> ~ld
> I'm awake, but I'm going into deep sleep for 5 seconds

<repeats every 5 seconds>

I only get down to about 17mA current draw during Deep Sleep with my D1 Mini due to the horrid linear regulator used on this cheap clone board.  A Holtek HT7833 would get lower as the quiescent current is better on their parts.

[image]

devyte, 2019-12-29:
Per @Tech-TX:
Closing due to unable to reproduce.

closed 2019-12-29 by devyte

Erriez, 2020-10-09:
@Tech-TX, @deive: Today I received a batch of Wemos D1 mini's and two of them failed to wake-up as well. I tried different resistor values and wires between D1 (GPIO15) and RST, but did not make sense.

After a lot of wasting time, I found this interesting thread:
https://github.com/esp8266/Arduino/issues/6007#issuecomment-542192213

With this sketch I'm able to wake the Wemos D1 mini:

```
#include "ESP8266WiFi.h"
extern "C" {
#include <user_interface.h>
}

#define ets_wdt_disable ((void (*)(void))0x400030f0)
#define ets_delay_us ((void (*)(int))0x40002ecc)

#define _R (uint32_t *)0x60000700

void nk_deep_sleep(uint64_t time)
{
    ets_wdt_disable();
    *(_R + 4) = 0;
    *(_R + 17) = 4;
    *(_R + 1) = *(_R + 7) + 5;
    *(_R + 6) = 8;
    *(_R + 2) = 1 << 20;
    ets_delay_us(10);
    *(_R + 39) = 0x11;
    *(_R + 40) = 3;
    *(_R) &= 0xFCF;
    *(_R + 1) = *(_R + 7) + (45*(time >> 8));
    *(_R + 16) = 0x7F;
    *(_R + 2) = 1 << 20;
    __asm volatile ("waiti 0");
}

void setup()
{
    delay(1000);

    nk_deep_sleep(2000000);
}

void loop()
{
}
```

However, I've no idea about the internals and how to wake with or without RF enabled. According to the thread, it looks like an issue with newer ESP-12F. Do you have more information on this?

devyte, 2020-10-09:
No new info.

studioant, 2020-10-09:
@devyte, I agree. And I think the person to do it is @nikolz0. He wrote that function above. Problem is, nk_deep_sleep consumes ~2mA after cutting the LDO and UART 3V3 trace, so it's nowhere near "deep sleep". If anything, nikolz0 has created a working "light sleep'. I'm trying to pick their brain to comment the code so we know whats happening, and if they have time, try and figure out the modification required to make it go "deep"

studioant, 2020-10-09:
@Erriez , according to the author, the line *(_R + 4) = 0; disables WiFi.
In light of this information the author has provided, we can assume this person is certainly capable of solving this issue once and for all. We need this hero in our lives

devyte, 2020-10-10:
@studioant that sounds like some pull-up resistor somewhere to me. Or a resistor voltage divider.

studioant, 2020-10-10:
@devyte You might be right, I found this Mux Register info:
[image]
I'm not sure what the hex should be for "Pull-down during sleep" but it looks like somewhere after 0x60000804.

This is on esp8266 wiki: https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map 
[image]

@nikolz0 defines his register variable _R starting at this hex value, so he's manipulating the RTC, but i can't find the register doc on this... where's the include/eagle_soc.h header file?

Erriez, 2020-10-10:
Thanks for looking into this. I checked my stock and discovered multiple ESP8266 chips which does not wake up after deep sleep.

@studioant You're right: The current consumption is 2.5mA with the magic workaround sketch. That explains why my batteries are drained quickly... :cry: 

@devyte 
> It would be interesting if somebody could reverse engineer the sdk deep sleep calls and figure out the difference vs the above code, especially the wakeup param. We could then implement our own version of deep sleep.

I'd like to summarize some facts about this issue, because I found a lot of wrong assumptions and discussions on other threads:
* This problem has been reported on newer ESP12F chips only.
* This repository uses the latest Espressif NONOS SDK version "2.2.2-dev(38a443e)", but needs volunteers to port it to the latest NONOS SDK version 3.0.0.

Please correct me if I'm wrong.
Did

nikolz0, 2020-10-10:
Hi, try downloading this test from the address 0x0000, terminal 74880, deepsleep 12 sec.

[testnk.zip](https://github.com/esp8266/Arduino/files/5359198/testnk.zip)

Erriez, 2020-10-10:
@nikolz0 The output is:

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 580, room 16 
tail 4
chksum 0x62
load 0x3ffe8004, len 268, room 4 
tail 8
chksum 0x2e
csum 0x2e

testnk deepsleep

...
Same output
```

It resets every ~12 seconds... Eh... this sounds promising. Can you share the contents of this bin?

nikolz0, 2020-10-10:
Hi,
This is code without the SDK, only the deepsleep function . Made for checking ESP.
You can do the same in your program using the SDK function 
**system_deep_sleep_instant(**12000000);
 In the user_init function, call 
**system_deep_sleep_set_option(2);**

fsommer1968, 2020-10-10:
>     * This repository uses the latest Espressif NONOS SDK version "2.2.2-dev(38a443e)", but needs volunteers to port it to the latest NONOS SDK version 3.0.2.
Arduino for 8266 does not support NONOSSDK 3.x, there is a long thread in the issues section about this.

devyte, 2020-10-10:
@Erriez you are correct: we're still on NONOS sdk 2.x, and we want to migrate to 3.x. We've had a few attempts, but walls have been hit. Volunteers in this case means either someone who has a deeper understanding of how our core interfaces with the sdk, or someone who is willing to stick around long term and aquire that understanding in time.
I outlined somewhere what I think needs to be done at high level.

Erriez, 2020-10-10:
@nikolz0 

> This is code without the SDK, only the deepsleep function .
I don't know how to build it without SDK. In my understanding include file "user_interface.h" is always required to access the Espressif SDK, right?

> In the user_init function, call `system_deep_sleep_set_option(2);`

For this repository, I found a `preinit()` function which is called from `user_init()` and added the `system_deep_sleep_instant(12000000);` to `setup()`. 

It wakes up twice as I encountered before:

```c++
extern "C" {
#include "user_interface.h"

void preinit() {
    system_deep_sleep_set_option(2);
}
}

void setup() {
    // Does not wake:
    system_deep_sleep_instant(12000000);
}

void loop() {
}
```

Output:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

No further resets...
```

Any Idea?

Tech-TX, 2020-10-10:
> @Tech-TX, @devyte: Today I received a batch of Wemos D1 mini's and two of them failed to wake-up as well. I tried different resistor values and wires between D1 (GPIO15) and RST, but did not make sense.
> 
> After a lot of wasting time, I found this interesting thread:
> [#6007 (comment)](https://github.com/esp8266/Arduino/issues/6007#issuecomment-542192213)

...
> However, I've no idea about the internals and how to wake with or without RF enabled. According to the thread, it looks like an issue with newer ESP-12F. Do you have more information on this?

I read reports over on esp8266.com about [Deep Sleep 'zombie mode'](https://www.esp8266.com/viewtopic.php?f=166&t=20500), but never saw it here at home.  Where did you get the D1 Mini boards from?  I'd like to be able to duplicate your results here.

For the GPIO16 > RST connection I use either a Schottky or germanium diode.  The cathode of the diode (the stripe) goes to GPIO16, and the anode goes to RST.  Using a resistor could effect the pulse getting to the RST line of the CPU, and exactly what it does depends on what they've done underneath the can on the ESP-12F.  I've seen 4 different RF shields from different companies in China, including one that has copied the Lolin logo (although I doubt Lolin made them.)

studioant, 2020-10-11:
@Tech-TX , The issue is clearly deeper than this, you even commented in February in the original thread.

>In one thread I read someone removed the RF shield and found a resistor in series between the RST pin on the module and the RST pin on the CPU. 

Yeah, the thread you literally included in this comment? There are even recorded attempts (in said thread) to remove the resistor, bypass it, etc to no avail. Obviously, not gonna help.

@nikolz0 explained it in that thread, plus he created a (almost spot-on) solution. It just needs further enhancement, atm it's allowing the ESP to draw too much current during deep sleep (I understand you have recorded lower current draw, but I have now recorded 2mA consistently across several bare-bone ESP's )

@nikolz0 - Do you have a copy of the RTC register manual? I can't seem to find it anywhere. It would be great to understand better what your function is doing. TIA

nikolz0, 2020-10-11:
Hi,
I connect the GPIO16 and RST using a Schottky diode. 
So I have no problem with the versions of the ESP-12.

nikolz0, 2020-10-11:
Hi, 
Are you using Wemos or ESP-12? 
is testnk working correctly?
[testnk.zip](https://github.com/esp8266/Arduino/files/5360595/testnk.zip)
There is an automatic mode "vemos" control. I believe that it interferes with deep sleep mode.
I write in C, but I'll try to tell you something.
try this:
`
#include <ESP8266WiFi.h>
void setup() {
  Serial.begin(115200);
  Serial.setTimeout(1000);
  while(!Serial) { }
}

void loop() {
    Serial.println("Test");
      delay(1000);
  ESP.deepSleep(12000000); 
  }
`

nikolz0, 2020-10-11:
https://github.com/esp8266/Arduino/issues/3408

Tech-TX, 2020-10-11:
I'm not sure what you're linking to @nikolz0 as I think that thread has been resolved.

If there's a specific post that's relevant, you can pick it with the three dots to the right of the post title bar,
[image]

Tech-TX, 2020-10-11:
FWIW if I run nicolz0's test binary it works as expected, and I get around 15uA sleep current.  If I run that test code that Erriez linked [above](https://github.com/esp8266/Arduino/issues/6318#issuecomment-706386677) it only sleeps for 1.4ms, no matter what I set 'time' to in the nk_deep_sleep(time) call.  It's sleeping so briefly that I can'[t measure the current, but it looks like it's on the order of 2 to 4mA.

Erriez, 2020-10-11:
@Tech-TX 
> @Erriez I read reports over on esp8266.com about Deep Sleep 'zombie mode', but never saw it here at home. here did you get the D1 Mini boards from? I'd like to be able to duplicate your results here.

I've purchased multiple batch from different shops and I did not mark them. My last batch came from Aliexpress [Sincere Company Store Wemos D1] Mini](https://www.aliexpress.com/item/32649468489.html?spm=a2g0s.9042311.0.0.ae2b4c4d5JeLK8). 
Only one of them cannot wake after deepsleep, others work as expected. I cannot find any differences in PCB, so it is concerning that nobody knows which version will be shipped.

> The cathode of the diode (the stripe) goes to GPIO16, and the anode goes to RST

I could not get it stable. A software solution would be appreciated.

@studioant There are different deepsleep issues, but I could not find a solution for this problem.

@nikolz0 

> Are you using Wemos or ESP-12?

Yes, this is a Wemos ESP12F D1 Mini (lite) with 4MB flash. That's the reason why I posted on this thread, but seems more related to newest ESP12F batches.

[image]

[image]

> is testnk working correctly? testnk.zip

Yes, the testnk is the only binary which works on my device with a simple 0 Ohm wire between RST and GPIO16 (D1): Wake works and 170uA in deep sleep. So now I'd like to know how to port this to my application by using this repository.

There is a magic difference between your binary which works and `ESP.deepSleep(12000000);` in this repository which does not work on several of my devices:

```
SDK version: 2.2.2-dev(a58da79)
Wake

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

=> Does not start the sketch.
```

I'll hookup my scope to measure the RST pulse and measure the current...

Erriez, 2020-10-11:
The RST pulse length for all devices and tests is 50us:

[image]

* Board 1: Wake works with ESP.deepsleep().
---
* Board 2: Wake does not work with ESP.deepsleep().
* Board 2: Wake works after flashing `testnk.bin`.

@nikolz0  @Tech-TX I'm curious about the differences between the `testnk.bin` and this repository...

Tech-TX, 2020-10-11:
I can't tell what is inside testnk.bin; perhaps nikolz0 would post the source,  I can't tell you how it's different from ESP.deepsleep().  If I had the elf for it I could run a disassembly of it. but that's a painful task since the disassembly frequently gets out of sync and starts puking up junk.  From the size of his file, it's assembly or compiled (something) without the SDK.

I've seen people selling those ESP-12F modules you [showed above](https://github.com/esp8266/Arduino/issues/6318#issuecomment-706751046), but with no vendor name and a (likely) forged copy of the Espressif logo on the RF can I always avoided them.  All of mine are either DOIT or AI Thinker, and those work fine.  I haven't tried any of DOIT's newer modules labeled DOITING, but I've heard of problems with them.

The [RST pulse](https://github.com/esp8266/Arduino/issues/6318#issuecomment-706755232) you're showing is on the module pin, not inside on the chip, correct?  With a (possible) series resistor inside the RF shield, all bets are off on what it looks like at the chip.  For sure it will be shorter duration and may not hit 0V; it depends on the size of the pull-up resistor (if any) inside the module plus the capacitor.

The reason I use a Schottky or germanium diode is because the RTC puts GPIO16 in OUTPUT mode when you're in Deep Sleep.  If you externally pull RST low you're shorting the GPIO16 output to ground.  With a diode it's protected.

With the LDO and CH340G installed, most of my modules run around 6mA during Deep Sleep.  If I cut them off, it drops to 16uA during Deep Sleep.  The LDO on my boards is marked 4A2D (no logo), which could be Torex or more likely one of the no-name Chinese clone parts.  The CH340G _looks_ like a real QinHeng part, but I know there are clones of it.

Your module looks like a D1 Mini, but it isn't Wemos (no 'WeMos.cc' name on it) so it's a clone.  I've seen a genuine Wemos board and it had the Wemos name.

nikolz0, 2020-10-12:
Hi, put not just a diode, but a Schottky SS14 diode. 
Connect the cathode to GPIO16. 
Do not set the pull-up resistor to RST.

nikolz0, 2020-10-12:
Hi,
this test is with the SDK. Address 0x00000,0x10000, terminal 74880.
[testnkSDK.zip](https://github.com/esp8266/Arduino/files/5362799/testnkSDK.zip)

nikolz0, 2020-10-12:
[image]

nikolz0, 2020-10-12:
Hi,
this test uses SDK_3.05. address 0x00000, 0x10000, terminal 74880.
[testnk_SDK_305.zip](https://github.com/esp8266/Arduino/files/5362857/testnk_SDK_305.zip)

[image]

Tech-TX, 2020-10-12:
Here's the schematic for the AI Thinker ESP-12F, from their a014ps01 V1.pdf:
[image]

Erriez, 2020-10-12:
@Tech-TX 

> If I run that test code that Erriez linked above it only sleeps for 1.4ms, no matter what I set 'time' to in the nk_deep_sleep(time) call. It's sleeping so briefly that I can'[t measure the current, but it looks like it's on the order of 2 to 4mA.

Correct, the power consumption of that skatch is too high for battery usage. So that's not a solution.

> The RST pulse you're showing is on the module pin, not inside on the chip, correct? 

Correct. this is measured at the RST pin and wire to GPIO16 D0.

> With a (possible) series resistor inside the RF shield, all bets are off on what it looks like at the chip. It may be shorter duration and may not hit 0V; it depends on what's under the can. A series resistor will mess up the RST pulse.

So next step would be to remove the shield and measure strait with a scope on the ESP8266EX `XPD_DCDC` pin 8? 

> It may be shorter duration and may not hit 0V;

I've doubts that we are looking at a hardware issue, because `testnk.zip` in https://github.com/esp8266/Arduino/issues/6318#issuecomment-706560461 works flawless without SDK, but it is a binary...

Note: I posted one scope picture of the RST pulse, but the signal is identical to another working board.

> I tried different resistor values and wires between D1 (GPIO15) and RST,

Good catch, that's a typo. I've updated the comment.

> If you externally pull RST low you're shorting the GPIO16 output to ground. With a diode it's protected.

Aha, now I understand the function of this diode.

> He also had ESP-12F modules with an Espressif logo on them.

Interesting. Is Espressif the only manufacture of the ESP8266EX chip, or are their counterfeit chips on the market? It becomes a big problem nowadays.

> Your module looks like a D1 Mini, but it isn't Wemos (no 'WeMos.cc' name on it) so it's a clone.

Yes, only clones left since WeMos.cc stopped selling it. The quality of the blue PCB and soldering is good. I'm not sure who manufactures the ESP8266 (small PCB with antenna). I considered to desolder this board from the blue PCB to exclude hardware problems, but I don't have the right equipment to do this. Now I see that there are some components under the shield. A strait forward schematic of the AI Thinker, but not sure if this matches with mine.

> With the LDO and CH340G installed, most of my modules run around 6mA during Deep Sleep.

I must say that the deep sleep of the working boards without modifications is around 170uA which is good. I also measured around 4..6 mA with NodeMCU's.

> If I cut them off, it drops to 16uA during Deep Sleep. 

A while ago I did the same, but it was hard to find the right trace as there are different Wemos D1 Mini PCB's.

The LDO is a `DE=A1D`. (Thanks to my cellphone microscope)

The mounted CH340G is stable with flashing at 3000000, so I think it's a genuine one:

```
Bus 002 Device 091: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
idVendor:  0x1a86 QinHeng Electronics
idProduct: 0x7523 HL-340 USB-Serial adapter
```

**MAC STA AP: C8:2B:96:2F:8E:C6**
Is a known Espressif MAC address: https://www.adminsub.net/mac-address-finder/espressif

With only 10 business days shipping to The Netherlands, it may be an idea to buy a few samples from Sincere.

I think I've answered your questions, otherwise let me know. Next post I'll show the zip results.

Erriez, 2020-10-12:
@nikolz0 

> put not just a diode, but a Schottky SS14 diode.

Unfortunately I don't have Schottky diods on stock, so I tried a germanium diode as @Tech-TX suggested.

**testnkSDK**

Thanks for the binaries. I've flashed:
* 0x00000 for eagle.flash.bin
* 0x10000 for eagle.irom0text.bin

[image]

With or without a wire or diode, the CPU resets every 166ms. Maybe I do something wrong.

This is the same for `testnk_SDK_305`. Screenshot download tool:

[image]

Did I configure it incorrectly, or miss a binary?

nikolz0, 2020-10-12:
Hi,
message rf_col[0]!=0x05 indicates
that you have incorrectly loaded the default system settings.

Erriez, 2020-10-12:
Ok, I've restored the AT commands first from https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v3.0.4

[image]

AT commands works. [AT instruction set](https://www.espressif.com/sites/default/files/documentation/4a-esp8266_at_instruction_set_en.pdf)

Then flash **testnkSDK**:

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29312, room 16 
tail 0
chksum 0x7d
load 0x08005210, len 1146759052, room 8 

===> No reset pulse generated on GPIO16
```

Then flash **testnk_SDK_305**:

```20:57:19.130 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
20:57:19.130 -> 
20:57:19.196 -> load 0x40100000, len 31216, room 16 
20:57:19.229 -> tail 0
20:57:19.229 -> chksum 0x4a
20:57:19.229 -> load 0xd8b10800, len 769779921, room 8 
20:57:23.249 -> flash read err, ets_unpack_flash_code
20:57:23.249 -> ets_main.c 
```

Do you know what `ets_unpack_flash_code` means?

Erriez, 2020-10-12:
1) bad PSU, 
Excluded: Same problem with external 3V3 power supply and capacitors.

2) bad flash, 
Excluded: Flashing works reliable.

3) auto-program circuit engaging at boot, 
Excluded with external FTD232 programmer.

4) whatever pulling up MISO might be.
Excluded: I tried a 47k pull-up on D6 MISO

5) Unknown yet...

Erriez, 2020-10-12:
I've successfully removed the shield and measured directly on the `nRST` pin as you suggested.

[image]

As you can see, two components are missing. (No idea if they are needed). I also noticed that pull-up resistor R2 on nRST is 5k6 instead of 12k! I see the evidence on the scope on pin 32 nRST:

[image]

This is the first time I'm convinced that nRST is not pulled low enough after `ESP.deepSleep()`.

Hmm... I'm thinking how to replace this tiny SMD...

nikolz0, 2020-10-12:
> @nikolz0 вы знаете, что `flash read err, ets_unpack_flash_code`это значит?
Hi,
This happens for two reasons.
1) faulty flash;
2) ESP in sleep mode. To get the ESP out of sleep mode, I write using the nodemcuflasher blank.bin, address 0x0000.
        
When writing esp_init......bin, 0x3fc000 should be used instead of 0x1fc000.

nikolz0, 2020-10-12:
> Я успешно снял щит и измерил прямо на `nRST`булавке, как вы и предлагали.
> 
> [image]
> 
> Как вы можете видеть, отсутствуют два компонента. (Понятия не имею, нужны ли они). Я также заметил, что подтягивающий резистор R2 на nRST равен 5k6 вместо 12k! Я вижу доказательства на прицеле на штыре 32 nRST:
> 
> [image]
> 
> Это первый раз, когда я убежден, что nRST не тянется достаточно низко после `ESP.deepSleep()`этого .
> 
> Хмм... Я думаю, как заменить этот крошечный SMD...
Hi,
what will be the pulse if you put a diode between RST and GPIO16?

Tech-TX, 2020-10-12:
I think your reset is fine... it goes HIGH when the chip acknowledges the RST and releases the low on GPIO16.  As I'd mentioned elsewhere, I can't track inside the reset detection code, but that 'release GPIO16' could be either in the RTC hardware or the reset logic in the boot code.  In either case, it's successfully come out of Deep Sleep.

If you're getting the "rst cause:2, boot mode:(3,6)" boot message but not the following LOAD message, then I'm in agreement with nikolz0: it might think it's getting a LOW on GPIO0 and is going into FLASH mode, and not RUN.  Check the level on GPIO0 (and maybe GPIO2) when it gets that reset pulse, if you would?  I doubt GPIO15 is pulling up high, and also doubt TX being pulled LOW.

Another thing you can try is to kill the modem during the boot.  I have some code [here](https://github.com/esp8266/Arduino/issues/7055#issuecomment-581183276) that'll minimize boot current to 35mA average, 54mA peak.  That could have an effect two different ways on the issue you're seeing, although it's less likely to help if you never get the LOAD message as the RF_PRE_INIT() is already after it's decided to load/flash/selftest.

You could still have funky flash.  There's some flash detection (in boot?) that tries to figure out how to proceed (DOUT, DIO, QOUT, QIO) and it's possible that one of those mysterious writes of nikolz0' to the RTC registers stops that or locks it in one of the workable flash modes.  It'd take me quite a while to test each of them to see if I can detect what they're doing to the CPU, and some of those writes I'd never guess.  If you have manufacturer ID = 0x0E like the other gentleman, then I have NO clue what that chip is.
esptool.py -p /dev/ttyUSB0 flash_id  (linux)
esptool.py -p COMx flash_id (Windows, where 'x' is your COM port)
(has to be in programming mode, of course!)

Erriez, 2020-10-13:
@nikolz0

> ESP in sleep mode. To get the ESP out of sleep mode, I write using the nodemcuflasher blank.bin, address 0x0000.

Oh, that makes sense. I've flashed with https://github.com/nodemcu/nodemcu-flasher

testnk_SDK_305.zip from https://github.com/esp8266/Arduino/issues/6318#issuecomment-706845183
* 0x00000 eagle.flash.bin
* 0x10000 eagle.irom0text.bin

```
8:04:17.410 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
18:04:17.410 -> 
18:04:17.410 -> load 0x40100000, len 31216, room 16 
18:04:17.443 -> tail 0
18:04:17.443 -> chksum 0xe9
18:04:17.443 -> load 0x3ffe8000, len 1288, room 8 
18:04:17.443 -> tail 0
18:04:17.443 -> chksum 0x14
18:04:17.443 -> load 0x3ffe8510, len 1508, room 8 
18:04:17.477 -> tail 12
18:04:17.477 -> chksum 0x2b
18:04:17.477 -> csum 0x2b
18:04:17.477 -> V2
18:04:17.477 -> Mo
18:04:17.510 -> ⸮rf cal sector: 1019
18:04:17.510 -> freq trace enable 0
18:04:17.543 -> rf[112] : 00
18:04:17.543 -> rf[113] : 00
18:04:17.543 -> rf[114] : 01
18:04:17.543 -> 
18:04:17.543 -> SDK ver: 3.0.5-dev(52383f9) compiled @ Jul  8 2020 16:00:05
18:04:17.543 -> phy ver: 1156_0, pp ver: 10.2
18:04:17.543 -> 
18:04:17.543 -> testnk
18:04:17.543 -> mode : softAP(ca:2b:96:2f:8e:c6)
18:04:17.543 -> add if1
18:04:17.543 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
18:04:17.576 -> bcn 100
18:04:17.576 -> enter deep sleep
⸮⸮
18:04:27.311 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
18:04:27.311 -> 
```

No further output with a wire between RST and D0.

> what will be the pulse if you put a diode between RST and GPIO16?

The pulse length is much longer now (with 1ms per division), measured on ESP8266EX pin 32 `nRST` with a 1N4148 between `RST` and `D0`:

[image]

> According to your photo it turns out that 470 ohms is enabled between the nRST pin and the external RST pin?

Yes, a 470 Ohm resistor is located between RST pin and nRST.

> If this is the case, then connect RST and GPIO16 should be a conductor, not a resistor.

Correct.

Erriez, 2020-10-13:
@Tech-TX

> In either case, it's successfully come out of Deep Sleep.

Yes, it wakes from deepsleep only once, but does not start the sketch.

> Check the level on GPIO0 (and maybe GPIO2) when it gets that reset pulse, if you would? I doubt GPIO15 is pulling up high, and also doubt TX being pulled LOW. If it's in FLASH mode, you should be able to query it with esptool (see below).

The Wemos D1 Mini has a 10k pull-up resistor on GPIO0 (D3) and GPIO2 (D4).

```c++
#include <user_interface.h>

#define _R (uint32 *)PERIPHS_RTC_BASEADDR
#define valRTC 0xFE000000  // normal value during CONT (modem turned on)

RF_PRE_INIT() {
  *(_R + 4) = 0;  // value of RTC_COUNTER for wakeup from light/deep-sleep (sleep the modem)
  system_phy_set_powerup_option(2);  // stop the RFCAL at boot
  wifi_set_opmode_current(NULL_MODE);  // set Wi-Fi to unconfigured, don't save to flash
  wifi_fpm_set_sleep_type(MODEM_SLEEP_T);  // set the sleep type to Forced Modem Sleep
  // moving the two SDK commands up here saves another 670us @ 8 mA of boot current :-)
}

void preinit() {
  *(_R + 4) = 0;  // sleep the modem again after another initialize attempt before preinit()
  wifi_fpm_open();  // enable Forced Modem Sleep, not in RF_PRE_INIT or it doesn't sleep
  wifi_fpm_do_sleep(0xFFFFFFF);  // enter Modem Sleep mode, in RF_PRE_INIT causes WDT
  // won't go into Forced Modem Sleep until it sees the delay() in setup()
}

void setup() {
  Serial.begin(74880);
  Serial.println("Wake");
   *(_R + 4) = valRTC;  // Force Modem Sleep engaged, set the modem back to 'normal' briefly
  delay(1000);  // and sleep the modem at 18mA nominal substrate current
  ESP.deepSleep(2000000);
}

void loop() {
}
```

```
20:34:57.507 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:34:57.507 -> 
20:34:57.507 -> load 0x4010f000, len 3584, room 16 
20:34:57.507 -> tail 0
20:34:57.507 -> chksum 0xb0
20:34:57.507 -> csum 0xb0
20:34:57.541 -> v2843a5ac
20:34:57.541 -> ~ld
20:35:21.800 -> 
20:35:21.800 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:35:21.800 -> 

===> No output
```

Chip ID:
```
$ ./esptool.py -p /dev/ttyUSB0 chip_id
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: c8:2b:96:2f:8e:c6
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x002f8ec6
Hard resetting via RTS pin...
```

> (laughs) Either you're an old fart like me or an RF geek. Most folks today have never SEEN a germanium.

Haha, I know what you mean. I got a lot of old components from my father. But he never ordered Schottky's and asked me: "What that?" Lol. Yes, germanium is a seldom article, but I can also hook-up an old tube if you like. It's time for me to order some "new" stuff. :-)

> It SHOULD be able to handle 300mA.

Good to know this.

> What I was describing was the Espressif logo on the RF can. 

Interesting. Where do you purchase your working ESP's?

Here are all signals at once:

[image]

If you want to open the logic analyzer file, you can download the free tool from
https://ideas.saleae.com/f/changelog/

[ESP8266WemosD1MiniDeepSleep.zip](https://github.com/esp8266/Arduino/files/5373471/ESP8266WemosD1MiniDeepSleep.zip)

What I see is a constant clock on GPIO0 (D3). Is that normal?

Tech-TX, 2020-10-14:
I'm probably as old as your dad or older: I remember dinosaurs. :D

If you see a clock on GPIO0 then it's likely stuck in test mode.  It thinks it saw a LOW on TXD0 during boot.  You're the first person mentioning that GPIO0 is toggling.

What I meant about the esptool test was while it was in 'zombie mode'.  If it's stuck in 'flash programming' mode then esptool should be able to query the flash.  If it's normally come out of Deep Sleep or it's doing something else weird then esptool can't communicate with it, only in flash mode.  However, with GPIO0 toggling, it's not in flash mode.

You ran the chip_id, I was hoping for flash_id to see which manufacturer of flash.  However, with the RF shield removed, you can read it directly.  What's on the flash chip? (just curious, in case it matters)

nikolz0, 2020-10-14:
@Erriez 
Hi, 
1) testnk without SDK works fine, but testNK with SDK only once? Right?
2) What is the voltage level on GPIO0 when testNK with SDK doesn't Wake up ?
3) If possible, check the operation of the tests not on Wemos, but on ESP-12. Or disable the control scheme on Wemos.

Erriez, 2020-10-14:
@Tech-TX @nikolz0, First I have to say thank you for your support so far. It is an interesting investigation and I appreciate your help.

> If you see a clock on GPIO0 then it's likely stuck in test mode.

I see this for the first time, but it looks like normal:
https://bbs.espressif.com/viewtopic.php?t=1485

> GPIO0 outputs the clock frequency of the external cystal(e.g. 26MHz) by default, which cannot be modified.
> Even you can modify the functions of GPIO0 in the codes, however, before programm starts to run, GPIO0 still outputs the clock frequency of the external cystal.

The clock on GPIO0 is around 1MHz.

> It thinks it saw a LOW on TXD0 during boot. 

I've added a 1k pull-up to TXD, but does not help.

> You're the first person mentioning that GPIO0 is toggling.

I don't think so when I search on internet. :-) I did not notice this before.

> However, with GPIO0 toggling, it's not in flash mode.

Confirmed:

* GPIO0 D3 low after reset: `ets Jan  8 2013,rst cause:2, boot mode:(1,6)`, no clock on GPIO0 (shorted)
* GPIO0 D3 high after reset: `ets Jan  8 2013,rst cause:2, boot mode:(3,6)`, 1MHz clock on GPIO0

> What I meant about the esptool test was while it was in 'zombie mode'. If it's stuck in 'flash programming' mode then esptool should be able to query the flash.

The esptool does not respond when RST and GPIO16 are connected together. Disconnecting displays:

```
$ ./esptool.py -p /dev/ttyUSB0 flash_id
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: c8:2b:96:2f:8e:c6
Uploading stub...
Running stub...
Stub running...
Manufacturer: 0b
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

Manufacturer 0b does not sound good. From the same seller I received in the same batch another ESP8266EX D1 Pro Mini  DOA which also contains damaged 16MB flash with chip type: 25Q12BFVG 1634.

> What's on the flash chip? (just curious, in case it matters)

XT26F32B 1824X9GB:
* Company: `XTX`
* Product family: `25F =2.7~3.6V Serial Flash Memory`
* Product density: `32B = 32M bit`

[Datasheet](https://datasheet.lcsc.com/szlcsc/2005251035_XTX-XT25F32BSOIGU-S_C558851.pdf)

> I got my D1 Mini boards from here.

Oh I did not expect E-Bay. I always thought that they deliver the same low quality as Aliexpress. You're lucky that they work, but you never know what's delivered tomorrow...

Today I've no other ideas.

> 1. testnk without SDK works fine, but testNK with SDK only once? Right?

Yes, your first [sketch](https://github.com/esp8266/Arduino/issues/6318#issuecomment-706560461) is the only one which wakes correctly after deepsleep. The others with SDK does not wake.

> 2. What is the voltage level on GPIO0 when testNK with SDK doesn't Wake up? 

Flash tool settings (for my reference):

[image]

Does not wake, measured on the RST pin (see Vmax1 = 3.751V and Vmin1 is a spike -439mV):

[image]

> 3. If possible, check the operation of the tests not on Wemos, but on ESP-12. Or disable the control scheme on Wemos.

Working ESP8266 bare board on RST pin, a much better reset pulse (see Vmax1 = 3.761V and Vmin1= 39mV):

[image]

Note: The broken Wemos contains an internal 5k6 pull-up resistor on the RST pin and the IA Tinker schematic shows 12k.

Do you recommend me to replace this incorrect resistor?

Tech-TX, 2020-10-15:
>testnk without SDK works fine, but testNK with SDK only once? Right?
Yes, your first sketch is the only one which wakes correctly after deepsleep. The others with SDK does not wake.

That means something in the SDK (boot or hand-over) is going awry.

re: the clock on GPIO0, I misunderstood.  The output on GPIO0 is normal, but it's brief.  I'd thought you said it was doing it continuously.  You don't want to short GPIO0 permanently low or high, a 10-12K pull-up and grounded until after RST for programming / flash upload.

I don't know anything about the XTX chips, but if they run code without the Deep Sleep it's less likely to be that.  This is more about the reset detection, I think.  Connecting a wire from GPIO16 > RST should have cured that, so we're still on the hunt.

I doubt the 5.6K pullup on RST will cause problems, but it will make the rising time shorter.  Can you see what value pullup they have on the CH_PD pin?  You can measure it with an ohmmeter to 3.3V (easiest).  I saw something in one of the databooks on RST / CH_PD sequencing which could conceivably cause your problem, although I'd expect a boot message with 'ets_main.c' in that case.

Those soft edges on the RST pulse with your D1 Mini are due to a capacitor to GND on the RST line.  That's usually both OK and preferred, as it reduces noise on RST.  A bare -12F module doesn't have the ~ .1uF to 1uF cap.  I'm not sure if the ESP-12S has the cap, although it has pullups / pulldowns, same as the other 'S' version modules.

WinBond W25Q12**8**FV is probably the most reliable Flash chip, but I'm not at all surprised if someone re-labeled a cheap flash with that number.  WinBond will have a Manufacturer ID of 0xEF, but if it's dead you can't read the ID (maybe, maybe not).

nikolz0, 2020-10-15:
@Erriez 
Hi, I propose to conduct the following experiment.
Upload the testnk c SDK to Wemos.
Check the operation of Wemos and the state of GPIO16.
Then remove the two s8050 transistors (U2 umh3n chip) from the Board.
https://esp8266.ru/easyeda-quick-start/wemos/
After that, wemos will always be in working mode.
You can set two buttons (GPIO0 and RST) to control the Wemos mode.
Check the operation of Wemos and the state of GPIO16.

nikolz0, 2020-10-15:
Hi,
This is a testnk with the SDK and WiFi turned off.
[testnk_SDK_Wifi_off.zip](https://github.com/esp8266/Arduino/files/5386194/testnk_SDK_Wifi_off.zip)

Erriez, 2020-10-15:
@Tech-TX 

> Can you see what value pullup they have on the CH_PD pin?

Yes, there is a 10k on the WeMos PCB between the ESP8266 PCB and 3V3, so that looks good.

> WinBond W25Q128FV is probably the most reliable Flash chip, but I'm not at all surprised if someone re-labeled a cheap flash with that number.

This chip is a good candidate for chip pirates...

> WinBond will have a Manufacturer ID of 0xEF, but if it's dead you can't read the ID (maybe, maybe not).

The esptool cannot read the chip ID on that Pro mini board, so I'm sure that there is a chip or PCB problem. I bought it for 16MB flash instead of 4MB to store sensor samples when WiFi is not available. I consider to remove that chip and wire it to an UNO to check if this chip is alive or not. But that's for later.

Let's check Nikolz0 output which is interesting...

Erriez, 2020-10-15:
@nikolz0 

> Here is a testnk with the SDK and with my deepsleep function.

Ok, I've flashed `testnkf_SDK.zip` from https://github.com/esp8266/Arduino/issues/6318#issuecomment-708884215

> I propose to conduct the following experiment.
> Upload the testnk c SDK to Wemos.
> Check the operation of Wemos and the state of GPIO16.

The output with a wire between RST and D0:

```
20:51:15.814 -> scandone
20:51:15.814 -> no alive found, reconnect after 1s
20:51:15.914 -> reconnect
20:51:17.179 -> sleep=4291035
20:51:26.378 -> 
20:51:26.378 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:51:26.378 -> 
20:51:26.444 -> load 0x40100000, len 31404, room 16 
20:51:26.478 -> tail 12
20:51:26.478 -> chksum 0xcb
20:51:26.478 -> ho 0 tail 12 room 4
20:51:26.478 -> load 0x3ffe8000, len 1288, room 12 
20:51:26.478 -> tail 12
20:51:26.478 -> chksum 0x2e
20:51:26.478 -> ho 0 tail 12 room 4
20:51:26.478 -> load 0x3ffe8510, len 1516, room 12 
20:51:26.511 -> tail 0
20:51:26.511 -> chksum 0xb2
20:51:26.511 -> csum 0xb2
20:51:26.511 -> V2
20:51:26.511 -> Mo
20:51:26.511 -> ⸮rf cal sector: 1019
20:51:26.511 -> freq trace enable 0
20:51:26.511 -> rf[112] : 00
20:51:26.511 -> rf[113] : 00
20:51:26.544 -> rf[114] : 01
20:51:26.544 -> 
20:51:26.544 -> SDK ver: 3.0.5-dev(52383f9) compiled @ Jul  8 2020 16:00:05
20:51:26.544 -> phy ver: 1156_0, pp ver: 10.2
20:51:26.544 -> 
20:51:26.544 -> testnk
20:51:26.544 -> SDK ver:3.0.5-dev(52383f9),T1=92096,T2=92392,vdd33=0
20:51:26.544 -> mode : sta(c8:2b:96:2f:8e:c6)
20:51:26.544 -> add if0
20:51:29.372 -> scandone
20:51:29.372 -> no alive found, reconnect after 1s
20:51:29.472 -> reconnect
20:51:30.738 -> sleep=4312955
20:51:39.893 -> 
20:51:39.893 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:51:39.893 -> 
20:51:39.960 -> load 0x40100000, len 31404, room 16 
20:51:39.993 -> tail 12
20:51:39.993 -> chksum 0xcb
20:51:39.993 -> ho 0 tail 12 room 4
20:51:39.993 -> load 0x3ffe8000, len 1288, room 12 
20:51:39.993 -> tail 12
20:51:39.993 -> chksum 0x2e
20:51:39.993 -> ho 0 tail 12 room 4
20:51:39.993 -> load 0x3ffe8510, len 1516, room 12 
20:51:40.026 -> tail 0
20:51:40.026 -> chksum 0xb2
20:51:40.026 -> csum 0xb2
20:51:40.026 -> V2
20:51:40.026 -> Mo
20:51:40.026 -> ⸮rf cal sector: 1019
20:51:40.026 -> freq trace enable 0
20:51:40.026 -> rf[112] : 00
20:51:40.026 -> rf[113] : 00
20:51:40.059 -> rf[114] : 01
20:51:40.059 -> 
20:51:40.059 -> SDK ver: 3.0.5-dev(52383f9) compiled @ Jul  8 2020 16:00:05
20:51:40.059 -> phy ver: 1156_0, pp ver: 10.2
20:51:40.059 -> 
20:51:40.059 -> testnk
20:51:40.059 -> SDK ver:3.0.5-dev(52383f9),T1=92044,T2=92339,vdd33=0
20:51:40.059 -> mode : sta(c8:2b:96:2f:8e:c6)
20:51:40.059 -> add if0
20:51:42.886 -> scandone
20:51:42.886 -> no alive found, reconnect after 1s
20:51:42.985 -> reconnect
20:51:44.249 -> sleep=4312903
20:51:53.398 -> 
20:51:53.398 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:51:53.398 -> 
20:51:53.465 -> load 0x40100000, len 31404, room 16 
20:51:53.499 -> tail 12
20:51:53.499 -> chksum 0xcb
20:51:53.499 -> ho 0 tail 12 room 4
20:51:53.499 -> load 0x3ffe8000, len 1288, room 12 
20:51:53.499 -> tail 12
20:51:53.499 -> chksum 0x2e
20:51:53.499 -> ho 0 tail 12 room 4
20:51:53.499 -> load 0x3ffe8510, len 1516, room 12 
20:51:53.532 -> tail 0
20:51:53.532 -> chksum 0xb2
20:51:53.532 -> csum 0xb2
20:51:53.532 -> V2
20:51:53.532 -> Mo
20:51:53.532 -> ⸮rf cal sector: 1019
20:51:53.532 -> freq trace enable 0
20:51:53.532 -> rf[112] : 00
20:51:53.532 -> rf[113] : 00
20:51:53.565 -> rf[114] : 01
20:51:53.565 -> 
20:51:53.565 -> SDK ver: 3.0.5-dev(52383f9) compiled @ Jul  8 2020 16:00:05
20:51:53.565 -> phy ver: 1156_0, pp ver: 10.2
20:51:53.565 -> 
20:51:53.565 -> testnk
20:51:53.565 -> SDK ver:3.0.5-dev(52383f9),T1=92099,T2=92395,vdd33=0
20:51:53.565 -> mode : sta(c8:2b:96:2f:8e:c6)
20:51:53.565 -> add if0
20:51:56.395 -> scandone
20:51:56.395 -> no alive found, reconnect after 1s
20:51:56.495 -> reconnect
20:51:57.762 -> sleep=4312958
...
```

I've flashed `testnk_SDK_Wifi_off.zip` as well from https://github.com/esp8266/Arduino/issues/6318#issuecomment-709420065

But unfortunately, this one does not wake with the same wire between RST and D0:

```
20:46:55.988 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:46:55.988 -> 
20:46:56.088 -> load 0x40100000, len 31400, room 16 
20:46:56.121 -> tail 8
20:46:56.121 -> chksum 0xeb
20:46:56.121 -> load 0x3ffe8000, len 1288, room 0 
20:46:56.121 -> tail 8
20:46:56.121 -> chksum 0x06
20:46:56.121 -> load 0x3ffe8510, len 1516, room 0 
20:46:56.121 -> tail 12
20:46:56.121 -> chksum 0x96
20:46:56.121 -> csum 0x96
20:46:56.154 -> V2
20:46:56.154 -> Mo
20:46:56.188 -> ⸮rf cal sector: 1019
20:46:56.188 -> freq trace enable 0
20:46:56.188 -> rf[112] : 00
20:46:56.188 -> rf[113] : 00
20:46:56.188 -> rf[114] : 01
20:46:56.188 -> 
20:46:56.188 -> SDK ver: 3.0.5-dev(52383f9) compiled @ Jul  8 2020 16:00:05
20:46:56.221 -> phy ver: 1156_0, pp ver: 10.2
20:46:56.221 -> 
20:46:56.221 -> testnk
20:46:56.221 -> enter deep sleep
```

Note: I've flashed both sketches twice after each other to double check that I don't make any mistakes here. Both with the same output.

I think there is a little break trough as the first binary wakes. Is this what you expected. :+1: for this.

nikolz0, 2020-10-15:
@Erriez,
Hi,
did I understand you correctly that the tests with my function works correctly?

nikolz0, 2020-10-16:
Hi,
I think the problem is the speed of the flash chip.
The size of the uploaded code for my test without the SDK is significantly smaller than for the test with the SDK.

Try setting DIO mode instead of QIO when programming the ESP.

Erriez, 2020-10-16:
@nikolz0 

> did I understand you correctly that the tests with my function works correctly, and tests with a function from the SDK stop?

I'm not sure what you mean, but I loaded your binaries on the non-working board:

testnkf_SDK.zip https://github.com/esp8266/Arduino/issues/6318#issuecomment-708884215 -> Wake-up works as expected. I hope you have more information about the contents of this binary.
testnk_SDK_Wifi_off.zip https://github.com/esp8266/Arduino/issues/6318#issuecomment-709420065 -> Wake-up does not work. 

QIO or DIO does not matter. I assume this setting is only used when writing the binaries to the flash chip. Writing to flash, booting from flash after power-on or pressing the reset button works reliable.

nikolz0, 2020-10-16:
Hi,
testnkf_SDK.zip,testnk.zip - this is a test with my deepsleep function.
testnk_SDK_Wifi_off.zip,testnk_SDK_305.zip - these are tests with the deepsleep function from the SDK.
Which of them work correctly?

nikolz0, 2020-10-16:
Hi,
I added the DeepSleepNK function to libespnow.a, which should be replaced in the lib SDK folder.
[libespnow.zip](https://github.com/esp8266/Arduino/files/5393817/libespnow.zip)

function description:
void DeepSleepNK(uint32 );
...
#define time1 1.32*10000000 //sleep 10 seconds
....
DeepSleepNK(time1); //function call

[99 more replies]