esp8266/Arduino #6322: SPIFFS setConfig

LeisureLadi, 2019-07-21:
----------------------------- Delete below -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

### Problem Description

Description of SPIFFS file system is wrong (https://github.com/esp8266/Arduino/blob/master/doc/filesystem.rst):

[image]

"SPIFFSConfig" has to be replaced by "FSConfig", otherwise the compiler will throw the error:

'SPIFFSConfig' was not declared in this scope

Please change the description or edit definition in FS.h

earlephilhower, 2019-07-21:
No, that won't work and the existing example is right.  The code compiles fine.  Maybe you're missing a `#include <FS.h>` in your code?  It's required to do any filesystem work:

````
#include <FS.h>
void setup() {
  SPIFFSConfig cfg;
  cfg.setAutoFormat(false);
  SPIFFS.setConfig(cfg);
}
void loop() {
}
````
gives
````
Sketch uses 296503 bytes (28%) of program storage space. Maximum is 1044464 bytes.
Global variables use 27528 bytes (33%) of dynamic memory, leaving 54392 bytes for local variables. Maximum is 81920 bytes.
````

LeisureLadi, 2019-07-21:
No. Clean install of core 2.5.2 on Windows 7 and Arduino 1.8.9 throws this message when trying to compile your code:
`#include <FS.h>

void setup() {
SPIFFSConfig cfg;
cfg.setAutoFormat(false);
SPIFFS.setConfig(cfg);
SPIFFS.begin();

}

void loop() {
  // put your main code here, to run repeatedly:

}`

Arduino: 1.8.9 (Windows 7), Board: "LOLIN(WEMOS) D1 R2 & mini, 160 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (2M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"
C:\Users\Ladi\Documents\Arduino\SPIFFStest\SPIFFStest.ino: In function 'void setup()':
SPIFFStest:4:1: error: 'SPIFFSConfig' was not declared in this scope
 SPIFFSConfig cfg;
 ^
C:\Users\Ladi\Documents\Arduino\SPIFFStest\SPIFFStest.ino:4:1: note: suggested alternative:
In file included from C:\Users\Ladi\Documents\Arduino\SPIFFStest\SPIFFStest.ino:1:0:
C:\Users\Ladi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/FS.h:167:7: note:   'fs::SPIFFSConfig'

class SPIFFSConfig : public FSConfig
       ^
SPIFFStest:4:14: error: expected ';' before 'cfg'
SPIFFSConfig cfg;
              ^

SPIFFStest:5:1: error: 'cfg' was not declared in this scope

 cfg.setAutoFormat(false);
 ^

exit status 1
'SPIFFSConfig' was not declared in this scope

However, replacing SPIFFSConfig with FSConfig, compiles well. Anyway, I hope this is going to help others running into the same issue.

earlephilhower, 2019-07-21:
Very odd.  I will try a 2.5.2 clean install.  In any case, the current GIT head is fine.

The problem with passing in a FSConfig is that it doesn't have the right header and I believe the code will return an error and ignore any settings therein.

So, unfortunately, your suggestion will compile fine but won't actually work. 

I'll update after I get the 2.5.2 toolchain installed/tested.

earlephilhower, 2019-07-21:
Yup, you're right that the 2.5.2 release has a bug but it's not that it doesn't define SPIFFSConfig, its that it doesn't have a `using fs::SPIFFSConfig;` in `<FS.h>`

The correct solution, on 2.5.2 only, is to use `fs::SPIFFSConfig cfg` instead.  Only that change will work properly.

GIT head is fine, so this has already been fixed.   Don't pass in a FSConfig.  It will compile, but if you check the return code you will find that it returns `false` and fails.  Only use `fs::SPIFFSConfig`.

````
#include <FS.h>
void setup() {
  fs::SPIFFSConfig cfg;
  cfg.setAutoFormat(false);
  SPIFFS.setConfig(cfg);
}
void loop() {
}
````

earlephilhower, 2019-07-21:
Actually, darn it, GIT head also seems wonky.  Good catch and thanks for following up!

PR #6324 (2019-07-21): Add `using fs::SPIFFSConfig` to FS.h
The SPIFFS config object was defined in FS.h in its own namespace, but
is not made easily available like other SPIFFS and FS objects because of
a missing `using` statement.  Add it in FS.h

Fixes #6322
commit 93ef2e29 (2019-07-21): Add `using fs::SPIFFSConfig` to FS.h (#6324)

The SPIFFS config object was defined in FS.h in its own namespace, but
is not made easily available like other SPIFFS and FS objects because of
a missing `using` statement.  Add it in FS.h

Fixes #6322
closed 2019-07-21 by earlephilhower