esp8266/Arduino #6323: espota.py passes wrong .bin file

thangktran, 2019-07-21:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:                 [ESP-12]
- Core Version:            [2.5.2]
- Development Env:     [Arduino IDE]
- Operating System:     [Ubuntu]

### Problem Description
I tried to upload a sketch using BasicOTA.ino example with bin signing and have put `private.key` and `public.key` in the same folder as the project folder.
The bin was signed, however, when `espota.py` is called from `Arduino IDE`, the standard bin `*.bin` is selected instead of the signed `*.bin.signed` one.

I would like to do a PR for this Issue as well if someone could guide me where to look into.

Thank you.

### Debug Messages

```
Creating BIN file "/tmp/arduino_build_392120/sketch_jul09a.ino.bin" using "/tmp/arduino_build_392120/sketch_jul09a.ino.elf"
/home/ubuntu/.arduino15/packages/esp8266/tools/python/3.7.2-post1/python /home/ubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/signing.py --mode sign --privatekey /home/ubuntu/Desktop/sketch_jul09a/private.key --bin /tmp/arduino_build_392120/sketch_jul09a.ino.bin --out /tmp/arduino_build_392120/sketch_jul09a.ino.bin.signed
Signed binary: /tmp/arduino_build_392120/sketch_jul09a.ino.bin.signed
Using library ESP8266WiFi at version 1.0 in folder: /home/ubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi 
Using library ESP8266mDNS at version 1.2 in folder: /home/ubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266mDNS 
Using library ArduinoOTA at version 1.0 in folder: /home/ubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ArduinoOTA 
Using library NTPClient at version 3.1.0 in folder: /home/ubuntu/Arduino/libraries/NTPClient 
/home/ubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-size -A /tmp/arduino_build_392120/sketch_jul09a.ino.elf
Sketch uses 347800 bytes (33%) of program storage space. Maximum is 1044464 bytes.
Global variables use 29312 bytes (35%) of dynamic memory, leaving 52608 bytes for local variables. Maximum is 81920 bytes.
/home/ubuntu/.arduino15/packages/esp8266/tools/python/3.7.2-post1/python /home/ubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/espota.py -i 192.168.0.101 -p 43434 --auth=********** -f /tmp/arduino_build_392120/sketch_jul09a.ino.bin
```

earlephilhower, 2019-07-21:
https://www.freecodecamp.org/news/a-simple-git-guide-and-cheat-sheet-for-open-source-contributors/ summarizes nicely how PRs are done.

thangktran, 2019-07-21:
@earlephilhower Thank you for your answer.
What i meant was `I would like to do a PR for this Issue as well if someone could guide me to which module/function/file i should look into and suggestions how should it be fixed`

earlephilhower, 2019-07-21:
It's probably being set in the espota.py command line from platform.txt.  Inserting logic there to swap between filenames may be a big ugly, good luck!

thangktran, 2019-07-21:
> It's probably being set in the espota.py command line from platform.txt. Inserting logic there to swap between filenames may be a big ugly, good luck!

@earlephilhower Thank you for your hint.
What is your opinion for the following solution:
remove the file extension `.bin` from `-f "{build.path}/{build.project_name}.bin"
https://github.com/esp8266/Arduino/blob/ac25f3276fc277391b883dd191c58fead6479c92/platform.txt#L144
==>
`tools.esptool.upload.network_pattern="{network_cmd}" "{runtime.platform.path}/tools/espota.py" -i "{serial.port}" -p "{network.port}" "--auth={network.password}" -f "{build.path}/{build.project_name}"

then we let `espota.py` decide to use `*.bin` or `*.bin.signed`.
Since the file `*.bin.signed` only exists when Signed Updates is used, this file should be uploaded.
Otherwise `*.bin` should be used instead.

devyte, 2019-07-23:
The solution sounds good to me. Please make a PR for review and further discussion.

thangktran, 2019-07-23:
> The solution sounds good to me. Please make a PR for review and further discussion.

@devyte Thank you for your feedback. I'll be working on this issue

closed 2019-07-23 by devyte