esp8266/Arduino #6330: ESP01 with Board Version > 2.42 does not connect to SMTP Server

BDCowboy, 2019-07-22:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-01]
- Core Version:			[Version 2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[1MB]
- lwip Variant:   [|v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Connecting to an SMTP Server using the WIFIClientSecure object fails with a  simple email test program with any version of the board manager > than 2.4.2
Also, WIFIClient also fails when used under the same circumstances.

WiFiClientSecure client;

  if (!client.connect(SMTP_SERVER, SMTP_PORT)) {
    error_message = "SMTP could not connect to the mail server";
    return;
  }

the code above fails (client.connect returns false) with any board manager version > 2.4.2. This is using an ESP01.
The above works with 2.4.1 and 2.4.2  WIFI connection to access point is not affected.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

earlephilhower, 2019-07-23:
Without actual sample code, unfortunately, there's not much we can do for you.  Can you give a real example?  BearSSL/AxTLS support TLS, not STARTTLS, so are you sure you're connecting to a system that really had a TLS connection and doesn't expect to be upgraded using the `STARTTLS` message after connecting unencrypted?  See https://www.fastmail.com/help/technical/ssltlsstarttls.html for a high level discussion of the differences.

BDCowboy, 2019-07-23:
Here is the full sketch. I have removed my wifi login info and my gmail login info. I also have an alternative email that connects with WiFiClient and not WiFeClientSecure. In both cases this code works (sends the email) in board versions up to 2.4.2. 
The wifi connection is fine and the http call (which is picked up by a little module with a buzzer attached and another ESP-01) also works.

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <ESP8266HTTPClient.h>
#include <base64.h>

// gmail requires wificlientsecure and alternate email uses wificlient

#pragma region Globals
const char* ssid = "";                           // WIFI network name
const char* password = "";                       // WIFI network password

const int     SMTP_PORT            = 485;
const char*   SMTP_SERVER          = "smtp.gmail.com";
String        error_message = "";
String        Response;

String Senders_Login    = "";
String Senders_Password = "";
String From;

String To, Subject, Message, Login_base64, Passwrd_base64;

// Uncomment & update if you want to use a Static IP 
// (Static IP connection is much faster than DHCP)
//#define IP
//IPAddress ip(192, 168, 1, 6);
//IPAddress gateway(192, 168, 1, 254);
//IPAddress subnet(255, 255, 255, 0);

int LED = LED_BUILTIN;                                        // Choose the pin for the LED 
int inputPin = 3;  //RX                             
int val = 0;                                            // Variable for reading the pin status 
int sendonce = 0;
String chipID;

WiFiClientSecure client;

// Blink the LED 
// Provides a visual display of activity

void blink() {

  for (int j=1; j<=3; j++) {
    // ON
    digitalWrite(LED, LOW);
    delay(100); 
    
    //OFF
    digitalWrite(LED, HIGH);
    delay(100);
  }

}

void setup() {
   Serial.begin(115200);

   Serial.println("Starting up");
   
  // put your setup code here, to run once:
  // prepare LED
  pinMode(LED, OUTPUT);
  digitalWrite(LED, HIGH);  
  
// Enter your own data for who you are sending the emails to 
  To      = "";
  From    = "";
  Subject = "ESP-01 Email Test for Gmail";
  
  // We start by connecting to a WiFi network

  // Set Static IP, if defined above
  #if defined(IP)
      WiFi.config(ip, gateway, subnet);
  #endif

  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }

 chipID = String(ESP.getChipId(), HEX);

  Message = "ESP-01 The motion sensor has been triggered<br />ESP Chip ID:" + chipID; 

  // Blink onboard LED to signify connection to WIFI
  for (int i = 1; i < 5; i++) {
    blink();
  }

        SendMail(From, To, Subject, Message);
         Serial.println(error_message);
         Serial.println("Message Sent");
         
        // now send the beep signal
        // to the esp-01 buzzer
        HTTPClient http;
  
        http.begin("http://192.168.1.17/buzz2"); //HTTP
        int httpCode = http.GET();
        http.end();  

}

void loop() {
  // put your main code here, to run repeatedly:

}

void SendMail(String From, String To, String Subject, String Message) {
  
  if (!client.connect(SMTP_SERVER, SMTP_PORT)) {
    error_message = "SMTP could not connect to the mail server";
    return;
  }
  if (ErrorWhileWaitingForSMTP_Response("220", 500)) {
    error_message = "SMTP Connection Error";
    return;
  }
  client.println("HELO server");
  if (ErrorWhileWaitingForSMTP_Response("250", 500)) {
    error_message = "SMTP Identification error";
    return;
  }
  client.println("AUTH LOGIN");
  WaitSMTPResponse(Response, 500);
  client.println(base64::encode(Senders_Login));
  
  WaitSMTPResponse(Response, 500);
  client.println(base64::encode(Senders_Password));

  if (ErrorWhileWaitingForSMTP_Response("235", 500)) {
    error_message = "SMTP Authorisation error";
    return;
  }

  // Note that mailFrom and recipient (To) do not support
  // multiple email addresses in the string. The sending of the
  // message will fail.
  
  String mailFrom = "MAIL FROM: <" + From + '>';
  client.println(mailFrom);
  WaitSMTPResponse(Response, 500);
  
  String recipient = "RCPT TO: <" + To + '>';
  client.println(recipient);
  WaitSMTPResponse(Response, 500);
  client.println("DATA");
  
  if (ErrorWhileWaitingForSMTP_Response("354", 500)) {
    error_message = "SMTP DATA error";
    return;
  }
  
  client.println("From: <" + String(From) + '>');
  client.println("To: <" + String(To) + '>');
  client.print("Subject: ");
  client.println(String(Subject));
  client.println("Mime-Version: 1.0");
  client.println("Content-Type: text/html; charset=\"UTF-8\"");
  client.println("Content-Transfer-Encoding: 7bit");
  client.println();
  
  String body = "<!DOCTYPE html><html lang=\"en\">" + Message + "</html>";
  client.println(body);
  client.println(".");
  
  if (ErrorWhileWaitingForSMTP_Response("250", 1000)) {
    error_message = "SMTP Message error";
    return;
  }
  
  client.println("QUIT");
  if (ErrorWhileWaitingForSMTP_Response("221", 1000)) {
    error_message = "SMTP QUIT error";
    return;
  }
  
  client.stop();
}

bool ErrorWhileWaitingForSMTP_Response(String Error_Code, int TimeOut) {
  int timer = millis();
  
  while (!client.available()) {
    
    if (millis() > (timer + TimeOut)) {
      error_message = "SMTP Timeout";
      return true;
    }
  }
  
  Response = client.readStringUntil('\n');
  
  if (Response.indexOf(Error_Code) == -1) return true;
  return false;
}

bool WaitSMTPResponse(String Error_Code, int TimeOut) {
  int timer = millis();
  
  while (!client.available()) {
    
    if (millis() > (timer + TimeOut)) {
      error_message = "SMTP Timeout";
      return false;
    }
  }
  Response = client.readStringUntil('\n');
  
  if (Response.indexOf(Error_Code) == -1) return false;
  return true;
}

BDCowboy, 2019-07-23:
I made an error in copying the code. The SMTP port for gmail is 465 not 485. I put the gmail port and server into the code after running with my alternate email (that uses port 587) and typed it in wrong!

earlephilhower, 2019-07-24:
The code never specifies a root certification authority, fingerprint, etc. so BSSL has no way of verifying the authenticity of the server.

Either add a `client.setInsecure()` and let BSSL connect without any verification (i.e. MITM is almost trivial), or add a proper certificate or certstore to the client before connecting.  See the BearSSL_Validation.ino example for the options.

You can also turn on full debugging and BSSL will telly you why it can't connect.  In this case you'd also get an extra warning telling you that it's impossible to connect w/o a verification option.

closed 2019-07-24 by earlephilhower

GurraE, 2019-08-06:
In order to make this work with G-mail  and core 2.5.0, add  the line earlephilhower mentioned, in this section 
 

`int LED = LED_BUILTIN;                                        // Choose the pin for the LED 
int inputPin = 3;  //RX                             
int val = 0;                                            // Variable for reading the pin status 
int sendonce = 0;
String chipID;

WiFiClientSecure client;

client.setInsecure();                       //   <----   **ADD LINE to make it work with core 2.5.0**

// Blink the LED 
// Provides a visual display of activity
` 

// Gurra