esp8266/Arduino #6331: Session Resumption over AWS IoT Core MQTT

grgoncal, 2019-07-23:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.5.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Fedora]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I'm using WiFiClientSecure (BearSSL) to connect to AWS IoT Core using all required certificates and it has been working pretty nicely. 

But recently I added _modem sleep_ to the code, to save battery between transmissions. This causes the client to resend all certificates using too much bandwidth. I really can't sacrifice battery, so I need to resume the connection session. The interval between transmissions is about a minute and I'm using PubSubClient. 

I have tried to change MQTT_KEEPALIVE and MQTT_MAX_PACKET_SIZE inside PubSub. Also changed the CleanSession flag to be able to not request a new session. Did not work.

I also tried to use BearSSL::Session session; to save a session by using the method .setSession(&session). Also did not work.

Does anyone have any idea of how I may be able to achieve this? Is this feature even possible?

earlephilhower, 2019-07-24:
Sorry, without code there's not much that we can do.  SSL sessions need to be supported by the server and the client.  The server is not required to preserve them, especially if there are load balancers along the path which might redirect a new incoming connection to a different VM (i.e. these are not like HTTP sessions with a cookie that might be globally shared among the load balancing backend).

earlephilhower, 2019-11-19:
No MCVE and no response in >5 months, so closing.

closed 2019-11-19 by earlephilhower