esp8266/Arduino #6332: GET requests not work properly

dmdt, 2019-07-23:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12E/NodeMCU v.10
- Core Version:			2.5.2
- Development Env:		Platformio/ArduinoIDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu v1.0
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

When requests running with delay several seconds between on core 2.5.2 the data (received with .getString()) not modifying (with reuse set to true).
When I set reuse false - response just get w/o body. And every second response return -2 HTTP error code. In both situations script on server the same (obviously). And this problem doesn't happen on 2.4.2 core. 
Here #6149 man stuck with the same problem, but no one respond.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>

BearSSL::WiFiClientSecure client;
HTTPClient https;

void setup() {
    Serial.begin(115200);
    WiFi.begin("ssid","pass");
    while(WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }
    Serial.println("\nWifi connected!");
    uint8_t cert[20] = {184,120,14,91,203,255,205,98,26,130,214,144,95,34,236,197,210,208,31,245};
    client.setFingerprint(cert);
    https.setReuse(true);
    https.begin(client, "https://link");
}

int code;
String response;

void loop() {
    code = https.GET();
    Serial.println("Code: " + String(code));
    response = https.getString();
    Serial.println("Body: " + response);
    delay(2000);
}

```

### Debug Messages

##With reuse
```
Code: 200
Body: mode?a=16711680&b=2
Code: 200
Body: mode?a=16711680&b=2
Code: 200
Body: mode?a=16711680&b=2
Code: 200
Body: mode?a=16711680&b=2
@@@Here I change data in database, but messages below are similar to above.
Code: 200
Body: mode?a=16711680&b=2
Code: 200
Body: mode?a=16711680&b=2
Code: 200
Body: mode?a=16711680&b=2
Code: 200
Body: mode?a=16711680&b=2
```
##Without reuse
```
@@@Each other request get failed, and all requests return empty body.
Code: -2
Body:
Code: 200
Body:
Code: -2
Body:
Code: 200
Body:
Code: -2
Body:
Code: 200
Body:
Code: -2
Body:
Code: 200
```

dmdt, 2019-07-24:
I also tested with https.useHTTP10(false), this also had no affect.
And I tried use latest version from git - same problem.

dmdt, 2019-08-04:
So, anyone can tell me about this situation? Is it known bug which will be fixed in next updates, or no one really knows about this?

d-a-v, 2019-08-04:
@dmdt are you able to try PR #6176 ?

dmdt, 2019-08-04:
Tried, this has no affect on behaviour.

dorindabassey, 2019-08-04:
hello there, am also having same issues, am trying to post data to my https://server online and am getting response code: 200 but the payload returns :datanotposted from the server

dorindabassey, 2019-08-04:
`//  FROM HERE!!!
  if (WiFi.status() == WL_CONNECTED) {
    HTTPClient http;

    // Your Domain name with URL path or IP address with path
    // http.begin(serverName);

    // Specify content-type header
    http.addHeader("Content-Type", "application/x-www-form-urlencoded");

    // Prepare your HTTP POST request data
    String httpRequestData = "api_key=" + apiKeyValue + "&name=" + incomingName + "&email=" + incomingEmail + "&userid=" + userId + "&useramount=" + userAmount + "";
    Serial.print("httpRequestData: ");
    Serial.println(httpRequestData);
    int httpsbegincode = http.begin(serverName, "5B FB D1 D4 49 D3 0F A9 C6 40 03 34 BA E0 24 05 AA D2 E2 01");
    Serial.println(httpsbegincode);

    // Send HTTP POST request
    int httpResponseCode = http.POST(httpRequestData);
    String payload = http.getString();
    Serial.println("PAYLOAD");
    Serial.println(payload);

    
    if (httpResponseCode > 0) {
      Serial.print("HTTP Response code: ");
      Serial.println(httpResponseCode);
    }
    else {
      Serial.print("Error code: ");
      Serial.println(httpResponseCode);
    }
    // Free resources''

    http.end();
    Serial.println("connection over");
  }
  else {
    Serial.println("WiFi Disconnected");
  }
`
below is a picture of my serial monitor
[image]

d-a-v, 2019-08-05:
#6176 will fix this

devyte, 2019-11-10:
#6176 is merged. Closing.
If you think the problem is still relevant, please open a new issue and follow the template instructions.

closed 2019-11-10 by devyte