esp8266/Arduino #6336: Leaving.. Hard reseting via RTS pin... on ESP8266 NODEMCU

Saketh-Bobbala, 2019-07-24:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					ESP8266 NODEMCU
- Core Version:			[latest git hash or date]
- Development Env:		Arduino ID
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 		512K
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		ck|
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description
Unable to upload code to nodemcu (Leaving... Resetting via RTS pin)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
#define BLYNK_PRINT Serial

#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "yPCwZ9AtBsjxhbzx2xGGieUodNAofDk";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "TP-Link";
char pass[] = "12345678";

void setup()
{
  // Debug console
  Serial.begin(11520);

  Blynk.begin(auth, ssid, pass);
}

void loop()
{
  Blynk.run();
}
```

### Debug Messages

Sketch uses 279800 bytes (55%) of program storage space. Maximum is 499696 bytes.
Global variables use 28856 bytes (35%) of dynamic memory, leaving 53064 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: cc:50:e3:c7:34:66
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 283952 bytes to 204290...

Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 283952 bytes (204290 compressed) at 0x00000000 in 18.0 seconds (effective 126.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

jandrassy, 2019-07-24:
it is a sucessful upload. the esptool resets the esp8266 after upload

Saketh-Bobbala, 2019-07-24:
Hi ,it is still not working.How long does it take to reset

jandrassy, 2019-07-24:
reset is immediate. how do you know it doesn't work?

Saketh-Bobbala, 2019-07-24:
The NodeMCU is not connecting to my wifi router, I set it to nodemcu and tried again but got the same message .

d-a-v, 2019-07-24:
Check GPIOs are correctly driven ***at boot or reset time***:
- GPIO0 and GPIO2 must be high
- GPIO15 must be low

[reference1](https://arduino-esp8266.readthedocs.io/en/latest/boards.html#minimal)
[reference2](https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process#esp-boot-modes)

jandrassy, 2019-07-24:
do you have some debug prints in `setup()`?

devyte, 2019-07-24:
Closing because I don't see a core issue here.

closed 2019-07-24 by devyte

sakthi123-s, 2019-09-21:
I am also get same error 
Hard resetting via rts pains how to solve it any suggestions

jandrassy, 2019-09-22:
> I am also get same error
> Hard resetting via rts pains how to solve it any suggestions

read the comments above

GonzalezAnguita, 2019-10-26:
After updating the Arduino IDE to 1.8.10 the ESP-01s with [this usb adapter](https://www.amazon.com/gp/product/B07KF119YB/) does not reset after successfully uploading a program using the generic configuration. I've tried with a NodeMCU 0.9 (ESP-12) and that one resets correctly using the board specific configuration.

d-a-v, 2019-10-26:
This adapter has a physical switch for running or programming an esp-01.
What is the exact version of the core and configuration of the menu that allows the board to reset and run right after programming ?

jandrassy, 2019-10-27:
@d-a-v they left the switch in programming mode. the old esptool didn't reset after upload so the sketch did run until next reset

d-a-v, 2019-10-27:
@GonzalezAnguita Are you able to try #6429 ?

DuKings, 2019-11-14:
@Saketh-Bobbala I Had the same issue, try changing the baud value in the serial monitor (do this if you are using Arduino) to the one that is in the code.

jithin-pal, 2019-11-16:
change the connection frequency to 2.4Ghz

VeeruSubbuAmi, 2019-12-10:
This article might help you to to understand anf fix the issue
https://electronicsinnovation.com/hard-resetting-via-rts-pin-fixed-explained/

MrPxx, 2020-02-17:
@Saketh-Bobbala This worked for me. Tools > Reset Method > "no dtr (aka ck)".

Then I got this :
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 277792 bytes (202445 compressed) at 0x00000000 in 18.0 seconds (effective 123.8 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...

rugimike, 2020-03-01:
I solved it this way:
-connect the serial port(GPIO 0) to ground. makes arduino act as a serial port to ESP8266
-run the sketch.
-while it starts connecting, connect the RESET of ESP8266 to ground. then disconnect before connecting... finishes.
-the above resets the ESP8266. remember, all this time, RESET of the arduino is grounded.
-Wait for the code to be uploaded to ESP8266.
-When you see. ' leaving..  hard reseting via RST pin', disconnect the serial port(GPIO 0).
 then connect the RESET of ESP8266 to ground for 1 second and then release it.
 when you release it, the program will start running. you will see this in the serial monitor.(make sure you 
 select the correct baud rate on the serial monitor)

-Above should work well

rugimike, 2020-03-01:
> @Saketh-Bobbala This worked for me. Tools > Reset Method > "no dtr (aka ck)".
>
> Then I got this :
> Writing at 0x00000000... (7 %)
> Writing at 0x00004000... (15 %)
> Writing at 0x00008000... (23 %) [...]

jandrassy, 2020-03-01:
@rugimike, if your board doesn't have autoreset feature, you don't need to make this adrenaline action. if io 0 is low at reset the esp8266 starts in flashing mode a stays in it until next reset. then at reset again the io 0 level counts

rugimike, 2020-03-01:
Hi. Its true.(even my board does not have autoreset). what you do after
seeing 'leaving..  hard/soft resetting via RTS pin..'  is to.
1. disconnect the GPIO 0 pin.
2.  connect the RESET of the ESP8266 to ground, for just a second and then
disconnect it.
 please see below.
[image: image.png]

rugimike, 2020-03-01:
[image: image.png]

Soubatan, 2020-03-29:
> [image: image.png]
> […](#)
> -- Best Regards Michael Rugi Njoroge

Thanks.great

ahmettsdvn, 2020-06-07:
> I solved it this way:
> -connect the serial port(GPIO 0) to ground. makes arduino act as a serial port to ESP8266
> -run the sketch.
> -while it starts connecting, connect the RESET of ESP8266 to ground. [...]

Thanks...

VeeruSubbuAmi, 2020-09-14:
You can find your answer here.
https://youtu.be/McjIEpc26Do