esp8266/Arduino #6344: WiFi.config with gateway in different subnet failed

TheChief79, 2019-07-25:
If have problems setting a static IP with a gateway in another subnet. Is this not possible?

For example, this is not working for me:

```
IPAddress ip(192, 168, 1, 20);
IPAddress gateway(192, 168, 2, 1);
IPAddress subnet(255, 255, 255, 0);
if (!WiFi.config(ip, gateway, subnet)){
      Serial.println("Wifi config failed");
}
```

This is working:

```
IPAddress ip(192, 168, 1, 20);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
if (!WiFi.config(ip, gateway, subnet)){
      Serial.println("Wifi config failed");
}
```

Pablo2048, 2019-07-25:
Well - gateway has to be in same subnet (this is what is gateway for :-) ). If you set gateway ip address outside actual subnet, it will become unreachable.

TheChief79, 2019-07-25:
Hm, strange, the WiFi AP has also a different gateway address. I can ping the AP and the esp is connected to it. But i cant ping the esp.

Pablo2048, 2019-07-25:
Maybe the AP has different net mask that suit the gateway address. You are pinging the ESP from WiFi device or wired one? If from WiFi there can be set Station isolation mode at the AP so you are unable to "see" other devices on WiFi.

TD-er, 2019-07-25:
192.168.x.y with a subnet of 255.255.255.0 can only reach IPs with the same x and different y values.
If you want to access IPs with different x values, you must lower the number of bits set to '1' in the subnet.
For example 255.255.254.0 has 9 bits set to 0.
What you need is 10 bits set to 0, so you need 255.255.252.0

Please note that the gateway also must use a larger subnet or else the packets cannot be sent back to the other subnet.
All nodes in the same subnet should use the same subnet or else you will see strange issues of unreachable nodes.

earlephilhower, 2019-07-25:
Thanks, @TD-er and @Pablo2048.  Closing.

closed 2019-07-25 by earlephilhower