esp8266/Arduino #6352: Some of my ESP8266 won't boot with this firmware

Nalorokk, 2019-07-27:
Some of my ESP8266 based boards (NodeMCU) won't boot with any sketch (even blink.io). But they do work with stock AT firmware or some other custom firmwares.

With Arduino firmware it stucks on boot like this:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

And that its, nothing happens

Nalorokk, 2019-07-27:
Seems like it is somehow related to flasher "esptool.py", when i'm export .bin from arduino ide, and flashing it's manually using "ESP8266 FIRMWARE DOWNLOAD TOOL V3.6.4" it works every time. When i'm flashing from platform io or arduino ide using standard settings: it does not work.

5chufti, 2019-07-27:
may be related to wiring and type of flash chip.
try setting flash mode to DIO or DOUT.
But this is definitely  not related to esp8266 arduino core.
So please cloe this issue and look for help in relevant forum like for platform io or www.esp8266.com

closed 2019-07-27 by devyte