esp8266/Arduino #6355: Including <iostream> will cause a crash on boot with version 2.5.1 and 2.5.2

dannyvanlierop, 2019-07-29:
### Basic Infos

- [✓] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [✓] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [✓] I have tested that the issue is present in current master branch (aka latest git).
- [✓] I have searched the issue tracker for a similar issue.
- [✓] If there is a stack dump, I have decoded it.
- [✓] I have filled out all fields below.

#### Platform

- Hardware:				[other]
- Core Version:			[2.5.1 and 2.5.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 mini]
- Flash Mode:			[not sure] 
- Flash Size: 		        [4MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[not sure]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200] (serial upload only)

### Problem Description

Including <iostream> will cause a crash on boot with version 2.5.1 and 2.5.2 of ESP8266/Arduino.

### Sketch

```cpp

#include <iostream>

void loop()
{
};
void setup()
{
};

```

### Debug Messages

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

Hopefully this is helpfull.

With kind regards, 
Danny

d-a-v, 2019-08-02:
That is quite expected.
Try with git version, and enable exceptions in tools menu.
Think however that we only have ~50KB of ram when nothing is included.
What would be that for ?

earlephilhower, 2019-08-02:
GIT head fixes this.  This was related to exceptions, but it works whether or not exceptions are enabled in the menu.

````
#include <iostream>
void loop()
{
  static int cnt=0;
 std::cout << "count: " << cnt++ << "\n";
 delay(1000); 
};
void setup()
{
  Serial.begin(115200);
  Serial.printf("\n\nhello\n");
};
````
gives
````
hello
count: 0
count: 1
count: 2
count: 3
count: 4
count: 5
count: 6
count: 7
...
````

closed 2019-08-02 by earlephilhower