esp8266/Arduino #6359: [FR]Item menu for Puya chips

luc-github, 2019-07-30:
Hi, 
I got several requests from people not able to enable Puya patch.   
If for common developer this action looks trivial :
```
Create platform.local.txt in esp8266 core directory and add these lines:

compiler.c.extra_flags= -DPUYA_SUPPORT=1
compiler.cpp.extra_flags={compiler.c.extra_flags}
like explained here: https://github.com/esp8266/Arduino/pull/5504#issuecomment-490111960
``` 

but for  many people, new in arduino development, they do not understand what actually need to be done, so I got several people who ,even after reading these lines,  are not able to get their esp8266 with puya chip working with SPIFFS and may be also littleFS in next release
Not mentioning people using google translator to understand it...

I know current menu is already heavy, but I think it would be a great benefit for new people to have an item menu to enable/not disable the puya patch, additionnaly there was a lot of work done on this patch,  so make it visible is also good IMHO.

What do you think ?

d-a-v, 2019-07-30:
I agree with enabling PUYA support by default, even without menu to disable it.
Today we need `-DPUYA_SUPPORT=1`.
I propose to change this to use instead `-DPUYA_SUPPORT=0` to disable it.
It also would remain compatible with those who already have `-DPUYA_SUPPORT=1` in their environment.

TD-er, 2019-07-30:
Just keep in mind the remark made here by @igrr : https://github.com/esp8266/Arduino/pull/5504#pullrequestreview-185428376
If that's not an issue, then I think having it enabled by default would save a lot of frustration.

closed 2019-07-31 by devyte

luc-github, 2019-07-31:
Thank you ^_^

adapso, 2019-12-20:
I'm on this list of beginners.
What core version is the platform.local.txt file to be created in?
His content is only: compiler.c.extra_flags= -DPUYA_SUPPORT=1
compiler.cpp.extra_flags={compiler.c.extra_flags}
like explained here: https://github.com/esp8266/Arduino/pull/5504#issuecomment-490111960

devyte, 2019-12-20:
@adapso you shouldn't need to do anything, puya support is enabled by default in our core, and has been for several releases now.

adapso, 2019-12-20:
Unfortunately with me it does not work.  I have about 30 PUYA units that I can't use.  I currently use Esp8266 arduino IDE 1.8 and core 2.4.0

d-a-v, 2019-12-20:
PUYA chips are not supported in release 2.4.0.
They are supported by default starting from 2.6.0.

adapso, 2020-02-27:
I have the following problem, I was waiting for a correction for the PUYA problem for over a year.  This week after resuming the project I upgraded to the core version 2.6.3 of ESP826, for my happiness ESP returned to save the information correctly in the flash using SPIFFS, but to my sadness this version caused some problem with the ESP connection, I fired  a notification using the Pushover API.  In core version 2.4.2 the shooting works very well, but in version 2.6.3 it is not working.  Can anyone help?

PR #6362 (2019-07-31): enable PUYA flash chips support by default 
It can be disabled with` -DPUYA_SUPPORT=0`.

closes #6359