esp8266/Arduino #6364: could not compile "Blink" example. Broken python symbolic link

ZachreyNM, 2019-07-31:
Just did a fresh install from Windows7 to Ubuntu 18.04.2 LTS
Just did fresh install of Arduino IDE from 
https://www.arduino.cc/download.php?f=/arduino-1.8.9-linux64.tar.xz
using "Archive Manager"
I extracted to the default location and then ran the install.sh script
then I added the URL in the Prefs and installed ESP8266in boards manager and loaded ESP8266 Blink example and pressed compile and got :
"fork/exec xxx/.arduino15/packages/esp8266/tools/python/3.7.2-post1/python: no such file or directory
Error compiling for board NodeMCU 1.0 (ESP-12E Module)."

I noticed broken symbolic link at:
~/.arduino15/packages/esp8266/tools/python/3.7.2-post1$
"python" points to /usr/bin/python, which does not exist in Ubuntu 18.04.2 LTS

So executed the following command:
~/.arduino15/packages/esp8266/tools/python/3.7.2-post1$ ln -sfn /usr/bin/python3 python

And pressed compile and it worked! But the ASUS K52F Core i3 laptop became unresponsive to keyboard and mouse. but then came back after the process finished.

Not sure if this is the right thing to do. But maybe the install script needs to be edited??

earlephilhower, 2019-07-31:
Thanks for the detailed explanation.  Can't do much about the compile, that's between you and virtual memory and cores, but the summons we can see if there is a better way to go about it.  On my install of 1804 I have Python 2 at /use/bin/python, and it seems most others have it, to (even though v2 is EOL next year).  Might be a /usr/bin/env way of doing it, if python 3 is not going to live at the usual spot.  That'd break lots of scripts in the real world, I imagine, so I'm surprised they'd drop it...

ZachreyNM, 2019-08-01:
Thank you Earl! 
Just to clarify, the code compiled just fine after I fixed the broken symbolic link to /usr/bin/python, which does not exist on my computer (I changed it to /usr/bin/python3).
I also compiled the basic web client example in Arduino IDE just fine after the change. I have not actually flashed anything yet but I am confident that it will work just fine. 

But I think it would throw a lot of people initially when the compiler would conk out when it could not find a valid python folder. I am still learning the Linux side of things since this is my first installation on a laptop (I got it working on a couple of OrangePi SBC running off a uSD card so far).

earlephilhower, 2019-08-02:
Hmm, looks like the Ubuntu Cloud VM image (not normally installed by an end user on bare metal) doesn't include /usr/bin/python at all.  Maybe others.

As Python 2 is EOL next year, we should probably move to `/usr/bin/python3` everywhere, including our other scripts.  Even using `/usr/bin/env python` won't help this case.

earlephilhower, 2019-10-15:
Closing as we went Python3 with 3 months to spare. :)

closed 2019-10-15 by earlephilhower

ZachreyNM, 2019-10-15:
Thank you so much for your careful attention on this issue! May YHVH bless you and keep you!