esp8266/Arduino #6365: WiFiClientSecureBearSSL.h & compiler issue - follow up

marcostrullato, 2019-07-31:
Hi all,

I'm following up again about my issue. Could you please give me time to reply before closing issues?

I can clearly reproduce the behavior. I have started from scratch, removing anything related to Arduino.

Since I can reproduce the issue, and the library conflict looks due to something shipped with the software, I'd appreciate if you could take a look
. 
I understand I can potentially go around a delete some files to avoid the conflict, however I'd be better such libraries don't hit each other from a default installation..  I really believe I'm not doing something wrong, I think the conflict is between the code shipped with the board and with arduino ide. Am I wrong? As you see form the error, the libraries are not from file system positions of the additional libraries. 

Steps to reproduce:

1) install arduino ide 1.8.9 (latest)
2) From the board manager, add https://arduino.esp8266.com/stable/package_esp8266com_index.json
3) Add the new board, esp8266, choosing the latest version
4) select the wemos d1 mini board
5) add the libraries for Adafruit_BMP085.h, NTPClient.h and PubSubClient.h. All other libraries are installed by default, and can't be removed from the GUI.
6) Run the following code:

```
#include <Adafruit_BMP085.h>
#include <DNSServer.h>
#include <ESP8266WiFi.h>
#include <EEPROM.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <NTPClient.h>
#include <PubSubClient.h>
#include <stdlib.h>
#include <WiFiClient.h>
#include <WiFiUdp.h>
#include <Wire.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

With this code I still get the following error:

Arduino: 1.8.9 (Linux), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecure.h:41:0,
Multiple libraries were found for "WiFiUdp.h"
                 from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiServerSecure.h:20,
                 from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/ESP8266WiFi.h:41,
                 from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.h:29,
                 from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServerSecureBearSSL.h:26,
                 from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServerSecureBearSSL.cpp:28:
 Used: /home/marco/opt/arduino-1.8.9/libraries/WiFi
 Not used: /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:258:36: error: expected ')' before '*' token
     WiFiClientSecure(ClientContext *client, const X509List *chain, unsigned cert_issuer_key_type,
                                    ^
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:260:35: error: expected ')' before '*' token
     WiFiClientSecure(ClientContext* client, const X509List *chain, const PrivateKey *sk,
                                   ^
In file included from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecure.h:41:0,
                 from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiServerSecure.h:20,
                 from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/ESP8266WiFi.h:41,
                 from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.h:29,
                 from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/Parsing.cpp:25:
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:258:36: error: expected ')' before '*' token
[... 139 lines cut ...]
/home/marco/opt/arduino-1.8.9/libraries/WiFi/src/WiFiServer.h:31:7: note:   candidate expects 1 argument, 2 provided
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp: In destructor 'virtual ESP8266WebServer::~ESP8266WebServer()':
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp:87:11: error: 'class WiFiServer' has no member named 'close'
   _server.close();
           ^
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp: In member function 'virtual void ESP8266WebServer::begin(uint16_t)':
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp:105:21: error: no matching function for call to 'WiFiServer::begin(uint16_t&)'
   _server.begin(port);
                     ^
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp:105:21: note: candidate is:
In file included from /home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp:26:0:
/home/marco/opt/arduino-1.8.9/libraries/WiFi/src/WiFiServer.h:38:8: note: virtual void WiFiServer::begin()
   void begin();
        ^
/home/marco/opt/arduino-1.8.9/libraries/WiFi/src/WiFiServer.h:38:8: note:   candidate expects 0 arguments, 1 provided
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp: In member function 'virtual void ESP8266WebServer::close()':
/home/marco/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp:342:11: error: 'class WiFiServer' has no member named 'close'
   _server.close();
           ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

JiriBilek, 2019-07-31:
You should not include 
```
#include <WiFiClient.h>
#include <WiFiUdp.h>

```
The two includes are for another wifi library. You should include only the wifi libraries that came with ESP distribution.

d-a-v, 2019-07-31:
@marcostrullato There were already two answers,

https://github.com/esp8266/Arduino/issues/6363#issuecomment-516729940
https://github.com/esp8266/Arduino/issues/6361#issuecomment-516591895

now @JiriBilek 's answer just above
https://github.com/esp8266/Arduino/issues/6365#issuecomment-517015950

Are these these answers sufficient for you to fix your issue ?

devyte, 2019-07-31:
@marcostrullato you have now been told 3 times what the problem is. This is not a support forum, this is an issue tracker, meant for tracking issues in the core. If you need further help, please ask at a forum like esp8266.com or stackoverflow.
Closing.

closed 2019-07-31 by devyte