esp8266/Arduino #6369: Feature request: implement ClientContext::wait_until_acked()

d-a-v, 2019-08-01:
Per #6348 and #6351,
It seems `ClientContext::wait_until_sent()` is not sufficient for a fully terminated transmission.

d-a-v, 2021-02-27:
Title was: "Feature request: implement ClientContext::wait_until_acked()"
Per #7896, renaming to "OTA: Investigate workaround delay when receiving 'OK'" implemented by #6351.

Miq1, 2021-04-02:
I would like to add my observations when chasing a missing "OK" response upon a successful authorization. It turned out that basically a router bug was responsible for the packet loss, but it seems to me slightly naive in the ArduinoOTA code to rely on the successful transmission of a single UDP packet. UDP is per se subject to packet losses, so at least the packets should be sent twice or thrice to increase the reliability. This was in the end what saved me from the bug, as the router (an AVM Fritzbox with OS 7.21) would only drop the very first packet.

d-a-v, 2022-06-11:
#7693 would have to be taken into consideration in this issue