esp8266/Arduino #6375: problem whit program

programnoober, 2019-08-02:
[image]
I
[image]
 bought the wifi-shield esp-wroom-02. I want to connect it whit my phone but i tryed several programs to test it but the program arduino IDE gif me always a error that says that failed to connect to ESP8266: Timed out waiting for packet header. I don't know what I need to do for to make sure the connecting don't failed and which program I need to use? Can someone help me?

d-a-v, 2019-08-02:
Check your shield documentation. The two blue jumpers probably have a "programmation" mode.

programnoober, 2019-08-02:
I just tried several combinations of the blue jumpers but I still get that error. or the error that my port didn't work when i set the first switch different.

devyte, 2019-08-02:
@programnoober this is not a support forum, this is an issue tracker, meant for tracking issues in the core. Please seek help at a forum like esp8266.com or stackoverflow.
Closing.

closed 2019-08-02 by devyte